ptrkrysik/gr-gsm #130: airprobe_rtlsdr.py: ImportError: cannot import name Range

ijarraki, 2015-09-25:
Same as issue #125 when compiling/installing on a clean system state (which has never had gr-gsm installed). I've tested the last few commits, same result (and yes, I did run "su"/sudo ldconfig).

I've added
>[grc]
>local_blocks_path=/usr/local/share/gnuradio/grc/blocks

to /home/user/.gnuradio/config.txt and ran an old airprobe_rtlsdr.grc from gnuradio-companion, the resulting .py is working fine.

ptrkrysik, 2015-10-01:
Hi @ijarraki ,

can you send the airprobe_rtlsdr.py generated by you?

ijarraki, 2015-10-02:
https://pastebin.com/HhvRQQ9w

djaCarletonS, 2015-10-11:
Hey there!

I have been working on this for a while and just found your thread.  I am having the same issue.  I have done what you mention above with no joy.  when I pull up the airprobe_rtlsdr.grc through gnuradio-companion I can see that there are 4 missing blocks. 3 are the variable_gtGui_range blocks and the other is the gtgui_freq_sink_x.
I also ran ldconfig and added the config.txt and a config.conf.  I am really not sure why this is not working.
I am about to try and find out if I can recreate the blocks in gnuradio-companion but if I cannot would you mind sending either the .grc file or the airprobe_rtlsdr.py generated by you?

ptrkrysik, 2015-10-12:
Hi @ijarraki,

I've commented out lines incompatible with GNU Radio 3.7.3 from airprobe_rtlsdr.py. 

@djaCarletonS 
It seems that your problem is different as you don't see GNU Radio that should be available in gnuradio-companion.

djaubin, 2015-10-14:
The problem was trying to install on a raspberry pi B+ with kali.

I have since reformatted an old MSI laptop I had. Downloaded UBUNTU 15.04, updated and upgraded 2 times and then went through the process. SUCCESS!!

I am now trying to understand the process of use.
I have found GSM850 siganls and try to view them in wireshark. Currently it seems as if I get one packet with a type one request but it also shows the packets rolling through quite quickly.

I have also tested capture and decode and both work. I just need to learn how it all works.
I was unsuccessful at seeing more than one packet. Until I see stuff flying accross the wireshark screen I will leave airprobe_decode alone.

Thank you for your quick reply.
hopefully I can get a cfile with useful data so that I can show my professor!

This is very exciting!

closed 2016-01-04 by ptrkrysik

rojavatechops, 2017-04-11:
I'm having same the issues with software compatibility, I think. Please let me know what gnu-radio version to downgrade to or help me make the script below compatible. Please and thank you. 

#!/usr/bin/env python
##################################################
# Gnuradio Python Flow Graph
# Title: Gr-gsm Livemon
# Author: Piotr Krysik
# Description: Interactive monitor of a single C0 channel with analysis performed by Wireshark (command to run wireshark: sudo wireshark -k -f udp -Y gsmtap -i lo)
# Generated: Tue Apr 11 13:59:26 2017
##################################################

from PyQt4 import Qt
from PyQt4.QtCore import QObject, pyqtSlot
from gnuradio import blocks
from gnuradio import eng_notation
from gnuradio import gr
from gnuradio import qtgui
from gnuradio.eng_option import eng_option
from gnuradio.filter import firdes
from math import pi
from optparse import OptionParser
import PyQt4.Qwt5 as Qwt
import grgsm
import osmosdr
import pmt
import sip
import sys

from distutils.version import StrictVersion
class grgsm_livemon(gr.top_block, Qt.QWidget):

    def __init__(self, args="", fc=939.4e6, gain=30, osr=4, ppm=0, samp_rate=2000000.052982, shiftoff=400e3):
        gr.top_block.__init__(self, "Gr-gsm Livemon")
        Qt.QWidget.__init__(self)
        self.setWindowTitle("Gr-gsm Livemon")
        try:
             self.setWindowIcon(Qt.QIcon.fromTheme('gnuradio-grc'))
        except:
             pass
        self.top_scroll_layout = Qt.QVBoxLayout()
        self.setLayout(self.top_scroll_layout)
        self.top_scroll = Qt.QScrollArea()
        self.top_scroll.setFrameStyle(Qt.QFrame.NoFrame)
        self.top_scroll_layout.addWidget(self.top_scroll)
        self.top_scroll.setWidgetResizable(True)
        self.top_widget = Qt.QWidget()
        self.top_scroll.setWidget(self.top_widget)
        self.top_layout = Qt.QVBoxLayout(self.top_widget)
        self.top_grid_layout = Qt.QGridLayout()
        self.top_layout.addLayout(self.top_grid_layout)

        self.settings = Qt.QSettings("GNU Radio", "grgsm_livemon")
        self.restoreGeometry(self.settings.value("geometry").toByteArray())

        ##################################################
        # Parameters
        ##################################################
        self.args = args
        self.fc = fc
        self.gain = gain
        self.osr = osr
        self.ppm = ppm
        self.samp_rate = samp_rate
        self.shiftoff = shiftoff

        ##################################################
        # Variables
        ##################################################
        self.ppm_slider = ppm_slider = ppm
        self.g_slider = g_slider = gain
[... 249 lines cut ...]
    parser.add_option("-f", "--fc", dest="fc", type="eng_float", default=eng_notation.num_to_str(939.4e6),
        help="Set fc [default=%default]")
    parser.add_option("-g", "--gain", dest="gain", type="eng_float", default=eng_notation.num_to_str(30),
        help="Set gain [default=%default]")
    parser.add_option("", "--osr", dest="osr", type="intx", default=4,
        help="Set OSR [default=%default]")
    parser.add_option("-p", "--ppm", dest="ppm", type="eng_float", default=eng_notation.num_to_str(0),
        help="Set ppm [default=%default]")
    parser.add_option("-s", "--samp-rate", dest="samp_rate", type="eng_float", default=eng_notation.num_to_str(2000000.052982),
        help="Set samp_rate [default=%default]")
    parser.add_option("-o", "--shiftoff", dest="shiftoff", type="eng_float", default=eng_notation.num_to_str(400e3),
        help="Set shiftoff [default=%default]")
    (options, args) = parser.parse_args()
    if(StrictVersion(Qt.qVersion()) >= StrictVersion("4.5.0")):
        Qt.QApplication.setGraphicsSystem(gr.prefs().get_string('qtgui','style','raster'))
    qapp = Qt.QApplication(sys.argv)
    tb = grgsm_livemon(args=options.args, fc=options.fc, gain=options.gain, osr=options.osr, ppm=options.ppm, samp_rate=options.samp_rate, shiftoff=options.shiftoff)
    tb.start()
    tb.show()
    def quitting():
        tb.stop()
        tb.wait()
    qapp.connect(qapp, Qt.SIGNAL("aboutToQuit()"), quitting)
    qapp.exec_()
    tb = None #to clean up Qt widgets

ptrkrysik, 2017-04-11:
@rojavatechops, can you try to run grgsm_livemon.grc file in
gnuradio-companion and let me know what is the result.

rojavatechops, 2017-04-12:
[grgsm_livemon.grc.txt](https://github.com/ptrkrysik/gr-gsm/files/916109/grgsm_livemon.grc.txt)