ptrkrysik/gr-gsm #135: GSM input adaptor(9): insufficient connected input ports (1 needed, 0 connected)

Mohitsharma44, 2015-12-09:
I am trying to use airprobe_rtlsdr.py using bladerf on an Ubuntu 14.10 machine. 
I have installed everything using pybombs but running the airprobe_rtlsdr.py (with or without sudo) returns GSM input adaptor(9): insufficient connected input ports (1 needed, 0 connected).
I am attaching the traceback:
```
linux; GNU C++ version 4.9.1; Boost_105500; UHD_003.010.git-89-gb17b440d

gr-osmosdr v0.1.4-54-g2ca720cf (0.1.5git) gnuradio v3.7.8.1-297-ga463fecc
built-in source types: file osmosdr fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy 
Opening nuand bladeRF with device identifier string: "*:instance=0"
 Serial # 282d...eb42
 FW v1.8.0 FPGA v0.4.1
Automatic DC correction mode is not implemented.
[INFO @ bladerf.c:696] Clamping bandwidth to 1500000Hz
Using Volk machine: avx_64_mmx
Traceback (most recent call last):
  File "/usr/local/bin/airprobe_rtlsdr.py", line 270, in <module>
    tb.start()
  File "/usr/local/lib/python2.7/dist-packages/gnuradio/gr/top_block.py", line 109, in start
    top_block_start_unlocked(self._impl, max_noutput_items)
  File "/usr/local/lib/python2.7/dist-packages/gnuradio/gr/runtime_swig.py", line 3668, in top_block_start_unlocked
    return _runtime_swig.top_block_start_unlocked(*args, **kwargs)
RuntimeError: GSM input adaptor(9): insufficient connected input ports (1 needed, 0 connected)
```

I tested the sdr using osmocom_fft and I can see the waveforms, so I guess the blade is installed correctly. Let me know if you need anymore information.

Mohitsharma44, 2015-12-09:
Looking at the traceback I think it was the Volk library causing the issue but runing osmocom_fft uses avx_64_mx and works without any issue.

krisk84, 2015-12-12:
I installed everything using pyboms on Linux Mint 17 and am experiencing the same issue:

linux; GNU C++ version 4.8.4; Boost_105400; UHD_003.010.git-101-g4a1cb1f2

gr-osmosdr v0.1.4-55-g7cec4c0f (0.1.5git) gnuradio v3.7.8.1-306-g484423e3
built-in source types: file osmosdr fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy 
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Elonics E4000 tuner
Exact sample rate is: 2000000.052982 Hz
Using Volk machine: sse4_2_64
Traceback (most recent call last):
  File "/usr/local/bin/airprobe_rtlsdr.py", line 270, in <module>
    tb.start()
  File "/usr/local/lib/python2.7/dist-packages/gnuradio/gr/top_block.py", line 109, in start
    top_block_start_unlocked(self._impl, max_noutput_items)
  File "/usr/local/lib/python2.7/dist-packages/gnuradio/gr/runtime_swig.py", line 3668, in top_block_start_unlocked
    return _runtime_swig.top_block_start_unlocked(*args, **kwargs)
RuntimeError: GSM input adaptor(9): insufficient connected input ports (1 needed, 0 connected)

ptrkrysik, 2015-12-12:
Please open apps/airprobe_rtlsdr.grc in gnuradio-companion and send
screenshot of the flowgraph.

p0acher-101, 2015-12-12:
Same issue on a fresh Ubuntu install with gr-gsm from pybombs.  (In fact same issue also on a new Kali install as well.)  
[image]

p0acher-101, 2015-12-12:
Output from Grc:
<<< Welcome to GNU Radio Companion v3.7.8.1-297-ga463fecc >>>

Preferences file: /home/poacher/.gnuradio/grc.conf
Block paths:
	/usr/local/share/gnuradio/grc/blocks
	/home/poacher/.grc_gnuradio

Loading: "airprobe.grc"
>>> Done

Showing: "/home/poacher/airprobe.grc"

Generating: '/home/poacher/airprobe_rtlsdr.py'

Executing: '/usr/bin/python2 -u /home/poacher/airprobe_rtlsdr.py'

linux; GNU C++ version 4.8.4; Boost_105400; UHD_003.010.git-89-gb17b440d

gr-osmosdr v0.1.4-54-g2ca720cf (0.1.5git) gnuradio v3.7.8.1-297-ga463fecc
built-in source types: file osmosdr fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy 
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Exact sample rate is: 2000000.052982 Hz
[R82XX] PLL not locked!
Using Volk machine: avx_64_mmx
Traceback (most recent call last):
  File "/home/poacher/airprobe_rtlsdr.py", line 294, in <module>
    main()
  File "/home/poacher/airprobe_rtlsdr.py", line 283, in main
    tb.start()
  File "/usr/local/lib/python2.7/dist-packages/gnuradio/gr/top_block.py", line 109, in start
    top_block_start_unlocked(self._impl, max_noutput_items)
  File "/usr/local/lib/python2.7/dist-packages/gnuradio/gr/runtime_swig.py", line 3668, in top_block_start_unlocked
    return _runtime_swig.top_block_start_unlocked(*args, **kwargs)
RuntimeError: GSM input adaptor(9): insufficient connected input ports (1 needed, 0 connected)

>>> Done

Mohitsharma44, 2015-12-12:
I don't know if it helps but I tried debugging by checking the output of every block using fft sink and everything works fine till the input reaches gsm input adaptor block.

ptrkrysik, 2015-12-12:
I can't reproduce the problem. Can you describe exactly how did you
install GNU Radio and gr-gsm?

Mohitsharma44, 2015-12-12:
I had first installed the bladerf drivers from ppa as mentioned here: https://github.com/Nuand/bladeRF/wiki/Getting-Started%3A-Linux#Easy_installation_for_Ubuntu_The_bladeRF_PPA

Then I followed the instructions for installing gr-gsm from the wiki (using pybombs): https://github.com/ptrkrysik/gr-gsm/wiki/Installation

It didn't work and crashed with above mentioned errors so I installed (only) libosmocore (to check using osmo_fft) based on instructions here: http://gmr.osmocom.org/trac/wiki/GettingStarted

And osmo_fft worked without issues but airprobe still gave me same errors.

Ps: I have formatted my laptop yesterday again and will reinstall (Ubuntu 14.04) everything back on Monday. Let me know if you would like to see installation logs or want me to follow particular instructions when I'm installing gr-gsm.

ptrkrysik, 2015-12-12:
Ahhh... you are using bladerf. I haven't ever tested gr-gsm with it.

Maybe gr-osmosdr has some problem with support of bladerf. Can you run
any gnuradio-companion flowgraph with RTL-SDR source and your bladerf?

If yes:
Do you have maybe RTL-SDR dongle and can check if airprobe_rtlsdr starts
with it?

Mohitsharma44, 2015-12-12:
I did test the airprobe with NooElec rtl sdr dongle (I mentioned in my 1st comment) but it too fails with the exact same error.

krisk84, 2015-12-12:
Fresh install with bladerf removed hard (removed recipe, removed gr-gsm dep, removed all traces from system) and same error:

linux; GNU C++ version 4.8.4; Boost_105400; UHD_003.010.git-101-g4a1cb1f2

gr-osmosdr v0.1.4-55-g7cec4c0f (0.1.5git) gnuradio v3.7.8.1-311-g8c42ed29
built-in source types: file osmosdr fcd rtl rtl_tcp uhd hackrf rfspace airspy 
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Elonics E4000 tuner
Exact sample rate is: 2000000.052982 Hz
Using Volk machine: sse4_2_64
Traceback (most recent call last):
  File "/usr/local/bin/airprobe_rtlsdr.py", line 270, in <module>
    tb.start()
  File "/usr/local/lib/python2.7/dist-packages/gnuradio/gr/top_block.py", line 109, in start
    top_block_start_unlocked(self._impl, max_noutput_items)
  File "/usr/local/lib/python2.7/dist-packages/gnuradio/gr/runtime_swig.py", line 3668, in top_block_start_unlocked
    return _runtime_swig.top_block_start_unlocked(*args, **kwargs)
RuntimeError: GSM input adaptor(9): insufficient connected input ports (1 needed, 0 connected)

Requested screenshot attached.
[image]

DesignerMix, 2015-12-14:
*Sorry for my bad english

Hi i'll get the same problem. I use Ubuntu 15.10 x64, GR-GSM installed with help of pybombs, HackRF One with firmware 2015.07.2. 

Earlier you asked if was any problem with installation, and i decide to tell - While i installing gr-gsm using pybomb i get problem because talloc was not installed, so i'll install it from tar.gz and after that start installing gr-gsm with pybombs again, now all install correctly.

I want noted that - if i delete block GSM Input adaptor, connect Rotator directly to GSM Receiver and execute flowgraph - all start working... But of course it work incorrect i think.

After that i clone this git and start airprobe_rtlsdr.grc from apps folder, here data from gnuradio-companion and screenshot (but i have small screen and i redistributed blocks):

```
Showing: "/home/designermix/gr-gsm/apps/airprobe_rtlsdr.grc"

Generating: '/home/designermix/gr-gsm/apps/airprobe_rtlsdr.py'

Executing: '/usr/bin/python2 -u /home/designermix/gr-gsm/apps/airprobe_rtlsdr.py'

linux; GNU C++ version 5.2.1 20151010; Boost_105800; UHD_003.010.git-101-g4a1cb1f2

gr-osmosdr v0.1.4-55-g7cec4c0f (0.1.5git) gnuradio 3.7.9rc1
built-in source types: file osmosdr fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy 
Number of USB devices: 15
USB device 1d50:6089: 000000000000000015d463dc18817825 match
Using HackRF One with firmware 2015.07.2 
gr::fft: can't import wisdom from /home/designermix/.gr_fftw_wisdom
Using Volk machine: sse4_1_64
Traceback (most recent call last):
  File "/home/designermix/gr-gsm/apps/airprobe_rtlsdr.py", line 294, in <module>
    main()
  File "/home/designermix/gr-gsm/apps/airprobe_rtlsdr.py", line 283, in main
    tb.start()
  File "/usr/local/lib/python2.7/dist-packages/gnuradio/gr/top_block.py", line 109, in start
    top_block_start_unlocked(self._impl, max_noutput_items)
  File "/usr/local/lib/python2.7/dist-packages/gnuradio/gr/runtime_swig.py", line 3668, in top_block_start_unlocked
    return _runtime_swig.top_block_start_unlocked(*args, **kwargs)
RuntimeError: GSM input adaptor(9): insufficient connected input ports (1 needed, 0 connected)
```

[image]

ptrkrysik, 2015-12-14:
W dniu 14.12.2015 o 07:25, DesignerMix pisze:
> *Sorry for my bad english
>
> Hi i'll get the same problem. I use Ubuntu 15.10 x64, GR-GSM installed
> with help of pybombs, HackRF One with firmware 2015.07.2.
>
> Earlier you asked if was any problem with [...]
Hi,

There is issue with installation - libosmocore requires talloc now, but
the recipe in pybombs wasn't updated to install it. If I install it using
apt-get install libtalloc-dev
then pybombs is able to compile libosmocore.

However I think that what you guys are experiencing might be caused by
something different. I will look into this at the end of this week.

-- 
Best Regards,
Piotr Krysik

DesignerMix, 2015-12-14:
>There is issue with installation - libosmocore requires talloc now, but
the recipe in pybombs wasn't updated to install it. If I install it using
apt-get install libtalloc-dev

Thanks, now i will know how to install talloc without tar.gz. and understand the problem a little.

> However I think that what you guys are experiencing might be caused by
something different.

Maybe it not important but - i remember that - while i install pybombs it installing gnuradio without problem but after installation it not associated with .grc and i install gnuradio once again with comand sudo apt-get install gnuradio. Maybe this is wrong and i need uninstall gnuradio or it is not a problem? I'm noob in linux, but Personally i think that is not problem because all block in gnuradio don't missing.

> I will look into this at the end of this week.

I, and in my opinion, other people, will wait, hope problem will be solved. Thanks for you attention

ptrkrysik, 2015-12-14:
I suggest that you remove completely gnuradio installed from Ubuntu's
repository, install talloc with command that I provided and install
gnuradio and gr-gsm from source with pybombs. I don't know if it'll help
because I'm still not able to reproduce the problem. I've used gr-gsm
with gnuradio installed both from source and from repository on Ubuntu
15.10.

Mohitsharma44, 2015-12-14:
It gives me the same error on Ubuntu 15.10 as well. (pybombs installs bladerf so I did not install it manually. I can confirm this since osmo_fft does work.)

I performed a fresh install of gr-gsm on Ubuntu 15.10:
```
sudo apt-get install git
sudo apt-get install libtalloc-dev
<and then the instructions for installation using pybombs>
```
> ps: Just a heads up to always keep `builddocs=OFF` in `./pybombs config` otherwise it will crash when installing gnuradio.

ptrkrysik, 2015-12-14:
@Mohit, to be sure, you are not following the part of the guide where
forcebuild parameter of pybombs is changed?

Mohitsharma44, 2015-12-14:
No. I followed the instructions from Wiki page here: https://github.com/ptrkrysik/gr-gsm/wiki/Installation

ptrkrysik, 2015-12-14:
Did you invoke:
./pybombs config forcebuild ' '
?
(in order gnuradio from source you shouldn't)

Mohitsharma44, 2015-12-14:
I see, from the wiki I inferred that since Ubuntu 15.10 has GNURadio >= 3.7.3, I should run `force build` to avoid the time that gnu takes to build from source.
Let me do a fresh install and I'll update you. Just to confirm, I will be running all the same steps Except ` ./pybombs config forcebuild ' ' `

Mohitsharma44, 2015-12-14:
@ptrkrysik So when I do it without forcebuild, It stops after installing ` boost_1_53_0 `. I re-execute the pybombs command to install gr-gsm and this time it populates the list of remaining packages and crashes on installing _uhd_. I have copied the log of installation here: http://pastebin.com/gFHJ6nU2

DesignerMix, 2015-12-14:
@Mohitsharma44 try to execute with -v switch, like this ```sudo ./pybombs install -v gr-gsm``` it will show all log and maybe you can find why it crash.

@ptrkrysik i install fresh ubuntu 15.10, then execute ```apt-get install libtalloc-dev```, then follow this instruction https://github.com/ptrkrysik/gr-gsm/wiki/Installation but without ```./pybombs config forcebuild ' '```. All installed normally and without problem.

But when i run flowgraph again by running ```designermix@SDR:~/pybombs/src/gr-gsm/apps$ gnuradio-companion airprobe_rtlsdr.grc```, i get the same error.

Mohitsharma44, 2015-12-14:
@DesignerMix I tried with -v it checks for installed components and then starts installing uhd but it crashes when building uhd package. (same error log as I have copied in above link). It is weird same distro but different results. I'll try again before calling it a day.

TotalDay, 2015-12-15:
RuntimeError: GSM input adaptor(9): insufficient connected input ports (1 needed, 0 connected)

Kali Linux 2

14 hours of installing and testing and no luck

Istalled via apt-get
Remove all and intalled from source
Installed to /usr - then to /usr/local and to opt - no luck

installed all dependences - no luck

устал и заебался - нихуя не работает.

J4r3tt, 2015-12-18:
I have Ubuntu 15.10 and same problem

zdroidan, 2015-12-18:
kali sana and same problem.

cahebebe, 2015-12-19:
I have the same problem in Ubuntu 14.04.

cahebebe, 2015-12-19:
@ptrkrysik if you wish I can give to you access to my ubuntu installation.

ptrkrysik, 2015-12-20:
I'm trying to reproduce the bug but I'm unable to obtain it. What would
be very helpful is virtualbox virtualmachine with installed gr-gsm and
the problem present.

cahebebe, 2015-12-20:
Yes, I’m currently have a virtual machine with Ubuntu with the problem present.

Teamviewer data:
ID: 727040156
Password: 3646

Please check it.

Regards,

Carlos

p0acher-101, 2015-12-20:
Likewise I have a Debian and an Ubuntu VM both with the problem.

carxan, 2015-12-21:
It is not solution, but
change in "GSM Input Adaptor" ===> "Advanced" ====> "Core Affinity" > = 0

cahebebe, 2015-12-21:
@ptrkrysik I've sent a mega link to your email.

radioman1511, 2015-12-21:
I changed the Value "Core Affinity" to > = 0, but it is running to Using Volk machine: sse4_1_64 and then stopped and is doing nothing.

Generating: '/home/sdrhts/gr-gsm/apps/airprobe_rtlsdr.py'

Executing: '/usr/bin/python2 -u /home/sdrhts/gr-gsm/apps/airprobe_rtlsdr.py'

linux; GNU C++ version 5.2.1 20151010; Boost_105800; UHD_003.010.git-101-g4a1cb1f2

gr-osmosdr v0.1.4-57-g55fe9619 (0.1.5git) gnuradio 3.7.9rc1
built-in source types: file osmosdr fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy soapy redpitaya 
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Elonics E4000 tuner
Exact sample rate is: 2000000,052982 Hz
Using Volk machine: sse4_1_64

>>> Done

cahebebe, 2015-12-21:
I had the same result.

carxan, 2015-12-21:
Not tested!
http://radiolay.ru/viewtopic.php?f=55&t=339

Mohitsharma44, 2015-12-21:
@carxan : I don't think it has the solution or the error that we are facing.
@cahebebe : Did you send the vdi or vmdk files using megalink to piotr? or should I?

cahebebe, 2015-12-21:
I've uploaded a zip file with VirtualBox files using Mega, I sent this to @ptrkrysik yesterday.

carxan, 2015-12-21:
page 2 radiolay.ru to test:

- VMWare xubuntu gnuradio 3.6 + Wireshark airprobe RTL-SDR + gqrx and 
- VMWare xubuntu + 3.7 + gnuradio wireshark + gr-GSM (airprobe RTL-SDR) + gqrx

Not have vmware 11
translator use

ptrkrysik, 2015-12-21:
Hi all, I see the problem on the virtualmachine from @cahebebe . I'm trying to find the reason.

Mohitsharma44, 2015-12-21:
@ptrkrysik : Thanks. Looking forward to understand what is causing the problem.
In the meantime, the link to the virtual image provided by @carxan works perfectly fine (I'm sorry, I missed the link somehow in my first glance.) Thanks for that.

haruotsa, 2015-12-22:
Hi! Could the above problem be related to the one discussed here on hier_block2 (?):
http://gnuradio.blogspot.co.at/2015/09/re-discuss-gnuradio-connecting-message.html

ptrkrysik, 2015-12-22:
@haruotsa - it could be the reason of the problem. I haven't found out
for sure yet, but I can share how I was able to install gr-gsm on the
virtual machine that I got from @cahebebe. It has Ubuntu 15.10
installed, so the GNU Radio version that is in the repository is
compatible with gr-gsm.

1. remove old installation of GNU Radio (here I removed everything from
 /usr/local directory)

2. download pybombs:
 git clone https://github.com/pybombs/pybombs.git

3. configure pybombs:

 cd pybombs
 ./pybombs

#-when asked about target directory give: /usr/local
#-when asked about "focebuid" give empty string: ' '
#(for some reason command "./pybombs config forcebuild ' ' " stopped
working in current pybombs

4. install libtalloc required by libosmocore (this should be installed
by pybombs, but it is not):
 sudo apt-get install libtalloc-dev

5. install gr-gsm with bybombs:
 sudo ./pybombs install gr-gsm

cahebebe, 2015-12-22:
I have a successful install in VirtualBox with Ubuntu Deskup x64 15.10, 

The issue is the ./pybombs config forcebuild ' ' doesn't work, then when asked in forcebuild left a space (with the space key).

fairlight1337, 2015-12-22:
I can confirm @ptrkrysik's experience, installing `gr-gsm` from pybombs works just fine. In fact, I removed everything gnuradio-related that was installed through debs and re-installed everything through pybombs. No forcebuild needed. Now everything works just nicely, except for running the grc (mein problem persists).

radioman1511, 2015-12-23:
no chance

radioman1511, 2015-12-23:
in virtualbox can i confirm works probely  without issue , but on reality machine with new installation ubuntu 15.10 not greetings

fairlight1337, 2015-12-23:
What version of Ubuntu did you install in your virtual machine?

radioman1511, 2015-12-23:
Hello in virtualbox i am using Ubuntu 15.10  64bit from osboxes.org. Merry Christmas

carxan, 2015-12-23:
1)Ubuntu 15.10 clean/pybombs:
last version libosmocore/libtalloc-dev not compatible with gr-gsm/gnuradio/etc
2))Replace latest version the libosmocore:
http://bb.osmocom.org/trac/wiki/libosmocore
-install preconditions
-libosmocore-0.6.6 does not use libtalloc-dev
Download: http://cgit.osmocom.org/libosmocore/
unzip and rename ==> "libosmocore"
3)-Install pybombs
   -git clone https://github.com/ptrkrysik/gr-gsm.git
4) -copy "version old" libosmocore ==> /pybombs/src/
    change file pybombs/recipes/libosmocore.lwr:
    comment:
    #source: git://git://git.osmocom.org/libosmocore.git
6) cd /pybombs
  sudo ./pybombs install gr-gsm
7)WARNING: no sources available for package libosmocore!
repeat: sudo ./pybombs install gr-gsm
good luck!

bikhial, 2015-12-23:
Same Problem Here, kali2 installed from source.
no chance.

radioman1511, 2015-12-24:
@carxan for me a good idea. I make it step by step but the same problem on the end.

finkata, 2015-12-24:
I am facing the same problem on Arch.

linux; GNU C++ version 5.2.0; Boost_105900; UHD_003.009.001-0-unknown

gr-osmosdr v0.1.4-55-g7cec4c0f (0.1.5git) gnuradio 3.7.9rc1
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy 
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
Exact sample rate is: 2000000.052982 Hz
Using Volk machine: avx_64_mmx_orc
Traceback (most recent call last):
  File "/usr/bin/airprobe_rtlsdr.py", line 270, in <module>
    tb.start()
  File "/usr/lib/python2.7/site-packages/gnuradio/gr/top_block.py", line 109, in start
    top_block_start_unlocked(self._impl, max_noutput_items)
  File "/usr/lib/python2.7/site-packages/gnuradio/gr/runtime_swig.py", line 4397, in top_block_start_unlocked
    return _runtime_swig.top_block_start_unlocked(r, max_noutput_items)
RuntimeError: GSM input adaptor(7): insufficient connected input ports (1 needed, 0 connected)

J4r3tt, 2015-12-25:
@ptrkrysik it works use ./pybombs to set config

pitikay, 2015-12-25:
I am getting the same error, tried with newest libosmocore and v 0.6.6 on ubuntu 15.10 and kali 2, no luck...

zdfrizzell1, 2015-12-25:
@carxan I believe I tried your method using Ubuntu 14.04 on VMware player, perhaps that was my issue. Here's what I did
I already had the newest libosmocore built from source. 
Then I downloaded libosmocore 0.6.6. Cloned pybombs. 
I placed the 0.6.6 unzipped libosmocore into pybombs/src and commented out the source: git://git://git.osmocom.org/libosmocore.git in pybombs/recipes/libosmocore.lwr
Then installed gr-gsm by running sudo ./pybombs install gr-gsm twice.
Installed successfully but still get GSM input adaptor(9): insufficient connected input ports (1 needed, 0 connected) when running gnuradio-companion airprobe_rtlsdr.grc in terminal. Execute graph
Thanks for the help, any other suggestions I will try.

carxan, 2015-12-25:
Ubuntu Software Center
search: libtalloc
NOT installed libtalloc-dev

haruotsa, 2015-12-25:
Here's my 2 cent:
I installed the Gnu Radio v3.7.9rc1 on Linux Mint 17.1 using Pybombs and received the same error as described in the title. After some googling I went on and "downgraded" the lines of gnuradio-runtime/lib/hier_block2_detail.cc regarding "hier_in" & "hier_out" variables as defined for releases before 3.7.9rc1. (See https://github.com/gnuradio/gnuradio/pull/694/files). With this experimental mod and after a fresh build of Gnu radio the airprobe_rtlsdr.grc became alive. Nonetheless, without expertice on inner life of GR I cannot tell if this is a real solution. Probably not. It would be the best to have one of the main contributors to look at this gr-gsm issue.

ptrkrysik, 2015-12-26:
@haruotsa, thank you for poitning this. It is probable that the source
of the problem is in GNU Radio. If it is so it would be good to have
some minimal working example of the bug in order to show it to the GNU
Radio developers, so they fix the issue.

finkata, 2015-12-27:
I can confirm that rolling back ./gnuradio-runtime/lib/hier_block2_detail.cc to the previous version of the file fixes the issue.

I used this:

1. git checkout e4aff27 gnuradio-runtime/lib/hier_block2_detail.cc
2. build gnuradio

Thanks

[30 more replies]

commit 8715da08 (2016-01-06): Fixed all hierarchical blocks in the project
by adding a hier_block class that solves incompatibility between GNU Radio >=3.7.9 and <=3.7.8
and inheriting all hierarchical blocks from it.
This commit fixes #135
closed 2016-01-06 by ptrkrysik