ptrkrysik/gr-gsm #136: No supported devices found to pick from

d3crypt, 2016-01-13:
root@kali:~/gr-gsm/apps# airprobe_rtlsdr.py 
linux; GNU C++ version 4.9.1; Boost_105500; UHD_003.007.003-0-unknown
I have installed gr-gsm manually and it still isn't working. I can verify its not my Rtlsdr because it works fine in gqrx. 

Here is the output when I try to run it:
root@kali:~/gr-gsm/apps# airprobe_rtlsdr.py 
linux; GNU C++ version 4.9.1; Boost_105500; UHD_003.007.003-0-unknown

gr-osmosdr v0.1.4-67-gac15e789 (0.1.5git) gnuradio 3.7.5
built-in source types: file fcd rtl_tcp uhd rfspace redpitaya 

FATAL: No supported devices found to pick from.

Trying to fill up 1 missing channel(s) with null source(s).
This is being done to prevent the application from crashing
due to gnuradio bug #528.

Using Volk machine: ssse3_64_orc

ptrkrysik, 2016-01-14:
Can you connect in gnuradio-companion RTL-SDR source to null sink and
check if it works?

d3crypt, 2016-01-15:
I will do so and post back.

d3crypt, 2016-01-21:
I fixed it. It was a problem with how i build gr-osmosdr.

closed 2016-02-14 by ptrkrysik

drsn0w, 2016-10-02:
How did you fix this?

Dr4s1l, 2017-08-03:
How did you fix this?

velichkov, 2017-08-03:
Hi @Dr4s1l, @drsn0w,

Check that the gr-osmosdr is compiled with RTL support. When compiling from source you could specify `-DENABLE_RTL=ON` cmake option and if you see the following error

```
$cmake -DENABLE_RTL=ON ..

-- Configuring Osmocom RTLSDR support...
--   Dependency LIBRTLSDR_FOUND = FALSE
CMake Error at /usr/lib/x86_64-linux-gnu/cmake/gnuradio/GrComponent.cmake:75 (message):
  user force-enabled Osmocom RTLSDR but configuration checked failed
Call Stack (most recent call first):
  lib/CMakeLists.txt:156 (GR_REGISTER_COMPONENT)
```
then you need to install the librtlsdr library and header files. On ubuntu 16.04 you could install it from the repository
```
sudo apt-get install librtlsdr-dev
```

Dr4s1l, 2017-08-03:
Workshop for me. On fedora using pybombs, install allô package, do
```
pybombs inv rtl-sdr
Pybombs remove rtl-sdr
pybombs install rtl-sdr
```
Workshop better with sudo for me

Rajatdongre, 2023-05-29:
if i am using hackrf what should I do ? for osmocomm. i also getting same error