ptrkrysik/gr-gsm #137: ImportError: cannot import name Range

ijarraki, 2016-01-25:
Ref issue #125 

This still happens with the latest gr-gsm on Debian Jessie and Kali (sana/2.0).

ijarraki, 2016-01-26:
I resolved this issue by compiling/installing latest/git gnuradio, libosmocore, rtl-sdr and gr-osmosdr before compiling/installing gr-gsm.

closed 2016-01-26 by ijarraki

jarodcs, 2017-07-20:
Hi, just want to share that I had this error:

xxxxxxx@debian:~/Documents/gr-gsm$ grgsm_livemon 
Traceback (most recent call last):
  File "/usr/local/bin/grgsm_livemon", line 32, in <module>
    import osmosdr
ImportError: No module named osmosdr          <<<<<<<<<<<<<<<<<< ERROR! <<<<<<<<

I and pass this through once  running
sudo apt-get install gr-osmosdr 
(No need to compile/install again, but just in case)

Cheers