ptrkrysik/gr-gsm #139: compiling gr-gsm

slick97477, 2016-02-05:
I am on Kali 2.0 sana 64 bit. I am trying to compile and build gr-gsm from git clone of this and when running the "cmake .." i am getting this error.

-- Build type not specified: defaulting to release.
-- Could NOT find Boost
CMake Error at CMakeLists.txt:64 (message):
  Boost required to compile gr-gsm

-- Configuring incomplete, errors occurred!
See also "/root/gr-gsm/build/CMakeFiles/CMakeOutput.log"

I am trying to get this functioning on Kali 2.0 sana. I have not tried the pybombs method of installing yet, have heard some serious horror stories about pybombs destroying installations, not sure if there are true but dont want to risk just yet, I am on a fresh install of kali 2.0 sana(not rolling release(too many bugs still)). Anyhelp would be appreciated. I am very familiar with kali linus have been using it since the backtrack days. Thank You

Yradio, 2016-02-06:
Take a look at the old **[Manual compilation and installation](https://github.com/ptrkrysik/gr-gsm/wiki/Manual-compilation-and-installation)** page.

KALI 2.0 **gnuradio** pakages are:
   `apt-get install gnuradio gnuradio-dev`

KALI 2.0 **rtl-sdr** packages are:
   `apt-get install rtl-sdr librtl-sdr-dev`

KALI 2.0 **GrOsmoSDR** pakages are:
   `apt-get install osmo-sdr libosmosdr-dev`

KALI 2.0 **liboscomore** packages are:
   `apt-get install libosmocore libosmocore-dev`

KALI 2.0 **libusb** pakages are also needed:
   `apt-get install libusb-1.0.0 libusb-dev`

And in order to build **gr-gsm**, the following packages must be installed:
   `apt-get install cmake **libboost-all-dev** libcppunit-dev swig doxygen liblog4cpp5-dev python-scipy`

scoyok, 2016-02-06:
Hello! My brother and I have successfully installed and decoded gsm using the rtl-sdr and the hackrf one. We literally just finshed yesterday with the last trials, we are putting together a full tutorial and should have it ready tomorrow. Its running perfect on Sana and rolling release. If you are interested in trialing our tutorial and giving us some feedback feel free to email us at 

 twobrothersdevs@gmail.com 

I posted to another community member yesterday in the kali

scoyok, 2016-02-06:
EDIT****

Kali.org forums.  Sorry sending from my phone and bumped comment. Hope this helps!

scoyok, 2016-02-08:
Done in Kali Rolling release. Pardon the formatting google did some weird stuff when I hit publish.

https://docs.google.com/document/d/1E_LQZ0xTs697L-0QsANQ7sDzc09QhO_7q4ue_WXezI0/pub

ptrkrysik, 2016-04-10:
@Yradio can you translate that into step by step guide how to install gr-gsm on Kali 2.0?

closed 2016-04-10 by ptrkrysik