ptrkrysik/gr-gsm #140: Using several RTL-SDR devices

MatejKovacic, 2016-02-11:
It would be nice if there would be an option to have connected several USB dongles and use a specific one (similar like kalibrate has a switch -d : rtl-sdr device index).

That would enable user to scan all the active frequencies and then (with the other RTL-SDR device) automatically tune to each one of them and capture a traffic for a while. For instance, you could detect there are 5 bases stations nearby, tune to which of them for a few minutes and collect some data, for instance gsm_a.rr.algorithm_identifier in Wireshark to detect bases stations using A5/0 cipher (i.e. IMSI catchers).

ptrkrysik, 2016-02-14:
I've added ability to select the device. Please look into the commit 7185b6664fc86e5ad75d0e1da9b2b0a5eebd20ae to see if this is what is needed. 
If yes I will close the issue.

MatejKovacic, 2016-02-15:
Hi,

> I've added ability to select the device. Please look into the commit
> 7185b66
> <https://github.com/ptrkrysik/gr-gsm/commit/7185b6664fc86e5ad75d0e1da9b2b0a5eebd20ae>
> to see if this is what is needed.
> If yes I will close the issue.

Great. Exactly what I had in mind.

Regards,

M.

-- 
PGP Fingerprint: 3B74 637D 8409 53F9 A704  F27C BEA5 286D A9CF 4A88
PGP Key:
https://keyserver.ubuntu.com/pks/lookup?op=get&search=0xBEA5286DA9CF4A88

closed 2016-02-15 by ptrkrysik