ptrkrysik/gr-gsm #141: Saving scanning results into a file

MatejKovacic, 2016-02-11:
It would be useful to have an option (switch) to write (append) results into a text file (or maybe database). My idea is to have an option to save the following data into tab-delimited or CSV file:

- timestamp
- ARFCN
- Freqency
- CID
- LAC
- MCC
- MNC
- Pwr
- some additional data (for instance ID of sensor, GPS location, etc.)

These additional data would be passed as parameter to an application, for instance:
`airprobe_rtslsdr_scanner.py -save-to-file <filename> -additional-parameter <GPS_X=xxx:GPS_Y=yyy:someother_data>`

The file would then look like:
```
date;time; ARFCN; Freq; CID; LAC; MCC; MNC; Pwr;additionals
2016-02-11;22:37:24;11;958.0M;122;50;292;21;-26;"GPS_X=xxx:GPS_Y=yyy:someother_data"
```

When importing this into the database it would be very simple to split the last column into several separate columns (in a given case separator is a semicolon).

Actually, something like this could be already done with some bash programming:
1. Create a file loopscan.sh:

```
#!/bin/bash

echo "date;time;ARFCN;Freq;CID;LAC;MCC;MNC;Pwr" > data.csv
while true;
do
  tmpf=$(tempfile)
  grtime="$(date +"%Y-%m-%d;%H":%M:%S)"
  echo "Scanning GSM... [Current scan started at: $grtime. Hit CTRL+C to
stop!]"
  stdbuf -oL airprobe_rtlsdr_scanner.py -p 34 >&1 > $tmpf
  more +3 $tmpf | awk -v x="$grtime" 'BEGIN {FS="[:, ]+"} {print
x";"$2";"$4";"$6";"$8";"$10";"$12";"$14}' >> data.csv
  rm $tmpf
done
```

2. Run it:
`sh loopscan.sh`

3. Stop it with **Ctrl-C** and view the results:
`cat data.csv`

Data from data.csv now can be imported into the database or analyzed in spreadsheet app... Anyway, timestamp here is fixed for the whole pass, and it would really be nice to have an option to save data into a file directly from GSM scanner.

MatejKovacic, 2016-02-12:
Example of analysis:
```

create table gsmscan (date date, time time, arfcn text, freq text, cid text, lac text, mcc text, mnc text, pwr text);
\COPY gsmscan from 'data.csv' with csv header delimiter ';';
alter table gsmscan add column bs_id text;
update gsmscan set bs_id = arfcn || '-' || freq || '-' || cid || '-' || lac || '-' || mcc || '-' || mnc;

select freq, count(*) as num_of_detections, min(pwr) as min_pwr, max(pwr) as max_pwr from gsmscan group by freq, bs_id order by freq, num_of_detections;
```

Result:
```
  freq  | num_of_detections | min_pwr | max_pwr 
--------+-------------------+---------+---------
 935.2M |                 1 | -30     | -30
 935.2M |                64 | -28     | -33
 936.2M |                 2 | -35     | -37
 936.2M |                 4 | -35     | -36
 936.2M |                 5 | -33     | -37
 937.2M |                 3 | -57     | -59
 937.2M |                 4 | -54     | -58
 937.2M |                11 | -56     | -59
 937.6M |                 1 | -57     | -57
 937.8M |                 1 | -55     | -55
 937.8M |                 4 | -49     | -56
 937.8M |                12 | -54     | -56
 938.6M |                65 | -26     | -26
 939.4M |                65 | -27     | -27
 956.6M |                65 | -26     | -27
 958.0M |                65 | -26     | -27
```

MatejKovacic, 2016-02-14:
There should be another option - when scanner is run in the loop, it should be autokalibrated each pass or each n-passes. This is important, because some RTL-SDR devices need to be rekalibrated after they warm up.

Also, in ouptut CSV file there should also be exported RTL-SDR device ID and kalibration parameter. That will be useful for analysis how kalibration of a RTL-SDR device changes over time (does it stabilize, does it start flipping,...).

petterreinholdtsen, 2017-09-17:
You can get an effect similar to the one you explain by using grgsm_livemon (or the -headless variant) for decoding with the code in https://github.com/Oros42/IMSI-catcher.git to collect information and store it in a sqlite database.

ptrkrysik, 2026-11-09:
The script this refers to (airprobe_rtlsdr_scanner.py) no longer exists, and logging/looping can be done with a small wrapper script around grgsm_scanner output, so we consider this out of scope for the core project; a new focused request is welcome if still needed.

Thank you for your interest in the project.

closed 2026-11-09 by ptrkrysik