ptrkrysik/gr-gsm #142: Automatic kalibration of RTL-SDR devices

MatejKovacic, 2016-02-11:
The idea is to add automatic kalibration of RTL-SDR devices to _airprobe_rtlsdr_scanner_ (**-p parameter**).

Right now this could be done with this script (first you need to install Kalibrate (i. e. kal):

```
tmpf=$(tempfile)
echo "Kalibrating... step 1"
stdbuf -oL kal -s GSM900 >&1 > $tmpf
channel=$(more +2 $tmpf | awk 'BEGIN {FS="[:, ]+"} {print $6"-"$2}' | sort | awk 'BEGIN {FS="[-, ]+"} {print $2}' | head -1)
stdbuf -oL kal -c $channel >&1 > $tmpf
ppm=$(tail -n 1 $tmpf | awk 'BEGIN {FS="[:, ]+"} {print $4}' | awk '{print int($1+0.5)}')
echo "Kalibration completed!"
echo "Run: airprobe_rtlsdr_scanner.py -p $ppm"
rm $tmpf
```

OK, it is ugly as hell, but it is working. Basically it scans GSM900 band and picks up the strongest channel. Then it computes average ppm for this channel and prints it.

P. S. It is good to know, that a RTL-SDR device needs re-kalibration after it warms up (after several minutes).

dydfrancis, 2020-04-05:
please sir why can't you update us with video tutorial on this i really need to learn

dydfrancis, 2020-04-05:
please can somebody assist me on this whasapp +2348087070541 thanks

ptrkrysik, 2026-11-09:
Frequency calibration is handled by external tools such as kalibrate-rtl, and the script mentioned here no longer exists, so we are closing this as out of scope.

Thank you for your interest in the project.

closed 2026-11-09 by ptrkrysik