ptrkrysik/gr-gsm #143: compile from source: cannot find -lgrgsm

sysrun, 2016-02-18:
All dependencies are met, cmake show now problems - running make, this is what happens:
>[ 75%] Built target grgsm_swig_swig_2d0df
>Scanning dependencies of target _grgsm_swig
>[ 77%] Building CXX object swig/CMakeFiles/_grgsm_swig.dir/grgsm_swigPYTHON_wrap.cxx.o
>Linking CXX shared module _grgsm_swig.so
>/usr/bin/ld: cannot find -lgrgsm
>collect2: error: ld returned 1 exit status
>make[2]: *** [swig/_grgsm_swig.so] Error 1
>make[1]: *** [swig/CMakeFiles/_grgsm_swig.dir/all] Error 2
>make: *** [all] Error 2

Any ideas?

sysrun, 2016-02-18:
Update: Commit 7185b6664fc86e5ad75d0e1da9b2b0a5eebd20ae is working - looks like the changes in 5c7a578e4fd8a55e7db15788a6cc6a04486e2a34 and/or 89cac10318884777423811e5569dfdb7b4db1c94 broke something

istais, 2016-02-19:
I can reproduce the issue

MatejKovacic, 2016-02-19:
Me too. With pybombs_legacy.

falcorocks, 2016-02-19:
N00b question, if 7185b66 is working, how do I install that version instead of the (I guess) last one that is installed automatically following the procedure showed [here](https://github.com/ptrkrysik/gr-gsm/wiki/Installation)?

istais, 2016-02-19:
wget https://github.com/ptrkrysik/gr-gsm/archive/7185b6664fc86e5ad75d0e1da9b2b0a5eebd20ae.zip
unzip 7185b6664fc86e5ad75d0e1da9b2b0a5eebd20ae.zip 
cmake CMakeLists.txt 
make 
make install

ptrkrysik, 2016-02-19:
I hope I've just corrected the problem. You can download the new version
of gr-gsm with:
git pull
called in install prefix directory in src/gr-gsm

Then go to gr-gsm/build directory, remove everything in there, and run:
cmake ..
make install

istais, 2016-02-19:
yes it's not fixed. Thanks

falcorocks, 2016-02-20:
Yep, now it works. Thank you very much @ptrkrysik

closed 2016-02-28 by ptrkrysik

peseta3, 2016-12-23:
I have the same Problem but i dont understand this.

> I hope I've just corrected the problem. You can download the new version of gr-gsm with: git pull called in install prefix directory in src/gr-gsm Then go to gr-gsm/build directory, remove everything in there, and run: cmake .. make install

Can you help me, I am a bad ubuntu user.

ptrkrysik, 2016-12-29:
@pseta3, this description was meant to speedup installation of gr-gsm
after I made a correction in the source. If you don't understand that
description, you can just remove gr-gsm and install it again from the
beginning. In the end the result will be the same.

peseta3, 2016-12-30:
Hallo Piotre,

 

thank you for your fast mail.

I have Ubuntu 14 installt as virtual mashine.

I did all as you write in the discribtion.

But I have also every time this errors.

 

PyBOMBS.Packager.source - ERROR - Configuration failed after running at least twice.
PyBOMBS.Packager.source - ERROR - Problem occurred while building package gr-gsm: Configuration failed

PyBOMBS.install_manager - ERROR - Error installing package gr-gsm. Aborting.

 

Best regards 

 

Sven (peseta3)