ptrkrysik/gr-gsm #144: Can't install gr-gsm

falcorocks, 2016-02-19:
I'm trying to install following this guide 
https://github.com/ptrkrysik/gr-gsm/wiki/Installation

But it fails all the time when installing the package gr-gsm.
Fails on Kali, Ubuntu 14.04, Ubuntu 15.10.
This is the output from terminal:

f..@ubuntu:~$ sudo pybombs install gr-gsm
(...)
|
\- gr-gsm
PyBombs.install - INFO - Installing package: gr-gsm
Building:    (100%) [====================================================]
[ 70%] Built target gnuradio-grgsm
[ 70%] Built target grgsm_swig_swig_doc
[ 71%] Built target _grgsm_swig_swig_tag
[ 75%] Built target grgsm_swig_swig_2d0df
Linking CXX shared module _grgsm_swig.so
/usr/bin/ld: cannot find -lgrgsm
collect2: error: ld returned 1 exit status
swig/CMakeFiles/_grgsm_swig.dir/build.make:90: recipe for target 'swig/_grgsm_swig.so' failed
make[2]: *** [swig/_grgsm_swig.so] Error 1
CMakeFiles/Makefile2:301: recipe for target 'swig/CMakeFiles/_grgsm_swig.dir/all' failed
make[1]: *** [swig/CMakeFiles/_grgsm_swig.dir/all] Error 2
Makefile:126: recipe for target 'all' failed
make: *** [all] Error 2
PyBombs.Packager.source - ERROR - Build failed. See output above for error messages.
PyBombs.Packager.source - ERROR - Problem occured while building package gr-gsm:
Build failed.
PyBombs.install - ERROR - Error installing package gr-gsm. Aborting.

istais, 2016-02-19:
it's the same issue with https://github.com/ptrkrysik/gr-gsm/issues/143

falcorocks, 2016-02-19:
yep, you are right! I’m still a bit of a n00b here in github, sorry.

closed 2016-02-28 by ptrkrysik