ptrkrysik/gr-gsm #146: Cannot write to prefix path `/usr/local'

MatejKovacic, 2016-02-19:
I am following instructions from wiki and after I write:
pybombs prefix init /usr/local -a default_prx

I got an error:
PyBombs.prefix - ERROR - Cannot write to prefix path `/usr/local'.

ptrkrysik, 2016-04-10:
I think that adding sudo solved this issue. In case it didn't I will reopen the issue.

closed 2016-04-10 by ptrkrysik

jacobitosuperstar, 2019-03-17:
hi, im here to comment that it didn't work

velichkov, 2019-03-17:
Hi @jacobitosuperstar,

What do you get when you execute it? Please provide the full command that you have executed including all arguments together with the whole output. And also which OS do you use and which version?

In my opinion the `pybombs` installation method is the hardest one and most error prone so my advice is to fist try some of the other installations methods described [here](https://osmocom.org/projects/gr-gsm/wiki/Installation).

jacobitosuperstar, 2019-03-17:
Yeah I just switched the installation and that’s it, sorry for the inconvenience... But in case it helps I use MacOS Mojave and i get the same error that guy above even after I use Sudo I get the same promp, “cannot write prefix path”

velichkov, 2019-03-17:
> I use MacOS Mojave

The MacOS is not well supported platform, I recommend you to first try installing and using gr-gsm on Linux, even in a VM.

> i get the same error that guy above even after I use Sudo I get the same promp, “cannot write prefix path”

I don't have any MacOS experience but maybe [this](https://support.apple.com/en-us/HT204012) and [this](
https://superuser.com/questions/1021141/problems-with-sudo-privileges-as-admin-in-osx) could help you resolve the `sudo` problem.