ptrkrysik/gr-gsm #147: Error installing package libosmocore

MatejKovacic, 2016-02-19:
When I say:

sudo pybombs install gr-gsm

I got this:

```
|
+- gr-osmosdr
|  |
|  +- hackrf
|  |
|  +- gnuradio
|  |  |
|  |  \- apache-thrift
|  |
|  +- bladeRF
|  |
|  +- gr-iqbal
|  |  |
|  |  +- libosmo-dsp
|  |  |
|  |  \- gnuradio
|  |     |
|  |     \- apache-thrift
|  |
|  +- osmo-sdr
|  |
|  \- rtl-sdr
|
+- gnuradio
|  |
|  \- apache-thrift
|
\- libosmocore
|
\- gr-gsm
   |
   +- gr-osmosdr
   |  |
   |  +- hackrf
   |  |
   |  +- gnuradio
   |  |  |
   |  |  \- apache-thrift
   |  |
   |  +- bladeRF
   |  |
   |  +- gr-iqbal
   |  |  |
   |  |  +- libosmo-dsp
   |  |  |
   |  |  \- gnuradio
   |  |     |
   |  |     \- apache-thrift
   |  |
   |  +- osmo-sdr
   |  |
   |  \- rtl-sdr
   |
   +- gnuradio
   |  |
   |  \- apache-thrift
   |
   \- libosmocore
PyBombs.install - INFO - Installing package: libosmocore
PyBombs.Fetcher - ERROR - Unexpected error while fetching git+git://git.osmocom.org/libosmocore.git.
PyBombs.Fetcher - ERROR - [Errno 2] No such file or directory: 'default_prx/src'
PyBombs.Packager.source - ERROR - Problem occurred while building package libosmocore:
Unable to fetch recipe libosmocore
PyBombs.install - ERROR - Error installing package libosmocore. Aborting.
```

ptrkrysik, 2016-02-28:
Hi Matej,

Are you able to install gr-gsm now?

MatejKovacic, 2016-02-28:
Hi,

> Are you able to install gr-gsm now?

In fact, I did manually install it, but have to manually compile and
install libosmocore too.

apt-cache search libosmocore returns no results...

Regards,

M.

-- 
PGP Fingerprint: 3B74 637D 8409 53F9 A704  F27C BEA5 286D A9CF 4A88
PGP Key:
https://keyserver.ubuntu.com/pks/lookup?op=get&search=0xBEA5286DA9CF4A88

ptrkrysik, 2016-04-10:
I'm not able to reproduce the problem.
In case you still have the problem - please post your ~/.pybombs/config.yml file.

closed 2016-04-10 by ptrkrysik

Gayathri-mohan, 2017-12-13:
research@research:~/gnuradio/build$ pybombs install glib
PyBOMBS - INFO - PyBOMBS Version 2.3.2
PyBOMBS.Packager.apt - INFO - Install python-apt to speed up apt processing.
PyBOMBS.install_manager - INFO - Phase 1: Creating install tree and installing binary packages:
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages were automatically installed and are no longer required:
  libsoapysdr0.5-2 linux-headers-4.8.0-36 linux-headers-4.8.0-36-generic
  linux-image-4.8.0-36-generic linux-image-extra-4.8.0-36-generic
Use 'sudo apt autoremove' to remove them.
The following additional packages will be installed:
  libpcre3-dev libpcre32-3 libpcrecpp0v5 zlib1g-dev
Suggested packages:
  libglib2.0-doc
The following NEW packages will be installed:
  libglib2.0-dev libpcre3-dev libpcre32-3 libpcrecpp0v5 zlib1g-dev
0 upgraded, 5 newly installed, 0 to remove and 8 not upgraded.
Need to get 2,054 kB/2,221 kB of archives.
After this operation, 12.5 MB of additional disk space will be used.
WARNING: The following packages cannot be authenticated!
  libglib2.0-dev
E: There were unauthenticated packages and -y was used without --allow-unauthenticated
PyBOMBS.PackageManager - ERROR - Something went wrong while trying to install glib using apt: Process returned value: 100
Install tree:
|
\- glib
PyBOMBS.install_manager - INFO - Phase 2: Recursively installing source packages to prefix:
PyBOMBS.install_manager - INFO - Installing package: glib
PyBOMBS.Packager.apt - INFO - Install python-apt to speed up apt processing.
You are using pip version 8.1.1, however version 9.0.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
PyBOMBS.Fetcher.wget - ERROR - Download using requests failed: [Errno 13] Permission denied: 'glib-2.27.2.tar.bz2'
PyBOMBS.Fetcher.wget - WARNING - Attempting to download using wget...
--2017-12-13 17:17:18--  http://ftp.gnome.org/pub/gnome/sources/glib/2.27/glib-2.27.2.tar.bz2
Resolving ftp.gnome.org (ftp.gnome.org)... 194.71.11.165, 194.71.11.173, 2001:6b0:19::165, ...
Connecting to ftp.gnome.org (ftp.gnome.org)|194.71.11.165|:80... connected.
HTTP request sent, awaiting response... 302 Found
Location: http://gensho.ftp.acc.umu.se/pub/gnome/sources/glib/2.27/glib-2.27.2.tar.bz2 [following]
--2017-12-13 17:17:18--  http://gensho.ftp.acc.umu.se/pub/gnome/sources/glib/2.27/glib-2.27.2.tar.bz2
Resolving gensho.ftp.acc.umu.se (gensho.ftp.acc.umu.se)... 194.71.11.176, 2001:6b0:19::176
Connecting to gensho.ftp.acc.umu.se (gensho.ftp.acc.umu.se)|194.71.11.176|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 6748249 (6.4M) [application/x-bzip2]
glib-2.27.2.tar.bz2: Permission denied

Cannot write to ‘glib-2.27.2.tar.bz2’ (Success).
PyBOMBS.Fetcher.wget - WARNING - Process returned value: 3
PyBOMBS.Packager.source - ERROR - Problem occurred while building package glib:
Unable to fetch recipe glib
PyBOMBS.install_manager - ERROR - Error installing package glib. Aborting

This is what i got.i tried a lot of instructions to clear it.but unable to remove the error.

Alster, 2017-12-14:
I see “Permission denied“
Try with sudo