ptrkrysik/gr-gsm #148: deadlock

alexsadik, 2016-02-20:
hello

sometimes i get deadlock when using the airprobe_decode.py (with my own cfile)
configuration does't matter BCCH or SDCCH same effect

OS Ubuntu 15.10
GnuRadio 3.7.8

strace -p (airprobe_decode pid):
futex(0x1bf2db4, FUTEX_WAIT_PRIVATE, 1, NULL

ptrkrysik, 2016-02-28:
Hi. I haven't use strace before. Can you tell me what the presented by you output mean (the " futex(0x1bf2db4, FUTEX_WAIT_PRIVATE, 1, NULL") and can you tell me how can I reproduce the problem (i.e. 'run grgsm_decode x times on a capture file y')?

ptrkrysik, 2016-04-10:
Hi, I can confirm that with GNU Radio 3.7.8 there is in fact deadlock. I'm working on finding this issue.
With GNU Radio installed from source on ubuntu 14.04 I didn't notice this problem.

@alexsadik it seems that you have some experience in diagnosing such problems. Are you able to help with it?

ptrkrysik, 2016-04-11:
Most probably this is a problem of GNU Radio prior to 3.7.9 version. I will try to prepare some simple reproduction of the bug that doesn't involve gr-gsm.

closed 2016-06-08 by ptrkrysik

ptrkrysik, 2016-06-09:
The problem doesn't occur on GNU Radio 3.7.9 - so the solution is to use this version of GNU Radio.