ptrkrysik/gr-gsm #152: If long time running livemon, it will output all zero value and display freeze.

nightcoffee, 2016-02-22:
If long time running livemon, it will output all zero value and display freeze.
[image]

nightcoffee, 2016-02-22:
Ubuntu 15.10 & RTL2823U with R820
also happen in Kali 2016.01.
But other applications(such SDR#, dump1090) are OK with long time running.

ptrkrysik, 2016-02-22:
How long you are running the program before the problem appears?

nightcoffee, 2016-02-22:
after few minutes, less than one hour.
I have another question, could I livemon two or three channel together by set wider bandwidth ?

ptrkrysik, 2016-02-22:
W dniu 22.02.2016 o 21:57, nightcoffee pisze:
> after few minutes, less than one hour.
> I have another question, could I livemon two or three channel together
> by set wider bandwidth ?
> 
Current livemon doesn't do that. But if your computer can process online
more channels than one - it is possible to setup gr-gsm blocks in such
way i.e. together with GNU Radio's polyphase channelizer.

nightcoffee, 2016-02-22:
I have modify the livemon.grc and replace the gsm_input with gsm_wideband_input.
But when I set the RTL2832U freq same as Wideband input settings, this problem appear again!
Note: It seems not Zero but "O"

Output:
Using Volk machine: avx2_64_mmx
Extracting channels [117, 120], given that the center frequency is at ARFCN 117 (958.4M)
ARFCN 117 is at C0 +0 KHz
ARFCN 120 is at C0 +600 KHz
OOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOO

nightcoffee, 2016-02-22:
Perhaps it relate to computre performance?
I am running it in VM which has 3 Core.(i5 6600K)

ptrkrysik, 2016-02-23:
It might be fault of the clock_offset_corrector (misc_utils/gsm_clock_offset_corrector.grc):

[image]

 I wanted to use as many of-the-shelf GNU Radio blocks as possible to implement it. But it was source of problems - when the ractional resampler was after the controlled rotator (more correct way of connecting the blocks as the rate of inputs at the rotator would be the same in this case) it ate randomly tags from the rotator. Currently the resampler is in front - this way rates of blue input and orange input of the rotator is slightly different. It might be causing problems. 

You can also try to add following change:
[image]
(there is additional fractional resampler resampling signal controlling frequency offset of the rotator)

zbt160, 2016-03-02:
I am facing the exact problem. I have tried both HackRF and RTLSDR but like after random period which is usually around half hour. @nightcoffee can you please update on whether you were able to solve this problem using the information provided or by any other means? 

The problem is annoying if you want to gather data for more than 1 hour for example. 

Please help.. Thank you

ptrkrysik, 2016-03-02:
The problem persists because it is hard to reproduce it in rational time
on my computer. I might need your cooperation in order to check if I
managed to fix it.

Mohitsharma44, 2016-03-10:
Similar issue here. No printing of O's though. It just stops recording after ~ 30-40mins. 
I wrote a small script that kills the process (along with child processes) and restarts recording every 10 minutes, but it still does stop recording. 
Device: Nuand Blade RF X40. 
dont know if it is device specific though.

zbt160, 2016-03-10:
I wrote a similar script to kill and turn on again and it is working fine in my case.

Mohitsharma44, 2016-03-21:
@zbt160 What's the *timeout* that you keep before restarting the application?

zbt160, 2016-03-21:
15 mins to be on safe side but i think 25 mins will also work.

ptrkrysik, 2016-03-21:
Just to let everyone know: I'm able to see the problem on my computer so
I can look for its cause.

tiger762, 2016-04-05:
Just FYI I work up to this problem this morning as well. I started grgsm_livemon in the evening. The wireshark trace has last packet received as 4am. So for me, it was about 8 hours. Could it be the clock in the RTL2832 drifts too much? I am quite happy with 8+ hours because the RTL dongle is 100 times cheaper than my Ettus B200 I use for OpenBTS :smile:

ptrkrysik, 2016-04-05:
W dniu 05.04.2016 o 16:24, tiger762 pisze:
> Just FYI I work up to this problem this morning as well. I started
> grgsm_livemon in the evening. The wireshark trace has last packet
> received as 4am. So for me, it was about 8 hours. Could it be the [...]

I think it is because of the way clock_offset_corrector is implemented.
I insisted on using as many stock GNU Radio blocks in it but probably it
wasn't the best idea. I won't go into details (explaining it will
consume some time) unless someone offer help with changing the block
and/or checking if this is causing the problem.

ptrkrysik, 2016-06-08:
The problem seems to be resolved by the commit 799133a. The source of the problem was in fact clock offset corrector. Everyone interested in having this bug resolved - please pull new version, switch to development branch and recompile and reinstall gr-gsm.

ptrkrysik, 2016-06-09:
My tests show that the new (simplified) version of the clock offset corrector doesn't cause this problem. I'm not 100% happy with the solution as I had to remove sample rate offset correction, due to different bug in GNU Radio (http://www.gnuradio.org/redmine/issues/846). I'm closing the issue here and merging the fix into master. In case of problems with the fix - please report the problem here.

closed 2016-06-09 by ptrkrysik

ironiksk, 2017-06-15:
Hi, the problem seems has not to been resolved yet.
I'm running the grgsm_livemon on my BPI M2U (Raspbian OS) and I have the O's in the output from time to time:
[image]

Perhaps it relate to computre performance, but processor is busy only on 80% or less. 
Information about the hardware, system and installed software:
Hardware:
`AllWinner R40 4x1500MHz (ARM Cortex-A7)`
`2G DDR3 SDRAM`

System:
`PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"`
`NAME="Raspbian GNU/Linux"`
`VERSION_ID="8"`
`VERSION="8 (jessie)"`

Software:
`GNU Radio Companion 3.7.10` installed from `jessie-backports`
`gr-gsm` installed from master branch using `Installation on RaspberryPi 3` Wiki Guide.