ptrkrysik/gr-gsm #155: SSLv3_method’ was not declared in this scope

tibas21, 2016-02-24:
Hello,
I have this error message when i want to build. 
Can you help me?! 
May be another issue had already answer, didn't found.
Thanks
Th
root@kali:~# sudo pybombs install gr-gsm
|
+- gr-osmosdr
|  |
|  +- airspy
|  |
|  +- hackrf
|  |
|  +- uhd
|  |  |
|  |  \- swig
|  |
|  +- gnuradio
|  |  |
|  |  +- swig
|  |  |
|  |  +- apache-thrift
|  |  |
|  |  \- uhd
|  |     |
|  |     \- swig
|  |
|  +- bladeRF
|  |
|  +- gr-iqbal
|  |  |
|  |  +- libosmo-dsp
|  |  |
|  |  \- gnuradio
|  |     |
|  |     +- swig
|  |     |
|  |     +- apache-thrift
|  |     |
|  |     \- uhd
|  |        |
|  |        \- swig
|  |
|  +- osmo-sdr
|  |
|  \- rtl-sdr
|
\- gnuradio
   |
   +- swig
   |
   +- apache-thrift
   |
   \- uhd
      |
      \- swig
|
\- gr-gsm
   |
   +- gr-osmosdr
   |  |
   |  +- airspy
   |  |
   |  +- hackrf
   |  |
   |  +- uhd
   |  |  |
   |  |  \- swig
   |  |
   |  +- gnuradio
   |  |  |
   |  |  +- swig
   |  |  |
   |  |  +- apache-thrift
   |  |  |
   |  |  \- uhd
   |  |     |
   |  |     \- swig
   |  |
   |  +- bladeRF
   |  |
   |  +- gr-iqbal
   |  |  |
   |  |  +- libosmo-dsp
   |  |  |
   |  |  \- gnuradio
   |  |     |
   |  |     +- swig
   |  |     |
   |  |     +- apache-thrift
   |  |     |
   |  |     \- uhd
   |  |        |
   |  |        \- swig
   |  |
   |  +- osmo-sdr
   |  |
   |  \- rtl-sdr
   |
   \- gnuradio
      |
      +- swig
      |
      +- apache-thrift
      |
      \- uhd
         |
         \- swig
PyBombs.install - INFO - Installing package: swig
Lecture des listes de paquets... Fait
Construction de l'arbre des dépendances       
Lecture des informations d'état... Fait
Le paquet suivant a été installé automatiquement et n'est plus nécessaire :
  swig3.0
Veuillez utiliser « sudo apt autoremove » pour le supprimer.
Paquets suggérés :
  swig2.0-examples swig2.0-doc
Les paquets suivants seront ENLEVÉS :
  swig
Les NOUVEAUX paquets suivants seront installés :
  swig2.0
0 mis à jour, 1 nouvellement installés, 1 à enlever et 0 non mis à jour.
Il est nécessaire de prendre 0 o/1 150 ko dans les archives.
Après cette opération, 3 990 ko d'espace disque supplémentaires seront utilisés.
(Lecture de la base de données... 364353 fichiers et répertoires déjà installés.)
Suppression de swig (3.0.7-2) ...
Sélection du paquet swig2.0 précédemment désélectionné.
(Lecture de la base de données... 364346 fichiers et répertoires déjà installés.)
Préparation du dépaquetage de .../swig2.0_2.0.12-1_amd64.deb ...
Dépaquetage de swig2.0 (2.0.12-1) ...
Traitement des actions différées (« triggers ») pour man-db (2.7.5-1) ...
Paramétrage de swig2.0 (2.0.12-1) ...
PyBombs.install - INFO - Installation successful.
PyBombs.install - INFO - Installing package: uhd
Cloning:     (100%) [===============================================================================================================================]
Configuring: (100%) [===============================================================================================================================]
Building:    (100%) [===============================================================================================================================]]
Installing:  (100%) [===============================================================================================================================]
PyBombs.install - INFO - Installation successful.
PyBombs.install - INFO - Installing package: apache-thrift
Building:    (100%) [===============================================================================================================================]
make  all-recursive
make[1] : on entre dans le répertoire « /usr/local/src/apache-thrift »
Making all in compiler/cpp
make[2] : on entre dans le répertoire « /usr/local/src/apache-thrift/compiler/cpp »
make  all-am
make[3] : on entre dans le répertoire « /usr/local/src/apache-thrift/compiler/cpp »
make[3]: rien à faire pour « all-am ».
make[3] : on quitte le répertoire « /usr/local/src/apache-thrift/compiler/cpp »
make[2] : on quitte le répertoire « /usr/local/src/apache-thrift/compiler/cpp »
Making all in lib
make[2] : on entre dans le répertoire « /usr/local/src/apache-thrift/lib »
Making all in cpp
make[3] : on entre dans le répertoire « /usr/local/src/apache-thrift/lib/cpp »
Making all in .
make[4] : on entre dans le répertoire « /usr/local/src/apache-thrift/lib/cpp »
depbase=`echo src/thrift/transport/TSSLSocket.lo | sed 's|[^/]*$|.deps/&|;s|\.lo$||'`;\
/bin/bash ../../libtool  --tag=CXX   --mode=compile g++ -DHAVE_CONFIG_H -I. -I../.. -I../../lib/cpp/src/thrift  -I/usr/include  -I./src -D__STDC_LIMIT_MACROS  -Wall -Wextra -pedantic -DNDEBUG -std=c++11 -MT src/thrift/transport/TSSLSocket.lo -MD -MP -MF $depbase.Tpo -c -o src/thrift/transport/TSSLSocket.lo src/thrift/transport/TSSLSocket.cpp &&\
mv -f $depbase.Tpo $depbase.Plo
libtool: compile:  g++ -DHAVE_CONFIG_H -I. -I../.. -I../../lib/cpp/src/thrift -I/usr/include -I./src -D__STDC_LIMIT_MACROS -Wall -Wextra -pedantic -DNDEBUG -std=c++11 -MT src/thrift/transport/TSSLSocket.lo -MD -MP -MF src/thrift/transport/.deps/TSSLSocket.Tpo -c src/thrift/transport/TSSLSocket.cpp  -fPIC -DPIC -o src/thrift/transport/.libs/TSSLSocket.o
src/thrift/transport/TSSLSocket.cpp: In constructor ‘apache::thrift::transport::SSLContext::SSLContext(const apache::thrift::transport::SSLProtocol&)’:
src/thrift/transport/TSSLSocket.cpp:143:37: error: ‘SSLv3_method’ was not declared in this scope
     ctx_ = SSL_CTX_new(SSLv3_method());
SSLv3_method’ was not declared in this scope
     ctx_ = SSL_CTX_new(SSLv3_method());
                                     ^
Makefile:1292 : la recette pour la cible « src/thrift/transport/TSSLSocket.lo » a échouée
make[4]: *** [src/thrift/transport/TSSLSocket.lo] Erreur 1
make[4] : on quitte le répertoire « /usr/local/src/apache-thrift/lib/cpp »
Makefile:1586 : la recette pour la cible « all-recursive » a échouée
make[3]: *** [all-recursive] Erreur 1
make[3] : on quitte le répertoire « /usr/local/src/apache-thrift/lib/cpp »
Makefile:550 : la recette pour la cible « all-recursive » a échouée
make[2]: *** [all-recursive] Erreur 1
make[2] : on quitte le répertoire « /usr/local/src/apache-thrift/lib »
Makefile:620 : la recette pour la cible « all-recursive » a échouée
make[1]: *** [all-recursive] Erreur 1
make[1] : on quitte le répertoire « /usr/local/src/apache-thrift »
Makefile:542 : la recette pour la cible « all » a échouée
make: *** [all] Erreur 2
PyBombs.Packager.source - ERROR - Build failed. See output above for error messages.
PyBombs.Packager.source - ERROR - Problem occurred while building package apache-thrift:
Build failed.
PyBombs.install - ERROR - Error installing package apache-thrift. Aborting

ptrkrysik, 2016-02-24:
This is not bug of gr-gsm but problem with installation of apache thrift
on your computer - can you ask the question on the mailing list?

closed 2016-02-24 by ptrkrysik

tibas21, 2016-02-24:
I will, Thanks !

mamoonali, 2016-03-03:
plzz help me with same problem

tibas21, 2016-03-03:
i tried to repair and compil by my self the file, without succes.
I asked to the mailing list of apache Thrift, no answer today.
This is an incompatibility with Kali 2.0. 
I installed Ubuntu 15.10 at a 2nd hard drive and everything is ok.

ptrkrysik, 2016-03-03:
I think you should ask GNU Radio guys on their mailing list or github page.
This is some specific problem with GNU Radio installation. It might be
possible to disable requirement for apache thrift during manual compilation.

mamoonali, 2016-03-03:
Thanx a lot..can u help me with manual installation

bastienbaranoff, 2016-06-15:
open /usr/local/src/apche-thrift/lib/cpp/src/thrift/transport/TSSLSocket.cpp with your favorite text editor search for SSLv3 change the second occurrence   ctx_ = SSL_CTX_new(SSLv3_method()); by  ctx_ = SSL_CTX_new(SSLv23_method()); worked for me 
[TSSLSocket.cpp.zip](https://github.com/ptrkrysik/gr-gsm/files/317396/TSSLSocket.cpp.zip)

Penzimo, 2016-07-27:
thanks working on kali.

Wangyao14cyy, 2016-07-28:
thanks working on arch

Marat1990, 2016-11-12:
Thank you bastienbaranoff!!!

fkelbert, 2017-03-30:
Thanks, @bastienbaranoff .

kermit6000, 2017-03-30:
Thanks ! Fixed it on xubuntu 16.04 :)

dyurnozz, 2017-08-09:
src/thrift/transport/TSSLSocket.cpp: In constructor ‘apache::thrift::transport::SSLContext::SSLContext(const apache::thrift::transport::SSLProtocol&)’:
src/thrift/transport/TSSLSocket.cpp:172:24: warning: ‘const SSL_METHOD* TLSv1_method()’ is deprecated [-Wdeprecated-declarations]
     ctx_ = SSL_CTX_new(TLSv1_method());
                        ^~~~~~~~~~~~
In file included from /usr/local/include/openssl/ct.h:13:0,
                 from /usr/local/include/openssl/ssl.h:61,
                 from src/thrift/transport/TSSLSocket.cpp:48:
/usr/local/include/openssl/ssl.h:1612:1: note: declared here
 DEPRECATEDIN_1_1_0(__owur const SSL_METHOD *TLSv1_method(void)) /* TLSv1.0 */
 ^
src/thrift/transport/TSSLSocket.cpp:172:37: warning: ‘const SSL_METHOD* TLSv1_method()’ is deprecated [-Wdeprecated-declarations]
     ctx_ = SSL_CTX_new(TLSv1_method());
                                     ^
In file included from /usr/local/include/openssl/ct.h:13:0,
                 from /usr/local/include/openssl/ssl.h:61,
                 from src/thrift/transport/TSSLSocket.cpp:48:
/usr/local/include/openssl/ssl.h:1612:1: note: declared here
 DEPRECATEDIN_1_1_0(__owur const SSL_METHOD *TLSv1_method(void)) /* TLSv1.0 */
 ^
src/thrift/transport/TSSLSocket.cpp:174:24: warning: ‘const SSL_METHOD* TLSv1_1_method()’ is deprecated [-Wdeprecated-declarations]
     ctx_ = SSL_CTX_new(TLSv1_1_method());
                        ^~~~~~~~~~~~~~
In file included from /usr/local/include/openssl/ct.h:13:0,
                 from /usr/local/include/openssl/ssl.h:61,
                 from src/thrift/transport/TSSLSocket.cpp:48:
/usr/local/include/openssl/ssl.h:1618:1: note: declared here
 DEPRECATEDIN_1_1_0(__owur const SSL_METHOD *TLSv1_1_method(void)) /* TLSv1.1 */
 ^
src/thrift/transport/TSSLSocket.cpp:174:39: warning: ‘const SSL_METHOD* TLSv1_1_method()’ is deprecated [-Wdeprecated-declarations]
     ctx_ = SSL_CTX_new(TLSv1_1_method());
                                       ^
In file included from /usr/local/include/openssl/ct.h:13:0,
                 from /usr/local/include/openssl/ssl.h:61,
                 from src/thrift/transport/TSSLSocket.cpp:48:
/usr/local/include/openssl/ssl.h:1618:1: note: declared here
 DEPRECATEDIN_1_1_0(__owur const SSL_METHOD *TLSv1_1_method(void)) /* TLSv1.1 */
 ^
src/thrift/transport/TSSLSocket.cpp:176:24: warning: ‘const SSL_METHOD* TLSv1_2_method()’ is deprecated [-Wdeprecated-declarations]
     ctx_ = SSL_CTX_new(TLSv1_2_method());
                        ^~~~~~~~~~~~~~
In file included from /usr/local/include/openssl/ct.h:13:0,
                 from /usr/local/include/openssl/ssl.h:61,
                 from src/thrift/transport/TSSLSocket.cpp:48:
/usr/local/include/openssl/ssl.h:1624:1: note: declared here
 DEPRECATEDIN_1_1_0(__owur const SSL_METHOD *TLSv1_2_method(void)) /* TLSv1.2 */
 ^
src/thrift/transport/TSSLSocket.cpp:176:39: warning: ‘const SSL_METHOD* TLSv1_2_method()’ is deprecated [-Wdeprecated-declarations]
     ctx_ = SSL_CTX_new(TLSv1_2_method());
                                       ^
In file included from /usr/local/include/openssl/ct.h:13:0,
                 from /usr/local/include/openssl/ssl.h:61,
                 from src/thrift/transport/TSSLSocket.cpp:48:
/usr/local/include/openssl/ssl.h:1624:1: note: declared here
 DEPRECATEDIN_1_1_0(__owur const SSL_METHOD *TLSv1_2_method(void)) /* TLSv1.2 */
 ^
src/thrift/transport/TSSLSocket.cpp: In member function ‘virtual void apache::thrift::transport::TSSLSocket::close()’:
src/thrift/transport/TSSLSocket.cpp:363:5: warning: ‘void ERR_remove_state(long unsigned int)’ is deprecated [-Wdeprecated-declarations]
     ERR_remove_state(0);
     ^~~~~~~~~~~~~~~~
In file included from /usr/local/include/openssl/crypto.h:32:0,
                 from /usr/local/include/openssl/bio.h:20,
                 from /usr/local/include/openssl/err.h:21,
                 from src/thrift/transport/TSSLSocket.cpp:46:
/usr/local/include/openssl/err.h:248:1: note: declared here
 DEPRECATEDIN_1_0_0(void ERR_remove_state(unsigned long pid))
 ^
src/thrift/transport/TSSLSocket.cpp:363:23: warning: ‘void ERR_remove_state(long unsigned int)’ is deprecated [-Wdeprecated-declarations]
     ERR_remove_state(0);
                       ^
In file included from /usr/local/include/openssl/crypto.h:32:0,
[... 67 lines cut ...]
 static void callbackLocking(int mode, int n, const char*, int) {
             ^~~~~~~~~~~~~~~
cc1plus: warning: unrecognized command line option ‘-Wno-deprecated-register’
Makefile:1348: set di istruzioni per l'obiettivo "src/thrift/transport/TSSLSocket.lo" non riuscito
make[4]: *** [src/thrift/transport/TSSLSocket.lo] Errore 1
make[4]: uscita dalla directory "/root/Scaricati/thrift/lib/cpp"
Makefile:1656: set di istruzioni per l'obiettivo "all-recursive" non riuscito
make[3]: *** [all-recursive] Errore 1
make[3]: uscita dalla directory "/root/Scaricati/thrift/lib/cpp"
Makefile:574: set di istruzioni per l'obiettivo "all-recursive" non riuscito
make[2]: *** [all-recursive] Errore 1
make[2]: uscita dalla directory "/root/Scaricati/thrift/lib"
Makefile:653: set di istruzioni per l'obiettivo "all-recursive" non riuscito
make[1]: *** [all-recursive] Errore 1
make[1]: uscita dalla directory "/root/Scaricati/thrift"
Makefile:567: set di istruzioni per l'obiettivo "all" non riuscito
make: *** [all] Errore 2

Who can help me?

openssl 1.1
jre8
pybombs 

Where is the error?  pls help me

bbaranoff, 2018-04-04:
dyurnozz 
apt install libssl1.0-dev

RunningSnailYang, 2021-09-15:
Thanks, @bastienbaranoff ！