ptrkrysik/gr-gsm #163: grgsm_livemon  run error

xrh003, 2016-03-01:
Help me!
Traceback (most recent call last):
  File "./grgsm_livemon", line 28, in <module>
    import grgsm
  File "/usr/local/lib/python2.7/dist-packages/grgsm/__init__.py", line 45, in <module>
    from grgsm_swig import *
  File "/usr/local/lib/python2.7/dist-packages/grgsm/grgsm_swig.py", line 28, in <module>
    _grgsm_swig = swig_import_helper()
  File "/usr/local/lib/python2.7/dist-packages/grgsm/grgsm_swig.py", line 24, in swig_import_helper
    _mod = imp.load_module('_grgsm_swig', fp, pathname, description)
ImportError: libgnuradio-runtime-3.7.10git.so.0.0.0: cannot open shared object file: No such file or directory

ptrkrysik, 2016-03-01:
Try this and check again:

sudo ldconfig

xrh003, 2016-03-01:
After  sudo ldconfig, i running grgsm_livemon got htis:
linux; GNU C++ version 4.8.4; Boost_105400; UHD_003.010.git-0-2d68f228

gr-osmosdr v0.1.x-xxx-xunknown (0.1.5git) gnuradio v3.7.9.1-97-gc0381e0e
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy soapy redpitaya 
Using HackRF One with firmware git-7e81647 
Using Volk machine: avx_64_mmx_orc
Could not find port: constant_msg in:
constant_msg
system

Traceback (most recent call last):
  File "./grgsm_livemon", line 270, in <module>
    tb = grgsm_livemon(fc=options.fc, gain=options.gain, ppm=options.ppm, samp_rate=options.samp_rate, shiftoff=options.shiftoff, args=options.args)
  File "./grgsm_livemon", line 150, in __init__
    samp_rate_in=samp_rate,
  File "/usr/local/lib/python2.7/dist-packages/grgsm/gsm_input.py", line 48, in __init__
    samp_rate_in=samp_rate_in,
  File "/usr/local/lib/python2.7/dist-packages/grgsm/clock_offset_corrector.py", line 64, in __init__
    self.msg_connect(self, "ppm_in", self.gsm_controlled_const_source_f_0, "constant_msg")
  File "/usr/lib/python2.7/dist-packages/gnuradio/gr/hier_block2.py", line 59, in wrapped
    func(self, src.to_basic_block(), srcport, dst.to_basic_block(), dstport)
  File "/usr/lib/python2.7/dist-packages/gnuradio/gr/hier_block2.py", line 131, in msg_connect
    self.primitive_msg_connect(*args)
  File "/usr/lib/python2.7/dist-packages/gnuradio/gr/runtime_swig.py", line 3043, in primitive_msg_connect
    return _runtime_swig.hier_block2_sptr_primitive_msg_connect(self, *args)
RuntimeError: invalid msg port in connect() or disconnect()

xrh003, 2016-03-03:
Hi, ptrkrysik 
Thanks for your help. But i still can not fix this error. Is there some wrong with the GNURadio ? Can you give me some advice.

ptrkrysik, 2016-03-03:
This looks like reincarnation of an older bug with connections of msg
ports in GNU Radio.

xrh003, 2016-03-04:
How to fix this older bug?

ptrkrysik, 2016-03-04:
I supposed that GNU Radio introduced a incompatibility in versions above
3.7.9.x. Maybe they did it somewhere around 3.7.9.2.

I need help from your side. Otherwise I will have to setup exactly the
same GNU Radio that you have or look for the source of the problem in
GNU Radio's repository (it will take some time for sure).

Can you change the content of gr-gsm/python/misc_utils/hier_block.py to:

```cpp
from gnuradio import gr
from distutils.version import LooseVersion as version

#class created to solve incompatibility of reginstration of message inputs
#that was introduced in gnuradio 3.7.9

class hier_block(gr.hier_block2):
    def message_port_register_hier_in(self, port_id):
        super(hier_block, self).message_port_register_hier_out(port_id)

    def message_port_register_hier_out(self, port_id):
        super(hier_block, self).message_port_register_hier_in(port_id)

```

and then install gr-gsm again manually:
cd gr-gsm/build
make install

xrh003, 2016-03-04:
I did the test and got the same error log:

linux; GNU C++ version 4.8.4; Boost_105400; UHD_003.010.git-0-2d68f228

gr-osmosdr v0.1.x-xxx-xunknown (0.1.5git) gnuradio v3.7.9.1-97-gc0381e0e
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy soapy redpitaya 
Using HackRF One with firmware git-7e81647 
Using Volk machine: avx_64_mmx_orc
Could not find port: constant_msg in:
constant_msg
system

Traceback (most recent call last):
  File "/usr/local/bin/grgsm_livemon", line 270, in <module>
    tb = grgsm_livemon(fc=options.fc, gain=options.gain, ppm=options.ppm, samp_rate=options.samp_rate, shiftoff=options.shiftoff, args=options.args)
  File "/usr/local/bin/grgsm_livemon", line 150, in __init__
    samp_rate_in=samp_rate,
  File "/usr/local/lib/python2.7/dist-packages/grgsm/gsm_input.py", line 48, in __init__
    samp_rate_in=samp_rate_in,
  File "/usr/local/lib/python2.7/dist-packages/grgsm/clock_offset_corrector.py", line 64, in __init__
    self.msg_connect(self, "ppm_in", self.gsm_controlled_const_source_f_0, "constant_msg")
  File "/usr/lib/python2.7/dist-packages/gnuradio/gr/hier_block2.py", line 59, in wrapped
    func(self, src.to_basic_block(), srcport, dst.to_basic_block(), dstport)
  File "/usr/lib/python2.7/dist-packages/gnuradio/gr/hier_block2.py", line 131, in msg_connect
    self.primitive_msg_connect(*args)
  File "/usr/lib/python2.7/dist-packages/gnuradio/gr/runtime_swig.py", line 3043, in primitive_msg_connect
    return _runtime_swig.hier_block2_sptr_primitive_msg_connect(self, *args)
RuntimeError: invalid msg port in connect() or disconnect()

my change:

from gnuradio import gr
from distutils.version import LooseVersion as version

#class created to solve incompatibility of reginstration of message inputs
#that was introduced in gnuradio 3.7.9

#class hier_block(gr.hier_block2):
#    def message_port_register_hier_in(self, port_id):
#        if version(gr.version()) >= version('3.7.9'):
#            super(hier_block, self).message_port_register_hier_in(port_id)
#        else:
#            super(hier_block, self).message_port_register_hier_out(port_id)
#
#    def message_port_register_hier_out(self, port_id):
#        if version(gr.version()) >= version('3.7.9'):
#            super(hier_block, self).message_port_register_hier_out(port_id)
#        else:
#            super(hier_block, self).message_port_register_hier_in(port_id)

class hier_block(gr.hier_block2): 
    def message_port_register_hier_in(self, port_id): 
        super(hier_block, self).message_port_register_hier_out(port_id) 
    def message_port_register_hier_out(self, port_id): 
        super(hier_block, self).message_port_register_hier_in(port_id)

xrh003, 2016-03-04:
I did the test and got the same error log:

linux; GNU C++ version 4.8.4; Boost_105400; UHD_003.010.git-0-2d68f228

gr-osmosdr v0.1.x-xxx-xunknown (0.1.5git) gnuradio v3.7.9.1-97-gc0381e0e
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy soapy redpitaya
Using HackRF One with firmware git-7e81647
Using Volk machine: avx_64_mmx_orc
Could not find port: constant_msg in:
constant_msg
system

Traceback (most recent call last):
File "/usr/local/bin/grgsm_livemon", line 270, in
tb = grgsm_livemon(fc=options.fc, gain=options.gain, ppm=options.ppm, samp_rate=options.samp_rate, shiftoff=options.shiftoff, args=options.args)
File "/usr/local/bin/grgsm_livemon", line 150, in init
samp_rate_in=samp_rate,
File "/usr/local/lib/python2.7/dist-packages/grgsm/gsm_input.py", line 48, in init
samp_rate_in=samp_rate_in,
File "/usr/local/lib/python2.7/dist-packages/grgsm/clock_offset_corrector.py", line 64, in init
self.msg_connect(self, "ppm_in", self.gsm_controlled_const_source_f_0, "constant_msg")
File "/usr/lib/python2.7/dist-packages/gnuradio/gr/hier_block2.py", line 59, in wrapped
func(self, src.to_basic_block(), srcport, dst.to_basic_block(), dstport)
File "/usr/lib/python2.7/dist-packages/gnuradio/gr/hier_block2.py", line 131, in msg_connect
self.primitive_msg_connect(*args)
File "/usr/lib/python2.7/dist-packages/gnuradio/gr/runtime_swig.py", line 3043, in primitive_msg_connect
return _runtime_swig.hier_block2_sptr_primitive_msg_connect(self, *args)
RuntimeError: invalid msg port in connect() or disconnect()

my change:

from gnuradio import gr
from distutils.version import LooseVersion as version

#class created to solve incompatibility of reginstration of message inputs
#that was introduced in gnuradio 3.7.9

#class hier_block(gr.hier_block2):
#    def message_port_register_hier_in(self, port_id):
#        if version(gr.version()) >= version('3.7.9'):
#            super(hier_block, self).message_port_register_hier_in(port_id)
#        else:
#            super(hier_block, self).message_port_register_hier_out(port_id)
#
#    def message_port_register_hier_out(self, port_id):
#        if version(gr.version()) >= version('3.7.9'):
#            super(hier_block, self).message_port_register_hier_out(port_id)
#        else:
#            super(hier_block, self).message_port_register_hier_in(port_id)

class hier_block(gr.hier_block2): 
    def message_port_register_hier_in(self, port_id): 
        super(hier_block, self).message_port_register_hier_out(port_id) 
    def message_port_register_hier_out(self, port_id): 
        super(hier_block, self).message_port_register_hier_in(port_id)

xrh003, 2016-03-04:
I did the test and got the same error log:

linux; GNU C++ version 4.8.4; Boost_105400; UHD_003.010.git-0-2d68f228

gr-osmosdr v0.1.x-xxx-xunknown (0.1.5git) gnuradio v3.7.9.1-97-gc0381e0e
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy soapy redpitaya
Using HackRF One with firmware git-7e81647
Using Volk machine: avx_64_mmx_orc
Could not find port: constant_msg in:
constant_msg
system

Traceback (most recent call last):
File "/usr/local/bin/grgsm_livemon", line 270, in
tb = grgsm_livemon(fc=options.fc, gain=options.gain, ppm=options.ppm, samp_rate=options.samp_rate, shiftoff=options.shiftoff, args=options.args)
File "/usr/local/bin/grgsm_livemon", line 150, in init
samp_rate_in=samp_rate,
File "/usr/local/lib/python2.7/dist-packages/grgsm/gsm_input.py", line 48, in init
samp_rate_in=samp_rate_in,
File "/usr/local/lib/python2.7/dist-packages/grgsm/clock_offset_corrector.py", line 64, in init
self.msg_connect(self, "ppm_in", self.gsm_controlled_const_source_f_0, "constant_msg")
File "/usr/lib/python2.7/dist-packages/gnuradio/gr/hier_block2.py", line 59, in wrapped
func(self, src.to_basic_block(), srcport, dst.to_basic_block(), dstport)
File "/usr/lib/python2.7/dist-packages/gnuradio/gr/hier_block2.py", line 131, in msg_connect
self.primitive_msg_connect(*args)
File "/usr/lib/python2.7/dist-packages/gnuradio/gr/runtime_swig.py", line 3043, in primitive_msg_connect
return _runtime_swig.hier_block2_sptr_primitive_msg_connect(self, *args)
RuntimeError: invalid msg port in connect() or disconnect()

xrh003, 2016-03-05:
I reinstalled GNU Radio from the source code and run the grgsm_livemon, then got a new error log.
The GNU Radio version is GNU Radio Companion v3.7.9.1-97-gc0381e0e.

Traceback (most recent call last):
  File "/usr/local/bin/grgsm_livemon", line 29, in <module>
    import osmosdr
  File "/usr/local/lib/python2.7/dist-packages/osmosdr/__init__.py", line 26, in <module>
    from osmosdr_swig import *
  File "/usr/local/lib/python2.7/dist-packages/osmosdr/osmosdr_swig.py", line 28, in <module>
    _osmosdr_swig = swig_import_helper()
  File "/usr/local/lib/python2.7/dist-packages/osmosdr/osmosdr_swig.py", line 24, in swig_import_helper
    _mod = imp.load_module('_osmosdr_swig', fp, pathname, description)
ImportError: libgnuradio-fcdproplus.so.0: cannot open shared object file: No such file or directory

xrh003, 2016-03-07:
Good news.
My grgsm_livemon can run successfully.
I think that the reason of the running error is something wrong with the installation of the GNU Radio.
When i located the libgnuradio-fcdproplus and i found a .deb file and then i installed it.
After installation this lib file, the grgsm_livemon can run normally.

The lib file is : '/var/cache/apt/archives/libgnuradio-fcdproplus0_10.0.1.0-myriadrf3~trusty_amd64.deb'.

ptrkrysik, 2016-04-12:
I've reproduced the problem with "Could not find port: constant_msg in:" on my computer with GNU Radio v3.7.9.2-107-g0ac619e3, so I'm reopening the issue.

ptrkrysik, 2016-04-12:
This problem can happen when there is GNU Radio installed from repository and with use of pybombs on the same machine.

closed 2016-04-12 by ptrkrysik

giomar207, 2017-09-21:
Hi ptrkrysik,
since I had the same problem in installing the gr-gsm software, do you suggest to install gnuradio without Pybombs?
Thank you in advance
Giovanni