ptrkrysik/gr-gsm #164: installation error, ubuntu 14.04 amd64

ddl131, 2016-03-02:
$ uname -a
Linux 3.19.0-25-generic #26~14.04.1-Ubuntu SMP Fri Jul 24 21:16:20 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux

sudo pybombs install gr-gsm
...
[ 84%] Built target gnuradio-uhd
[ 84%] Built target tags_demo
[ 84%] Built target uhd_swig_swig_doc
[ 84%] Built target _uhd_swig_swig_tag
[ 84%] Built target uhd_swig_gr_uhd_swig_5e3ce
[ 84%] Building CXX object gr-uhd/swig/CMakeFiles/_uhd_swig.dir/uhd_swigPYTHON_wrap.cxx.o
/usr/local/src/gnuradio/build/gr-uhd/swig/uhd_swigPYTHON_wrap.cxx: In function ‘PyObject* _wrap_dboard_iface_set_gpio_debug(PyObject*, PyObject*, PyObject*)’:
/usr/local/src/gnuradio/build/gr-uhd/swig/uhd_swigPYTHON_wrap.cxx:27154:15: error: ‘class uhd::usrp::dboard_iface’ has no member named ‘set_gpio_debug’
       (arg1)->set_gpio_debug(arg2,arg3);
               ^
/usr/local/src/gnuradio/build/gr-uhd/swig/uhd_swigPYTHON_wrap.cxx: In function ‘PyObject* _wrap_dboard_iface_sptr_set_gpio_debug(PyObject*, PyObject*, PyObject*)’:
/usr/local/src/gnuradio/build/gr-uhd/swig/uhd_swigPYTHON_wrap.cxx:28517:16: error: ‘class uhd::usrp::dboard_iface’ has no member named ‘set_gpio_debug’
       (*arg1)->set_gpio_debug(arg2,arg3);
                ^
/usr/local/src/gnuradio/build/gr-uhd/swig/uhd_swigPYTHON_wrap.cxx: In function ‘void init_uhd_swig()’:
/usr/local/src/gnuradio/build/gr-uhd/swig/uhd_swigPYTHON_wrap.cxx:46466:91: error: ‘ATR_REG_IDLE’ is not a member of ‘uhd::usrp::dboard_iface’
   SWIG_Python_SetConstant(d, "dboard_iface_ATR_REG_IDLE",SWIG_From_int(static_cast< int >(uhd::usrp::dboard_iface::ATR_REG_IDLE)));
                                                                                           ^
/usr/local/src/gnuradio/build/gr-uhd/swig/uhd_swigPYTHON_wrap.cxx:46467:94: error: ‘ATR_REG_TX_ONLY’ is not a member of ‘uhd::usrp::dboard_iface’
   SWIG_Python_SetConstant(d, "dboard_iface_ATR_REG_TX_ONLY",SWIG_From_int(static_cast< int >(uhd::usrp::dboard_iface::ATR_REG_TX_ONLY)));
                                                                                              ^
/usr/local/src/gnuradio/build/gr-uhd/swig/uhd_swigPYTHON_wrap.cxx:46468:94: error: ‘ATR_REG_RX_ONLY’ is not a member of ‘uhd::usrp::dboard_iface’
   SWIG_Python_SetConstant(d, "dboard_iface_ATR_REG_RX_ONLY",SWIG_From_int(static_cast< int >(uhd::usrp::dboard_iface::ATR_REG_RX_ONLY)));
                                                                                              ^
/usr/local/src/gnuradio/build/gr-uhd/swig/uhd_swigPYTHON_wrap.cxx:46469:98: error: ‘ATR_REG_FULL_DUPLEX’ is not a member of ‘uhd::usrp::dboard_iface’
   SWIG_Python_SetConstant(d, "dboard_iface_ATR_REG_FULL_DUPLEX",SWIG_From_int(static_cast< int >(uhd::usrp::dboard_iface::ATR_REG_FULL_DUPLEX)));
                                                                                                  ^
make[2]: *** [gr-uhd/swig/CMakeFiles/_uhd_swig.dir/uhd_swigPYTHON_wrap.cxx.o] Ошибка 1
make[1]: *** [gr-uhd/swig/CMakeFiles/_uhd_swig.dir/all] Ошибка 2
make: *** [all] Ошибка 2
PyBombs.Packager.source - ERROR - Build failed. See output above for error messages.
PyBombs.Packager.source - ERROR - Problem occurred while building package gnuradio:
Build failed.
PyBombs.install - ERROR - Error installing package gnuradio. Aborting.

ddl131, 2016-03-02:
reinstalled everything, compiled ok, but, segmentation error at gnuradio-companion ./grgsm_livemon.grc

ptrkrysik, 2016-04-10:
I'm closing the issue as the install went ok. If you still have problem with segmentation error please send describe it in separate issue.

closed 2016-04-10 by ptrkrysik