ptrkrysik/gr-gsm #166: grgsm_scanner error

falcorocks, 2016-03-06:
Suddendly grgsm_scanner stopped working.
Everything else works so it's not the dongle...
This is what I'm getting

`linux; GNU C++ version 5.2.1 20151010; Boost_105800; UHD_003.010.git-0-2d68f228

Traceback (most recent call last):
  File "/usr/local/bin/grgsm_scanner", line 393, in <module>
    for info in sorted(found_list):
  File "/usr/local/bin/grgsm_scanner", line 293, in __cmp__
    return self.getKey().__cmp__(other.getKey())
AttributeError: 'NoneType' object has no attribute '__cmp__'
Reattached kernel driver`

otherwise i'm also getting this

`linux; GNU C++ version 5.2.1 20151010; Boost_105800; UHD_003.010.git-0-2d68f228

Traceback (most recent call last):
  File "/usr/local/bin/grgsm_scanner", line 394, in <module>
    print info
  File "/usr/local/bin/grgsm_scanner", line 296, in __repr__
    return "ARFCN: %4u, Freq: %6.1fM, CID: %5u, LAC: %5u, MCC: %3u, MNC: %3u, Pwr: %3i" % (self.arfcn, self.freq/1e6, self.cid, self.lac, self.mcc, self.mnc, self.power)
TypeError: %u format: a number is required, not NoneType
Reattached kernel driver

`

ptrkrysik, 2016-04-10:
It seems like some bug in the scanner. One of the parameters had value None. Probably ARFCN.
I'm marking it as bug.

romankh, 2016-12-05:
@falcorocks: does this bug still exist with current master ? think it may have been fixed by the changes on arfcn calculation in commit 385a606119ce43fd575c604717463f1453625c46

petterreinholdtsen, 2017-08-29:
The problem still exist in master.  It has been fixed in the packaging branch with commit aed9f5cc80e2c123eb0294dff21e853c1ffa376b .

petterreinholdtsen, 2017-09-02:
A fix for this is now included in master and was part of the 0.41.1 release.  Time to close this issue as fixed?

closed 2017-09-05 by falcorocks