ptrkrysik/gr-gsm #167: cfile hopping tchf

alex0500, 2016-03-09:
can any one share wideband test cfile with hopping tchf traffic?
thanks

ptrkrysik, 2016-04-10:
I have such file but it has 3GB. Are you able to provide some hosting for it (I'm not talking about file sharing site where file will be kept for short period).

alex0500, 2016-04-10:
Yes. I can give to you google drive access. Its ok?

Отправлено с iPhone

ptrkrysik, 2016-04-10:
I want to publish the file with guide how to decode it and I don't have
hosting for it.
Google drive won't be good because it will be available only for you (I
can agree for to send the file just to you if you are working on
development of new functionality to gr-gsm and you will share this work).

W dniu 10.04.2016 o 12:19, alex0500 pisze:

alex0500, 2016-04-10:
I see. 
Ok i will send you hosting details. 

Отправлено с iPhone

alex0500, 2016-04-10:
you can use youfilelink.com. they have no time limit
or create google drive and give access to public.

ptrkrysik, 2016-06-09:
I will send the file with description/example after I resolve other issues that are more pressing currently.

closed 2016-06-09 by ptrkrysik