ptrkrysik/gr-gsm #169: Clean Debian - gnuradio missed modules

aloz, 2016-03-11:
Hello!

My environment is:

3.16.0.4-amd64 GNU/Linux
Distributor ID: Debian
Release: 8.3
Codename: jessie

Under Oracle Virtualbox, just after inslalling the system from an official ISO, applying the latest updates.

I did everything described into https://github.com/ptrkrysik/gr-gsm/wiki/Installation

pybombs did everything without errors, but starting gnuradio-companion (during the load I got the message into the terminal console:
Gtk-Message: Failed to load module "canberra-gtk-module"
) and after loading of grgsm_livemon.grc I've got an errors, and the missing blocks into the scheme. Please see the window content. How it could be fixed? Is in an issue of https://gnuradio.org/redmine/projects/gnuradio/wiki/FAQ#How-do-I-install-GNU-Radio ? Thank you!

<<< Welcome to GNU Radio Companion 3.7.5 >>>

Preferences file: /home/local/.grc
Block paths:
    /usr/share/gnuradio/grc/blocks
    /home/local/.grc_gnuradio

Showing: ""

Loading: "/home/local/src/gr-gsm/apps/grgsm_livemon.grc"
>>> Error: Block key "gsm_bcch_ccch_demapper" not found in Platform - grc(GNU Radio Companion)
>>> Error: Block key "gsm_clock_offset_control" not found in Platform - grc(GNU Radio Companion)
>>> Error: Block key "gsm_control_channels_decoder" not found in Platform - grc(GNU Radio Companion)
>>> Error: Block key "gsm_control_channels_decoder" not found in Platform - grc(GNU Radio Companion)
>>> Error: Block key "gsm_decryption" not found in Platform - grc(GNU Radio Companion)
>>> Error: Block key "gsm_input" not found in Platform - grc(GNU Radio Companion)
>>> Error: Block key "gsm_message_printer" not found in Platform - grc(GNU Radio Companion)
>>> Error: Block key "gsm_receiver" not found in Platform - grc(GNU Radio Companion)
>>> Error: Block key "gsm_sdcch8_demapper" not found in Platform - grc(GNU Radio Companion)
>>> Error: Block key "rtlsdr_source" not found in Platform - grc(GNU Radio Companion)
>>> Done

Showing: "/home/local/src/gr-gsm/apps/grgsm_livemon.grc"

MatejKovacic, 2016-03-30:
Hi, I have similar problems. I have Raspbian Jessine and RPi 3.

This is the error when I try to load grgsm_livemon in GRC:

```
<<< Welcome to GNU Radio Companion 3.7.5 >>>

Preferences file: /home/pi/.grc
Block paths:
	/usr/share/gnuradio/grc/blocks
	/home/pi/.grc_gnuradio

Showing: ""

Loading: "/usr/local/bin/grgsm_livemon"
Error: /usr/local/bin/grgsm_livemon:1:1:FATAL:PARSER:ERR_DOCUMENT_EMPTY: Start tag expected, '<' not found
>>> Failure

Loading: "/home/pi/gr-gsm/apps/grgsm_livemon.grc"
>>> Error: Block key "gsm_bcch_ccch_demapper" not found in Platform - grc(GNU Radio Companion)
>>> Error: Block key "gsm_clock_offset_control" not found in Platform - grc(GNU Radio Companion)
>>> Error: Block key "gsm_control_channels_decoder" not found in Platform - grc(GNU Radio Companion)
>>> Error: Block key "gsm_control_channels_decoder" not found in Platform - grc(GNU Radio Companion)
>>> Error: Block key "gsm_decryption" not found in Platform - grc(GNU Radio Companion)
>>> Error: Block key "gsm_input" not found in Platform - grc(GNU Radio Companion)
>>> Error: Block key "gsm_message_printer" not found in Platform - grc(GNU Radio Companion)
>>> Error: Block key "gsm_receiver" not found in Platform - grc(GNU Radio Companion)
>>> Error: Block key "gsm_sdcch8_demapper" not found in Platform - grc(GNU Radio Companion)
>>> Done

Showing: "/home/pi/gr-gsm/apps/grgsm_livemon.grc"

```

MatejKovacic, 2016-03-31:
Heh, stupid.

I forgot to create the ~/.gnuradio/config.conf config file (so gnuradio-companion can find custom blocks of gr-gsm).

Add to it this:

```
[grc]
local_blocks_path=/usr/local/share/gnuradio/grc/blocks
```
...and save it. That is it.

ptrkrysik, 2016-04-10:
@aloz did you try the advice of @MatejKovacic ?

ptrkrysik, 2016-04-10:
Also... lets continue the discussion on the mailing list. You can reverence the issue #169 there.

closed 2016-04-10 by ptrkrysik

aloz, 2016-04-17:
Hi Piotr!

Yes, it helped, thank you!

jbarea, 2016-05-06:
Hello, i don't know exactly what the path does in the case the grgsm blocks are stored something else. I mean, i downloaded the grgsm_master zip and decompress it on a directory in my machine. Now, do i need to copy the blocks on the gnuradio installation or simply i need to put the path to the grgsm blocks location?I'm trying to get it running on win and on mac os x, and while gnuradio is running ok, i cannot run the grgsm_livemonitor because it complains about missing blocks.
Thanks in advance!!!

rojavatechops, 2017-04-09:
I had the same problem. It was fixed  moving /.gnuradio to ~/.gnuradio.

RBXSystems, 2017-04-23:
I had the same problem on Raspiberry Pi running Ubuntu-Mate. I solved this by adding the following to my ~/.gnuradio/config.conf config file:
[grc]
local_blocks_path=/home/richard/gr-gsm/grc
This is the folder that contains the gsm_block_tree.xml file (plus folders)

huzaifa2457, 2018-03-07:
Hey. I cannot find the config file to edit and if I enter ~/.gnuradio/config.conf in terminal it says access denied .What should I do?

velichkov, 2018-03-07:
Hi @huzaifa2457,

> if I enter ~/.gnuradio/config.conf in terminal it says access denied 

You cannot start/execute config files. Try to create the directory first
```
mkdir ~/.gnuradio/
vim ~/.gnuradio/config.conf
```
and then open it with your favorite text editor (`vim` in the example above)