ptrkrysik/gr-gsm #171: I can not install gr-gsm

warpiper, 2016-03-15:
root@COMPUTER:/home/xxxxx# pybombs install gr-gsm
PyBombs.Packager.pip - ERROR - Could not run pip list. Hm.
PyBombs.Packager.pip - ERROR - Command '['pip', 'list']' returned non-zero exit status 2

That it mean -  "returned non-zero exit status 2"   ???

krysiek, 2016-03-15:
I have the same on Ubuntu 12. What I have found till now, is only this , bot no idea what to do with this, In my case pip list does not return zero. Check this out
http://osdir.com/ml/discuss-gnuradio-gnu/2016-02/msg00054.html

warpiper, 2016-03-15:
I have Ubuntu 15. What i must to see on this page? Looking like japan ieroglific...   :)

warpiper, 2016-03-15:
Is it possible, to download gr-gsm.tgz manually somethere, and to install it himself, manually?

krysiek, 2016-03-16:
what about this? https://github.com/ptrkrysik/gr-gsm/wiki/Manual-compilation-and-installation

warpiper, 2016-03-16:
ubuntu@computer:~/gr-gsm/build$ cmake ..
-- The CXX compiler identification is GNU 4.9.2
-- The C compiler identification is GNU 4.9.2
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Build type not specified: defaulting to release.
-- Boost version: 1.55.0
-- Found the following Boost libraries:
--   filesystem
--   system
--   thread
CMake Warning at CMakeLists.txt:92 (find_package):
  By not providing "FindGnuradio.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "Gnuradio",
  but CMake did not find one.

  Could not find a package configuration file provided by "Gnuradio" with any
  of the following names:

    GnuradioConfig.cmake
    gnuradio-config.cmake

  Add the installation prefix of "Gnuradio" to CMAKE_PREFIX_PATH or set
  "Gnuradio_DIR" to a directory containing one of the above files.  If
  "Gnuradio" provides a separate development package or SDK, be sure it has
  been installed.

-- Found PkgConfig: /usr/bin/pkg-config (found version "0.28") 
-- checking for module 'volk'
--   package 'volk' not found
-- Could NOT find VOLK (missing:  VOLK_LIBRARIES VOLK_INCLUDE_DIRS) 
-- checking for module 'cppunit'
--   found cppunit, version 1.13.2
-- Found CPPUNIT: /usr/lib/i386-linux-gnu/libcppunit.so;dl  
-- Found Doxygen: /usr/bin/doxygen (found version "1.8.9.1") 
-- checking for module 'libosmocore'
--   package 'libosmocore' not found
-- Could NOT find libosmocore (missing:  LIBOSMOCORE_LIBRARY LIBOSMOCORE_INCLUDE_DIR) 
CMake Error at CMakeLists.txt:99 (message):
  GnuRadio Runtime required to compile gr-gsm

-- Configuring incomplete, errors occurred!
See also "/home/ubuntu/gr-gsm/build/CMakeFiles/CMakeOutput.log".
ubuntu@computer:~/gr-gsm/build$

warpiper, 2016-03-16:
git clone https://github.com/ptrkrysik/gr-gsm.git

This depository is not ready for cloning and compilyng :(

warpiper, 2016-03-16:
Yes! I made it! :)

KALI 2.0 libosmocore
apt-get install libosmocore libosmocore-dev
                                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^  - this is not working

But   
KALI 2.0 libosmocore  
                 ^^^^^^^^^^^^^   - Is good link, I manually compiled libosmocore CC  And all is OK!  gr-gsm is compyling now also!

krysiek, 2016-03-16:
have you followed the manual installation tutorial from the link above?

warpiper, 2016-03-16:
WTF

At first, all my trying to install gr-gsm on UBUNTU 15 not sucsess, gr-gsm was istalled, but Scarry Ugly Curve and not working.
I deleted it and install UBUNTU 14.4

And Your package begin to install!
But...
only till this:

root@COMPUTER:/home/ubuntu# sudo pybombs install gr-gsm

|
\- gr-osmosdr
   |
   +- airspy
   |
   +- hackrf
   |
   \- bladeRF
|
\- gr-gsm
   |
   \- gr-osmosdr
      |
      +- airspy
      |
      +- hackrf
      |
      \- bladeRF
PyBombs.install - INFO - Installing package: bladeRF
PyBombs.Packager.source - ERROR - Problem occurred while building package bladeRF:
There should be a source dir in /home/ubuntu/src/bladeRF, but there isn't.
PyBombs.install - ERROR - Error installing package bladeRF. Aborting.
root@COMPUTER:/home/ubuntu# 

WTF!!!  Where must be this bladeRF???

tiger762, 2016-04-05:
The bladeRF is a type of SDR hardware. I am not familiar with it. I use the Realtek dongle and the Ettus B200 (UHD). Somehow you have PyBOMBS believing that you want it to build with bladeRF support.

What hardware do you intend to use?

ptrkrysik, 2016-04-10:
There are many configurations that gr-gsm doesn't install on without problems but it is not always issue with code of gr-gsm. Lets have this discussion on the mailing list. You can reference this issue (#171) there so we don't start the discussion from 0.

closed 2016-04-10 by ptrkrysik