ptrkrysik/gr-gsm #172: what to do next.?

newlearnergrgsm, 2016-03-16:
hi all.
ı am using kali linux 2.0 (64 bit) with vbox , ı have downloaded ''gr-gsm'' successfull and my all files are in ''/usr/local/src/'' ı do open ''grgsm_livemon.grc'' and works great. I open wireshark with ''sudo wireshark -k -f udp -Y gsmtap -i lo'' everything is okey ı see codecs and checking cellphone id etc...
everything is okey but what todo next. 
1. how to decode and listen !
thanks for help..specially who works on gr-gsm..

newlearnergrgsm, 2016-03-21:
in ''Airprobe'' ı do listen my own voice ''speech.au.gsm''  but with out any knowledge , it really funny for me because ''gr-gsm'' easy then airprobe but ı cant decode how funny me .. Problem is
''gsm_decode -c vf_call6_a725_d174_g5_Kc1EF00BAB3BAC7002.cfile -s $((100000000/174)) -a 725 -m BCCH -t 0'' how can ı make my own cfile from .bin file 
thanks for your help..

ptrkrysik, 2016-03-21:
Hi, we use issue tracker for tracking bugs and enhancements in gr-gsm.
For general discussions use mailing list.

Regarding the question - did you try to decode voice with:
grgsm_decode -c vf_call6_a725_d174_g5_Kc1EF00BAB3BAC7002.cfile -s
$((100000000/174)) -a 725 -m TCHF -t 5 -e 1 -k
0x1E,0xF0,0x0B,0xAB,0x3B,0xAC,0x70,0x02 -d FR -o /tmp/speech.au.gsm

Regarding capturing own test data - I'm thinking about writing a guide
but this will require some time.

I'm closing the issue.

closed 2016-03-21 by ptrkrysik