ptrkrysik/gr-gsm #174: bash parameters grgsm_capture not working correctly

Eauwzeauw, 2016-03-17:
Hi,

I have multiple receivers (all of type Realtek RTL2838UHIDIR with Rafael Micro R820T tuner) which i would like to use at the same time. You recently added functionality to do this (commit 7185b6664fc86e5ad75d0e1da9b2b0a5eebd20ae, thanks!) though this doesn't seem to work flawlessly with grgsm_capture.

I can for example run grgsm_scanner and grgsm_capture at the same time using the device serial numbers as argument. If i want to run two captures (of different frequencies) at the same time i get an error because the wrong device is selected.

I have two devices connected, one with serial number 00000001 and one with 00000006

Output of terminal:
Running first capture with first device:

> eauwzeauw@ubuntu:~$ grgsm_capture.py -c capture01.cfile -f 937200000 -T 30 --args="rtl=00000001"
> linux; GNU C++ version 5.2.1 20151010; Boost_105800; UHD_003.010.git-0-2d68f228
> 
> gr-osmosdr v0.1.4-67-gac15e789 (0.1.5git) gnuradio v3.7.9.1-95-g1683e58d
> built-in source types: file osmosdr fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy redpitaya 
> Using device #0 Realtek RTL2838UHIDIR SN: 00000001
> Detached kernel driver
> Found Rafael Micro R820T tuner
> [R82XX] PLL not locked!
> Exact sample rate is: 2000000.052982 Hz
> [R82XX] PLL not locked!
> Using Volk machine: avx_64_mmx
> OOOOOOOOOOOOOReattached kernel driver
> eauwzeauw@ubuntu:~$ 

(Some seconds later, but during other capture) Running second capture using other device:

> eauwzeauw@ubuntu:~grgsm_capture.py -c capture02.cfile -f 926800000 -T 30 --args="rtl=00000006"
> linux; GNU C++ version 5.2.1 20151010; Boost_105800; UHD_003.010.git-0-2d68f228
> 
> gr-osmosdr v0.1.4-67-gac15e789 (0.1.5git) gnuradio v3.7.9.1-95-g1683e58d
> built-in source types: file osmosdr fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy redpitaya 
> Using device #0 Realtek RTL2838UHIDIR SN: 00000001
> usb_claim_interface error -6
> 
> FATAL: Failed to open rtlsdr device.
> 
> Trying to fill up 1 missing channel(s) with null source(s).
> This is being done to prevent the application from crashing
> due to gnuradio bug #528.
> 
> Using Volk machine: avx_64_mmx

As you can see, the wrong device is used during the second capture, which obviously fails as it is already capturing. Also, grgsm_scanner -h, grgsm_livemon-h and grgsm_decode -h all work, but grgsm_capture -h is failing, which makes me think that something is wrong with the option interpreter of grgsm_capture. 

Running Ubuntu 15.10. And also; now that i have the chance: Thank you for making this! A great part of our project is possible because of your work ;)

Eauwzeauw, 2016-03-17:
Looking at it a bit longer; it seems that your forgot to change two lines in the commit.
In grgsm_capture;
line 64: change last """ to args ,
Line 261: add , 'args=options.args' in argument list

ptrkrysik, 2016-03-18:
@Eauwzeauw, thank you for this information. I applied the fix in the
development branch.

closed 2016-04-10 by ptrkrysik