ptrkrysik/gr-gsm #176: cannot import name Range

tiger762, 2016-04-04:
Hello,

Apologies for bringing up a common issue but from all I have read I cannot figure out what specific steps are taken to fix it.

[root@neutron apps]# ./grgsm_livemon
Traceback (most recent call last):
  File "./grgsm_livemon", line 25, in <module>
    from gnuradio.qtgui import Range, RangeWidget
ImportError: cannot import name Range

Versions:
[root@neutron apps]# rpm -qa | grep gnuradio
gnuradio-3.7.5.1-2.fc21.x86_64
gnuradio-devel-3.7.5.1-2.fc21.x86_64
[root@neutron apps]# rpm -qa | grep osmo
libosmo-dsp-0.3-1.fc21.x86_64
gr-osmosdr-0.1.3-1.20141023git42c66fdd.fc21.x86_64
[root@neutron apps]# rpm -qa | grep rtl
rtl-sdr-0.5.3-3.fc21.x86_64

I am well-versed in the use of the RTL2832, and regularly use applications like dump1090, rtl_flex.py, rtl_acars, etc. I am quite familiar with how GSM 2G TDMA works. Used to use gsm_receiver. 

Any help would be appreciated!
-Keith

tiger762, 2016-04-04:
Additional information:

[root@neutron apps]# echo $PYTHONPATH
/usr/local/lib64/python2.7/site-packages/
[root@neutron apps]# uname -a
Linux neutron 4.1.13-100.fc21.x86_64 #1 SMP Tue Nov 10 13:13:20 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux

ptrkrysik, 2016-04-04:
W dniu 04.04.2016 o 17:11, tiger762 pisze:
> Additional information:
> 
> [root@neutron apps]# echo $PYTHONPATH
> /usr/local/lib64/python2.7/site-packages/
> [root@neutron apps]# uname -a
> Linux neutron 4.1.13-100.fc21.x86_64 #1
> <https://github.com/ptrkrysik/gr-gsm/issues/1> SMP Tue Nov 10 13:13:20
> UTC 2015 x86_64 x86_64 x86_64 GNU/Linux
> 
>

Hi,

Please try to open gr-gsm/apps/grgsm_livemon.grc in gnuradio-companion.
Generate python file with f5 and run it with f6. If it works without the
error mentioned by you - please send me generated python file (its name
will be grgsm_livemon.py)

If it doesn't work please report it here.

tiger762, 2016-04-04:
Hi Piotr, thank you for such a fast response. This might be a dumb question, but where do I find that GRC file? I have looked for it under /root/gr-gsm/ and cannot find any GNURADIO companion files :(

tiger762, 2016-04-04:
OK, found it :)

When I try to run it within companion, I get:

<<< Welcome to GNU Radio Companion 3.7.5.1 >>>

Preferences file: /root/.grc
Block paths:
	/usr/local/share/gnuradio/grc/blocks
	/usr/share/gnuradio/grc/blocks
	/root/.grc_gnuradio

Loading: "/root/sdr/receivers/flex/rtl_fft.grc"
>>> Done

Showing: "/root/sdr/receivers/flex/rtl_fft.grc"

Loading: "/root/gr-gsm/apps/grgsm_livemon.grc"
>>> Done

Showing: "/root/gr-gsm/apps/grgsm_livemon.grc"

Generating: "/root/gr-gsm/apps/grgsm_livemon.py"

Generating: "/root/gr-gsm/apps/grgsm_livemon.py"

Executing: "/root/gr-gsm/apps/grgsm_livemon.py"

Traceback (most recent call last):
  File "/root/gr-gsm/apps/grgsm_livemon.py", line 21, in <module>
    import grgsm
  File "/usr/local/lib64/python2.7/site-packages/grgsm/__init__.py", line 45, in <module>
    from grgsm_swig import *
ImportError: No module named grgsm_swig

>>> Done

tiger762, 2016-04-04:
OK, I saw someone else had an issue with missing grgsm_swig. His resolution was to create the grgsm subdirectory under dist-packages. I copied grgsm_swig.i to the grgsm subdirectory:

cp   grgsm_swig.i    /usr/local/lib64/python2.7/site-packages/grgsm

Restarted grc but made no difference

tiger762, 2016-04-04:
OK, please disregard my SWIG issues. I didn't have swig installed. It compiled/built anyway, so that's why I thought my environment was good. I still have the "cannot import Range" but after I ran it in companion and generated the py file, I can run the py from the command line.

All is good. This is a wonderful software package you have created. Thank you!

ptrkrysik, 2016-04-04:
W dniu 04.04.2016 o 20:55, tiger762 pisze:
> OK, please disregard my SWIG issues. I didn't have swig installed. It
> compiled/built anyway, so that's why I thought my environment was good.
> I still have the "cannot import Range" but after I ran [...]
Thank you for the information.
Yesterday I also had similar situation when I tried to install gr-gsm in
Ubuntu docker image. There was missing grgsm_swig but I haven't found
why it compiled anyway. I will have to add check for SWIG.

ptrkrysik, 2016-04-10:
I've added check for SWIG in the cmake file.

closed 2016-04-10 by ptrkrysik

vallettea, 2017-07-31:
does anyone has this file generated by gnuradio companion. I'm running headless and I can't use the  interface ?

vallettea, 2017-07-31:
found it here https://groups.google.com/forum/#!topic/gr-gsm/_BIaHdb3UoM

vallettea, 2017-08-01:
i followed @ptrkrysik indications about running in gnuradio compagnion but i get:

```
gr-osmosdr 0.1.3 (0.1.3) gnuradio 3.7.5
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
Exact sample rate is: 2000000.052982 Hz
Using Volk machine: neon_hardfp_orc
Traceback (most recent call last):
  File /root/gr-gsm/apps/grgsm_livemon.py, line 335, in <module>
    tb = grgsm_livemon(args=options.args, fc=options.fc, gain=options.gain, osr=options.osr, ppm=options
.ppm, samp_rate=options.samp_rate, shiftoff=options.shiftoff)
  File /root/gr-gsm/apps/grgsm_livemon.py, line 177, in __init__
    timeslot_nr=1,
  File /usr/local/lib/python2.7/dist-packages/grgsm/gsm_sdcch8_demapper.py, line 39, in __init__
    self.msg_connect((self.gsm_universal_ctrl_chans_demapper_0, 'bursts'), (self, 'bursts'))
TypeError: msg_connect() takes exactly 5 arguments (3 given)
```

velichkov, 2017-08-01:
Hi @vallettea,

> I'm running headless and I can't use the interface ?

You could regenerate it from the command line using `grcc`
```
$grcc  -d $(pwd)  grgsm_livemon.grc
```

ptrkrysik, 2017-08-31:
W dniu 04.04.2016 o 20:55, tiger762 pisze:
> OK, please disregard my SWIG issues. I didn't have swig installed. It
> compiled/built anyway, so that's why I thought my environment was good.
> I still have the "cannot import Range" but after I ran [...]
Hi,

I don't think SWIG problem is related to the problem with importing
Range. This is probably related to your GNU Radio installation.

But thanks for report about the SWIG. Cmake in gr-gsm probably should
inform about missing SWIG and drop an error.