ptrkrysik/gr-gsm #177: Decode of fast and slow associated channels? (enhancement request)

tiger762, 2016-04-06:
Hi Piotr, the thought occurred to me today that after the BTS instructs the MS to tune to a dedicated TCH (half or full), there are fast and slow associated control channels that share the traffic channel. As long as we are before the MSC instructing the handset to set ciphering, there may be some messaging worth decoding.

As you're aware, when a handset requests service from the network, it will send a Service Request on the SDCCH (uplink) with the BTS hopefully giving a Service Accept (downlink). After call setup, the BTS will send a Assignment Command to the handset with the channel description (ARFCN and timeslot) of where the handset needs to tune itself to. The handset then replies with a Assignment Complete on the FACCH in that new timeslot.

I think if your livemon could tune into other timeslots on the primary ARFCN, it *might* get lucky and pick up some unencrypted call setup messages.

Curious what your thoughts are on this? Thanks!

ptrkrysik, 2016-04-06:
I think that ability to configure what combination of channels are in
timeslots as a program parameter (or inside of gui) should improve a lot
in this regards.

Parsing of Assignment Commands and then feeding it back as a new
configuration for the receiver is harder and I think it would be the
next step after the application can have this information set manually
by the user. Automatic configuration could be especially useful for
setting up hopping channels reception.

ptrkrysik, 2016-06-09:
Hi @tiger762, I don't have time to work on such enhancement currently. I'm closing the issue. If you want to work on it I can reopen it.

closed 2016-06-09 by ptrkrysik