ptrkrysik/gr-gsm #179: Compilation error on Fedora 22 64-bit --> grgsm_swigPYTHON_wrap.cxx.o

tiger762, 2016-04-06:
I think this may be a GCC compiler version issue?

[ 61%] Built target grgsm
[ 64%] Built target _grgsm_swig_doc_tag
[ 67%] Built target grgsm_swig_swig_doc
[ 70%] Built target _grgsm_swig_swig_tag
[ 71%] Building CXX object swig/CMakeFiles/_grgsm_swig.dir/grgsm_swigPYTHON_wrap.cxx.o
/root/gr-gsm/build/swig/grgsm_swigPYTHON_wrap.cxx: In function ‘PyObject* _wrap_receiver_sptr_pc_throughput_avg(PyObject*, PyObject*)’:
/root/gr-gsm/build/swig/grgsm_swigPYTHON_wrap.cxx:7965:32: error: ‘class gr::gsm::receiver’ has no member named ‘pc_throughput_avg’
       result = (float)(*arg1)->pc_throughput_avg();
                                ^
/root/gr-gsm/build/swig/grgsm_swigPYTHON_wrap.cxx: In function ‘PyObject* _wrap_receiver_sptr_set_processor_affinity(PyObject*, PyObject*, PyObject*)’:
/root/gr-gsm/build/swig/grgsm_swigPYTHON_wrap.cxx:8012:110: error: no matching function for call to ‘gr::gsm::receiver::set_processor_affinity(const std::vector<long unsigned int>&)’
       (*arg1)->set_processor_affinity((std::vector< std::size_t,std::allocator< std::size_t > > const &)*arg2);
                                                                                                              ^
In file included from /root/gr-gsm/build/swig/grgsm_swigPYTHON_wrap.cxx:4349:0:
/usr/local/include/gnuradio/block.h:537:10: note: candidate: virtual void gr::block::set_processor_affinity(const std::vector<int>&)
     void set_processor_affinity(const std::vector<int> &mask);
          ^
/usr/local/include/gnuradio/block.h:537:10: note:   no known conversion for argument 1 from ‘const std::vector<long unsigned int>’ to ‘const std::vector<int>&’
/root/gr-gsm/build/swig/grgsm_swigPYTHON_wrap.cxx: In function ‘PyObject* _wrap_receiver_sptr_processor_affinity(PyObject*, PyObject*)’:
/root/gr-gsm/build/swig/grgsm_swigPYTHON_wrap.cxx:8077:14: error: no match for ‘operator=’ (operand types are ‘SwigValueWrapper<std::vector<long unsigned int> >’ and ‘std::vector<int>’)
       result = (*arg1)->processor_affinity();
              ^
/root/gr-gsm/build/swig/grgsm_swigPYTHON_wrap.cxx:24:21: note: candidate: SwigValueWrapper<T>& SwigValueWrapper<T>::operator=(const SwigValueWrapper<T>&) [with T = std::vector<long unsigned int>]
   SwigValueWrapper& operator=(const SwigValueWrapper<T>& rhs);
                     ^
/root/gr-gsm/build/swig/grgsm_swigPYTHON_wrap.cxx:24:21: note:   no known conversion for argument 1 from ‘std::vector<int>’ to ‘const SwigValueWrapper<std::vector<long unsigned int> >&’
/root/gr-gsm/build/swig/grgsm_swigPYTHON_wrap.cxx:28:21: note: candidate: SwigValueWrapper<T>& SwigValueWrapper<T>::operator=(const T&) [with T = std::vector<long unsigned int>]
   SwigValueWrapper& operator=(const T& t) { SwigMovePointer tmp(new T(t)); pointer = tmp; return *this; }
                     ^
/root/gr-gsm/build/swig/grgsm_swigPYTHON_wrap.cxx:28:21: note:   no known conversion for argument 1 from ‘std::vector<int>’ to ‘const std::vector<long unsigned int>&’
/root/gr-gsm/build/swig/grgsm_swigPYTHON_wrap.cxx: In function ‘PyObject* _wrap_clock_offset_control_sptr_pc_throughput_avg(PyObject*, PyObject*)’:
/root/gr-gsm/build/swig/grgsm_swigPYTHON_wrap.cxx:11223:32: error: ‘class gr::gsm::clock_offset_control’ has no member named ‘pc_throughput_avg’
       result = (float)(*arg1)->pc_throughput_avg();

tiger762, 2016-04-06:
Linux tampa 4.4.6-200.fc22.x86_64 #1 SMP Wed Mar 16 22:13:40 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux
gcc version 5.3.1 20151207 (Red Hat 5.3.1-2) (GCC) 
swig-3.0.7-5.fc22.x86_64 reverted to swig-3.0.7-4.fc21.x86_64, same results

ptrkrysik, 2016-04-10:
I think that your advice in #180 solves this issue. In case it doesn't I will reopen it.

closed 2016-04-10 by ptrkrysik