ptrkrysik/gr-gsm #181: capture wideband in gsm

hydraOghost, 2016-04-17:
i want decode GSM frequency hopping when i sniffing BCCH in type inf. 1 there is two ARFCN so i have freq hopping with MAIO 0 and HSF 2 so :
can u pls tell me how i can capture wideband in GSM using RTL2832U to put it in input for channelize
   tnxs for any help

ptrkrysik, 2016-04-17:
W dniu 17.04.2016 o 18:08, hydraOghost pisze:
> i want decode GSM frequency hopping when i sniffing BCCH in type inf. 1
> there is two ARFCN so i have freq hopping with MAIO 0 and HSF 2 so :
> can u pls tell me how i can capture wideband in GSM using RTL2832U to
> put it in input for channelize
> tnxs for any help
> 
Hi,

Which ARFCNs are used for the hopping?

It's a bit non-typical that hopping is used for BCCH channel. Are you
sure this is the case?

-- 
Best Regards,
Piotr Krysik

hydraOghost, 2016-04-17:
[image]
[image]

ptrkrysik, 2016-04-18:
W dniu 17.04.2016 o 20:05, hydraOghost pisze:
> screenshot from 2016-04-17 14-03-26
> <https://cloud.githubusercontent.com/assets/18515922/14588993/6a330aa8-04a5-11e6-83ef-2b5e2e6f93d3.png>
> screenshot from 2016-04-17 14-03-45
> <https://cloud.githubusercontent.com/assets/18515922/14588994/6a36727e-04a5-11e6-90fc-8e0fb594b1c7.png>
> 

Ok,

So it is rather typical setup: you have your BCCH channel in timeslot 0
on a non-hopping channel and SDCCH channel in timeslot 1 of a hopping
channel.

What hardware do you have? Separation of these channels is 2.6MHz - it
is a bit over the specification of reliable bandwidth of RTL-SDR but you
can try. Record the signal at carrier frequency 928.70MHz and with
bandwidth of 3MHz. Do you know how to do it or need help?

-- 
Best Regards,
Piotr Krysik

ptrkrysik, 2016-04-18:
also @hydraOghost - this is topic for the mailing-list, not a bug/features tracker.
I'm closing the issue here. We can continue it on the mailing-list.

closed 2016-04-18 by ptrkrysik

hydraOghost, 2016-04-18:
good idea but thats not work with me i scan at 928.7MHZ and then used channelize at 986 999 
when i try to decode  BCCH out.986.cfile no output at wireshark

do u have any suggestions

ptrkrysik, 2016-04-18:
W dniu 18.04.2016 o 19:35, hydraOghost pisze:
> good idea but thats not work with me i scan at 928.7MHZ and then used
> channelize at 986 999
> when i try to decode BCCH out.986.cfile no output at wireshark
> 
> do u have any suggestions
> 

Can you send tell me how you captured and channelized the data?

hydraOghost, 2016-04-19:
https://www.youtube.com/watch?v=SzbyGobhGlk&feature=youtu.be

hydraOghost, 2016-04-19:
i really thank for ur help

Varbaek, 2016-07-20:
> So it is rather typical setup: you have your BCCH channel in timeslot 0
> on a non-hopping channel and SDCCH channel in timeslot 1 of a hopping
> channel.
>
> What hardware do you have? Separation of these [...]

I've been looking for this information everywhere on the web, thanks xD 
@hydraOghost: Try recording at 927.4MHz and channelize that and another channel. If you get output from 927.4MHz and not the other channel, then the program is working, but it also tells you that your RTL-SDR is unable to capture a "wide enough" spectrum/band/whatever it is called xD (RTL-SDR can only capture between 2.4-3.2MHz depending on the model/brand/components.)

ptrkrysik, 2016-07-20:
W dniu 20.07.2016 o 16:55, Hans-Michael Varbaek pisze:
> So it is rather typical setup: you have your BCCH channel in timeslot 0
>     on a non-hopping channel and SDCCH channel in timeslot 1 of a hopping
>     channel.
>
>     What hardware do you have? [...]

If you want to receive channels separated beyond one RTL-SDR receiver's
bandwidth you might want to try Multi-RTL:
https://ptrkrysik.github.io/
https://github.com/ptrkrysik/multi-rtl/

I used it in order to receive both uplink and downlin (which are
separated by 45MHz for GSM900). Should be good also for reception of
signals with frequency hopping.