ptrkrysik/gr-gsm #187: TypeError: msg_connect() takes exactly 5 arguments (3 given)

ifan-boy, 2016-05-12:
I have encountered a problem when I am trying to run airprobe_rtlsdr.py. I have also tried to run the airprobe_rtlsdr.grc in GNU Radio Companion before run the airprobe_rtlsdr.py. Unfortunately, the both methods is failed. I have take the #135 as my reference in order to solve the problem. However, the error stills exist. The errors as below:

Do not run the airprobe_rtlsdr.grc in GNU Radio Companion before execute the airprobe_rtlsdr.py.
`linux; GNU C++ version 4.9.1; Boost_105500; UHD_003.007.003-0-unknown

Gtk-Message: Failed to load module "canberra-gtk-module"
gr-osmosdr 0.1.3 (0.1.3) gnuradio 3.7.5
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy 
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Exact sample rate is: 2000000.052982 Hz
[R82XX] PLL not locked!
Using Volk machine: avx_64_mmx
Using Volk machine: sse4_2_64_orc
Traceback (most recent call last):
  File "airprobe_rtlsdr.py", line 269, in <module>
    tb = airprobe_rtlsdr(fc=options.fc, gain=options.gain, ppm=options.ppm, samp_rate=options.samp_rate, shiftoff=options.shiftoff)
  File "airprobe_rtlsdr.py", line 166, in __init__
    self.msg_connect((self.gsm_bcch_ccch_demapper_0, 'bursts'), (self.gsm_control_channels_decoder_0, 'bursts'))    
TypeError: msg_connect() takes exactly 5 arguments (3 given)`

Run the airprobe_rtlsdr.grc in GNU Radio Companion before execute the airprobe_rtlsdr.py.
The error of GNU Radio Companion when I execute the .airprobe_rtlsdr.grc flow graph
`linux; GNU C++ version 4.9.1; Boost_105500; UHD_003.007.003-0-unknown

Gtk-Message: Failed to load module "canberra-gtk-module"
gr-osmosdr 0.1.3 (0.1.3) gnuradio 3.7.5
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy 
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Exact sample rate is: 2000000.052982 Hz
[R82XX] PLL not locked!
Using Volk machine: avx_64_mmx
Using Volk machine: sse4_2_64_orc
Traceback (most recent call last):
  File "/home/dbsuser/Downloads/software/pybombs/src/gr-gsm/apps/airprobe_rtlsdr.py", line 311, in <module>
    tb.start()
  File "/usr/lib/python2.7/dist-packages/gnuradio/gr/top_block.py", line 104, in start
    top_block_start_unlocked(self._tb, max_noutput_items)
  File "/usr/lib/python2.7/dist-packages/gnuradio/gr/runtime_swig.py", line 4726, in top_block_start_unlocked
    return _runtime_swig.top_block_start_unlocked(*args, **kwargs)
RuntimeError: Clock offset corrector(10): insufficient connected input ports (1 needed, 0 connected)`

The error of airprobe_rtlsdr.py
`linux; GNU C++ version 4.9.1; Boost_105500; UHD_003.007.003-0-unknown

Gtk-Message: Failed to load module "canberra-gtk-module"
gr-osmosdr 0.1.3 (0.1.3) gnuradio 3.7.5
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy 
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Exact sample rate is: 2000000.052982 Hz
[R82XX] PLL not locked!
Using Volk machine: avx_64_mmx
Using Volk machine: sse4_2_64_orc
Traceback (most recent call last):
  File "airprobe_rtlsdr.py", line 269, in <module>
    tb = airprobe_rtlsdr(fc=options.fc, gain=options.gain, ppm=options.ppm, samp_rate=options.samp_rate, shiftoff=options.shiftoff)
  File "airprobe_rtlsdr.py", line 158, in __init__
    self.blocks_socket_pdu_0_0 = blocks.socket_pdu("UDP_SERVER", "127.0.0.1", "4729", 10000)
  File "/usr/lib/python2.7/dist-packages/gnuradio/blocks/blocks_swig5.py", line 919, in make
    return _blocks_swig5.socket_pdu_make(*args, **kwargs)
RuntimeError: bind: Address already in use`

The details of the platform:
Debian 8.4 
gnuradio 3.7.5
gr-osmosdr 0.1.3

Sorry for my sucks english...

Danikon, 2016-06-14:
user@ubuntu:~/gr-gsm/apps$ grgsm_decode -c vf_call6_a725_d174_g5_Kc1EF00BAB3BAC7002.cfile -s $((100000000/174)) -a 725 -m BCCH -t 0
Using Volk machine: ssse3_32_orc
Traceback (most recent call last):
  File "/usr/local/bin/grgsm_decode", line 378, in <module>
    print_bursts=options.print_bursts)
  File "/usr/local/bin/grgsm_decode", line 77, in __init__
    self.input_adapter = grgsm.gsm_input(ppm=0, osr=4, fc=self.fc, samp_rate_in=samp_rate)
  File "/usr/local/lib/python2.7/dist-packages/grgsm/gsm_input.py", line 48, in __init__
    samp_rate_in=samp_rate_in,
  File "/usr/local/lib/python2.7/dist-packages/grgsm/clock_offset_corrector.py", line 47, in __init__
    self.msg_connect((self, 'ppm_in'), (self.gsm_controlled_const_source_f_0, 'constant_msg'))    
TypeError: msg_connect() takes exactly 5 arguments (3 given)

Danikon, 2016-06-14:
i have same problem please help  me

ubuntu 15.04 vmware images 
gnuradio 3.7.5

ifan-boy, 2016-06-14:
I guess the gnuradio 3.7.5 is not support the newest version of gr-gsm. 
I have used the old version of gr-gsm and it is work fine for me.

Danikon, 2016-06-14:
can u give me gr gsm zip file

ptrkrysik, 2016-06-14:
W dniu 14.06.2016 o 10:39, ifan-boy pisze:
> I guess the gnuradio 3.7.5 is not support the newest version of gr-gsm.
> I have used the old version of gr-gsm and it is work fine for me.
> 
Hi,

I thought that my patch was backward-compatible but apparently it wasn't
enough. Generally the GNU Radio that works best with gr-gsm currently is
3.7.9 (older versions - 3.7.8 for sure - tended to have problems with
message passing that resulted with hanging up).

I will try to investigate it but it will take some time as I have to set
up some machine with Ubuntu 15.04.

-- 
Best Regards,
Piotr Krysik

Danikon, 2016-06-14:
ok i will downlod ubuntu 14.04 /15.10  ??? but i don't no GNURadio version on 14.04/15.10

ptrkrysik, 2016-06-14:
W dniu 14.06.2016 o 11:20, Danikon pisze:
> ok i will downlod ubuntu 14.04 /15.10 ??? but i don't no GNURadio
> version on 14.04/15.10
> 
With Ubuntu 16.04 you get GNU Radio 3.7.9 straigh away - from the
repository.

-- 
Piotr Krysik

Danikon, 2016-06-14:
Ok thanks,  Are you tested 16.04 + gr-gm

ptrkrysik, 2016-06-14:
W dniu 14.06.2016 o 11:24, Danikon pisze:
> Ok thanks, Are you tested 16.04 + gr-gm
> 
Yes,

Here you can view automatic tests that are run after each commit to the
gr-gsm's repository:
https://travis-ci.org/ptrkrysik/gr-gsm/builds

(these tests were added thanks to Ash Wilson:
https://github.com/ashmastaflash).

-- 
Best Regards,
Piotr Krysik

Danikon, 2016-06-14:
Hai
new ubuntu 16.04 + VMware player
gnuradio radio 7.9.1
Now no error but not decoding

user@ubuntu:~/gr-gsm/apps$ grgsm_decode -c vf_call6_a725_d174_g5_Kc1EF00BAB3BAC7002.cfile -s $((100000000/174)) -a 725 -m BCCH -t 0
Using Volk machine: ssse3_32_orc
user@ubuntu:~/gr-gsm/apps$

Danikon, 2016-06-14:
user@ubuntu:~/gr-gsm/apps$ grgsm_decode -c sms_multirtl_downlink_tail.cfile
You haven't provided a frequency or an ARFCN - working without automatic frequency offset correction.
Using Volk machine: ssse3_32_orc
user@ubuntu:~/gr-gsm/apps$

ptrkrysik, 2016-06-14:
W dniu 14.06.2016 o 17:56, Danikon pisze:
> Hai
> new ubuntu 16.04 + VMware player
> gnuradio radio 7.9.1
> Now no error but not decoding
> 
> user@ubuntu:~/gr-gsm/apps$ grgsm_decode -c
> vf_call6_a725_d174_g5_Kc1EF00BAB3BAC7002.cfile -s $((100000000/174)) -a
> 725 -m BCCH -t 0
> Using Volk machine: ssse3_32_orc
> user@ubuntu:~/gr-gsm/apps$
> 
Hi,

grgsm_decode wont print messages on the stdout unless you add -v option.

-- 
Best Regards,
Piotr Krysik

Danikon, 2016-06-14:
it is working well
Thanks....

Danikon, 2016-06-17:
Hai, today I got Rtl Sdr stick, i try livemon but problam

**user@ubuntu:~/gr-gsm/apps$** gnuradio-companion grgsm_livemon.grc

<<< Welcome to GNU Radio Companion 3.7.9 >>>

Preferences file: /home/user/.gnuradio/grc.conf
Block paths:
	/usr/local/share/gnuradio/grc/blocks
	/usr/share/gnuradio/grc/blocks
	/home/user/.grc_gnuradio

Loading: "grgsm_livemon.grc"
>>> Error: Block key "rtlsdr_source" not found
>>> Done

Showing: "/home/user/gr-gsm/apps/grgsm_livemon.grc"
>>> Error: Block key "rtlsdr_source" not found

Danikon, 2016-06-19:
it possble to decode TCH/H-half rate channel?

ptrkrysik, 2016-06-19:
Currently it isn't possible.

P.S. please use the mailing-list for questions. Github issues in gr-gsm
are exclusively for bugs/issues and approved (after discussion on the
mailing-list) enhancement requests.

Danikon, 2016-07-04:
Hai @ ptrkrysik please tell me how to decode sdcch8 time slot T1/T2 livemon

ptrkrysik, 2016-10-01:
I don't have time to reproduce the problem with GNU Radio 3.7.5. I'm closing the issue unless someone prepare i.e. docker image that reproduces the problematic setup. Solution is to use systems with newer versions of GNU Radio.

closed 2016-10-01 by ptrkrysik