ptrkrysik/gr-gsm #193: grgsm_decode doesn't decode AMR12.2 correctly

ptrkrysik, 2016-05-30:
grgsm_decode doesn't decode AMR12.2 correctly while hand crafted flowgraph does

ptrkrysik, 2016-06-17:
Notes: reason - lack of ability to turn off voice boundaries detection. Add parameter that enables user to do that.

romankh, 2016-09-17:
Need to get up to date with the current state of the code. 
There is a parameter / selector in gnuradio for that, "Voice boundary detection"
Is it not working ?

ptrkrysik, 2016-09-18:
Hi Roman,

Nice to hear from you again :).

I've changed a little bit checking of checksum for Full Rate codec so
less number of false positives pass through the decoder in case when
voice boundary detection is turned off:
https://github.com/ptrkrysik/gr-gsm/commit/5c9afd38f066bc428bb50d5a0782bd7408f54be7

Don't know if it might have any effect for AMR12.2. As far as I remember
AMR12.2 is the same as EFR. In this case my change might affect decoding
of AMR12.2.

ptrkrysik, 2016-09-18:
I've changed a little bit in TCHF decoding - I removed storing to the file from the decoder and added voice output. It seems in the process I fixed this bug (gsgsm_decode works well now with AMR12.2) - or just turned off the voice boundary detection by default. Here is example how to store voice to a file:
https://github.com/ptrkrysik/examples/blob/f4f7c95b491fd6ce0b47d4f16b5019d69e5e22e8/voice_decoding_in_grc/tch_f_reception.grc
Playback of Full Rate coded voice in GNU Radio is also possible. I also tried to send voice to VLC through UDP - with mixed success.

What is missing is ability to turn on (or off) voice boundary detection in gsgsm_decode with a parameter.

romankh, 2016-09-18:
Hi Piotr !
It's been a while... 
Hope I will be around more often from now on

Adding the option shouldn't be a problem, will assign that to me

romankh, 2016-10-26:
Hi again @ptrkrysik,
I think this issue is fixed and can be closed also, option was implemented and merged

closed 2016-10-27 by ptrkrysik