ptrkrysik/gr-gsm #198: help with grgsm_channelize?

Nik100, 2016-06-17:
Hello! Help me please to understand with decoding hopping channels! Doesn't work «grgsm_channelize.py»: 

1)	monitoring with «grgsm_livemon» ARFCN=99 (954700 kHz) ;
2)	analysis performed by Wireshark, program print, that BS have two carries ARFCN=99 (954700 kHz) and  95 (953900 kHz). Used hopping channels (MAIO 1, HSN 7);
[image]
3)	monitoring ARFCN=95 (953900 kHz ), I see splashes in a signal (I have highlighted with red color);
[image]
4)	wideband capturing with «grgsm_capture» data to cfile, centered 954300 kHz, sample rate 2 MHz (wideband capture contain 95 and 99 ARFCN);
sudo grgsm_capture.py -f 9543e5 -c /home/…/capture/111.cfile -T 60 -v -s 2M
[image]
5)	as a result, 1 file «111.cfile » will be generated (size 960 Mb);
6)	testing my «111.cfile ». Wideband capture cfile contain two signal (95 and 99 ARFCN);
[image]
7)	decoding hopping channels with «grgsm_channelize.py»;
sudo grgsm_channelize.py -s 2e6 -i /home/…/capture/111.cfile -d /home/…/capture/out -f 9543e5 -b P-GSM -o 2e6 95 99
There are mistakes
Input sample rate: 2M
Output sample rate: 2M
==> using resample rate of 1.0
Traceback (most recent call last):
  File "/usr/local/bin/grgsm_channelize.py", line 164, in <module>
    data_type=args.data_type
  File "/usr/local/bin/grgsm_channelize.py", line 94, in __init__
    print("Extracting channels %s, given center frequency at %sHz (ARFCN %d)" % (str(ca), fc_str, center_arfcn))
NameError: global name 'ca' is not defined

Help to understand, please, what’s wrong, I tried a lot of things!

ptrkrysik, 2016-08-01:
I don't know why it didn't work. It does now. Maybe I corrected the problem in the process. 
P.S. if you describe some problem please try to keep the description short. When I looked at this issue first I was convinced that it is a support request and as it required some time to process it I left it for the future. But there was in fact some problem but it was described in the last point of 7 after some graphics that are not so relevant.

closed 2016-08-01 by ptrkrysik

Nik100, 2016-08-07:
Ok! Thank you very much. If you corrected the problem, tell me about it, please!

Varbaek, 2016-08-29:
I've recently tested grgsm_channelize and it does work if you have a valid cfile.

Nik100, 2016-09-04:
Hello,C-file is good,I can send it for you.
But I am not sure.

Понедельник,

Nik100, 2017-04-06:
Всех приветствую!

Хотелось бы познакомить Вас с ситуацией моей племянницы - http://savemenast.blogspot.com.es/

Спасибо!

-- 
Олег Суворов

Varbaek, 2017-04-06:
@Nik100 Reported for spam.