ptrkrysik/gr-gsm #202: A5/3 Decipher

tmeghe, 2016-07-15:
Hi,

I'm trying to decipher a specific assignment on one SDCCH8 using my own Kc (gathered one my phone). The cap file is one the GSM900 band however the BTS ciphering is A5/3. grgsm_decode supports A5/3 but it doesn't work there, the output is still ciphered. Any hint (maybe A5/3 deciphering only works on some specific cases ?)

Regards,

ptrkrysik, 2016-07-15:
W dniu 15.07.2016 o 14:02, Thomas pisze:
> Hi,
>
> I'm trying to decipher a specific assignment on one SDCCH8 using my own
> Kc (gathered one my phone). The cap file is one the GSM900 band however
> the BTS ciphering is A5/3. grgsm_decode supports [...]

Hi Thomas,

A5/3 wasn't tested because I didn't have any test data for that. The
implementation was taken from libosmocore but it's usage wasn't ever tested.

-- 
Best Regards,
Piotr Krysik

tmeghe, 2016-07-15:
Thanks for the information. I will look further.
Cheers,

closed 2016-07-15 by tmeghe

Varbaek, 2016-08-29:
I've tested A5/3 with SMS and it worked in my case.

danyhm, 2016-08-29:
Did you get Kc from simcard? What band did you try? 3G ? I'm really
interested.

Varbaek, 2016-08-29:
I extracted the Kc from the USB modem on my mobile phone after I had received an SMS.
The band I used was P-GSM, even though this shouldn't matter. I forced my phone to use 2G, as this project (gr-gsm) is based on 2G work, not 3G.

danyhm, 2016-08-29:
It's true gr-gsm is based on 2G. But as far as i know (i think) A5/3 is
used in 3G networks. So are you saying your operator is using A5/3 for
encryption in GSM band? And that you could decrypt it using the same
commands?

Varbaek, 2016-08-29:
When I force my Samsung to use 2G, it uses A5/3 with my main provider on the P-GSM (GSM900) band. No special commands or modifications were made. Frequency hopping is enabled in my case, so I can only decode SMS, but yes it works. Try it out yourself, set your phone into 2G mode only, and if the cipher mode command says A5/3, then decode your own SMS. Just make sure that when you extract the Kc, that you remove the last byte (2 characters), at least when you extract it from a USB modem you have to do that.

arifkyi, 2019-03-29:
> It's true gr-gsm is based on 2G. But as far as i know (i think) A5/3 is
> used in 3G networks. So are you saying your operator is using A5/3 for
> encryption in GSM band? And that you could decrypt it using the same
> commands?
> 
> 
> > I extracted the Kc from the USB modem on my mobile phone after I had
> > received an SMS.
> > The band I used was P-GSM, even though this shouldn't matter. I forced my
> > phone to use 2G, as this project (gr-gsm) is based on 2G work, not 3G.
> > [#202 (comment)](https://github.com/ptrkrysik/gr-gsm/issues/202#issuecomment-243216340),

one of Operator (from total 4 operators) in Indonesia they use A5/3 in their 2G networks

rajanpasupuleti, 2020-09-15:
> I've tested A5/3 with SMS and it worked in my case.

Hi - could help with the steps that you proceeded