ptrkrysik/gr-gsm #204: TypeError: Required argument 'samp_rate' (pos 2) not found

nikosvir, 2016-07-17:
Hi,

I have the following issue with airprobe_rtlsdr_capture.py:

root@ubuntu:~# airprobe_rtlsdr_capture.py -s 1M -f 936600000 -g 44 -b /tmp/gsm
linux; GNU C++ version 5.4.0 20160609; Boost_105800; UHD_003.010.000.git-249-gef57ffcb

gr-osmosdr v0.1.4-72-g164a09fc (0.1.5git) gnuradio v3.7.10-1-ge55666b7
built-in source types: file osmosdr fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy redpitaya 
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Exact sample rate is: 1000000.026491 Hz
[R82XX] PLL not locked!
Traceback (most recent call last):
  File "/usr/local/bin/airprobe_rtlsdr_capture.py", line 264, in <module>
    rec_length=options.rec_length, args=options.args)
  File "/usr/local/bin/airprobe_rtlsdr_capture.py", line 89, in __init__
    self.gsm_clock_offset_control = grgsm.clock_offset_control(fc-shiftoff)
  File "/usr/local/lib/python2.7/dist-packages/grgsm/grgsm_swig.py", line 482, in make
    return _grgsm_swig.clock_offset_control_make(*args, **kwargs)
**TypeError: Required argument 'samp_rate' (pos 2) not found**

This happens regardless if I give the -s parameter on not.
When I run the same command again I get a core dump:

root@ubuntu:~# airprobe_rtlsdr_capture.py -s 1M -f 936600000 -g 44 -b /tmp/gsm
linux; GNU C++ version 5.4.0 20160609; Boost_105800; UHD_003.010.000.git-249-gef57ffcb

gr-osmosdr v0.1.4-72-g164a09fc (0.1.5git) gnuradio v3.7.10-1-ge55666b7
built-in source types: file osmosdr fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy redpitaya 
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Exact sample rate is: 1000000.026491 Hz
[R82XX] PLL not locked!
Traceback (most recent call last):
  File "/usr/local/bin/airprobe_rtlsdr_capture.py", line 264, in <module>
    rec_length=options.rec_length, args=options.args)
  File "/usr/local/bin/airprobe_rtlsdr_capture.py", line 89, in __init__
    self.gsm_clock_offset_control = grgsm.clock_offset_control(fc-shiftoff)
  File "/usr/local/lib/python2.7/dist-packages/grgsm/grgsm_swig.py", line 482, in make
    return _grgsm_swig.clock_offset_control_make(*args, **kwargs)
TypeError: Required argument 'samp_rate' (pos 2) not found
*** Error in `/usr/bin/python2': double free or corruption (!prev): 0x000000000374c2c0 ***
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x77725)[0x7f13d8177725]
/lib/x86_64-linux-gnu/libc.so.6(+0x7ff4a)[0x7f13d817ff4a]
/lib/x86_64-linux-gnu/libc.so.6(cfree+0x4c)[0x7f13d8183abc]
/usr/local/lib/libgnuradio-runtime-3.7.11git.so.0.0.0(+0x7746a)[0x7f13d5a3e46a]
/usr/local/lib/libgnuradio-runtime-3.7.11git.so.0.0.0(+0x7743c)[0x7f13d5a3e43c]
/usr/local/lib/libgnuradio-runtime-3.7.11git.so.0.0.0(+0x7743c)[0x7f13d5a3e43c]
/lib/x86_64-linux-gnu/libc.so.6(+0x39fe8)[0x7f13d8139fe8]
/lib/x86_64-linux-gnu/libc.so.6(+0x3a035)[0x7f13d813a035]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf7)[0x7f13d8120837]
/usr/bin/python2(_start+0x29)[0x49dbf9]
======= Memory map: ========
00400000-006e9000 r-xp 00000000 08:01 5372                               /usr/bin/python2.7
008e9000-008eb000 r--p 002e9000 08:01 5372                               /usr/bin/python2.7
008eb000-00962000 rw-p 002eb000 08:01 5372                               /usr/bin/python2.7
00962000-00985000 rw-p 00000000 00:00 0 
0294b000-0385a000 rw-p 00000000 00:00 0                                  [heap]
7f13b8000000-7f13b8021000 rw-p 00000000 00:00 0 
7f13b8021000-7f13bc000000 ---p 00000000 00:00 0 
7f13bea92000-7f13bea9d000 r-xp 00000000 08:01 923120                     /lib/x86_64-linux-gnu/libnss_files-2.23.so
7f13bea9d000-7f13bec9c000 ---p 0000b000 08:01 923120                     /lib/x86_64-linux-gnu/libnss_files-2.23.so
7f13bec9c000-7f13bec9d000 r--p 0000a000 08:01 923120                     /lib/x86_64-linux-gnu/libnss_files-2.23.so
7f13bec9d000-7f13bec9e000 rw-p 0000b000 08:01 923120                     /lib/x86_64-linux-gnu/libnss_files-2.23.so
7f13bec9e000-7f13beca4000 rw-p 00000000 00:00 0 
7f13beca4000-7f13beca5000 ---p 00000000 00:00 0 
[... 274 lines cut ...]
7f13cbde0000-7f13cbde6000 r-xp 00000000 08:01 12360                      /usr/lib/x86_64-linux-gnu/libfftw3f_threads.so.3.4.4
7f13cbde6000-7f13cbfe5000 ---p 00006000 08:01 12360                      /usr/lib/x86_64-linux-gnu/libfftw3f_threads.so.3.4.4
7f13cbfe5000-7f13cbfe7000 r--p 00005000 08:01 12360                      /usr/lib/x86_64-linux-gnu/libfftw3f_threads.so.3.4.4
7f13cbfe7000-7f13cbfe8000 rw-p 00007000 08:01 12360                      /usr/lib/x86_64-linux-gnu/libfftw3f_threads.so.3.4.4
7f13cbfe8000-7f13cc1df000 r-xp 00000000 08:01 12356                      /usr/lib/x86_64-linux-gnu/libfftw3f.so.3.4.4
7f13cc1df000-7f13cc3de000 ---p 001f7000 08:01 12356                      /usr/lib/x86_64-linux-gnu/libfftw3f.so.3.4.4
7f13cc3de000-7f13cc3f2000 r--p 001f6000 08:01 12356                      /usr/lib/x86_64-linux-gnu/libfftw3f.so.3.4.4
7f13cc3f2000-7f13cc3f3000 rw-p 0020a000 08:01 12356                      /usr/lib/x86_64-linux-gnu/libfftw3f.so.3.4.4
7f13cc3f3000-7f13cc41e000 r-xp 00000000 08:01 302121                     /usr/local/lib/libgnuradio-fft-3.7.11git.so.0.0.0
7f13cc41e000-7f13cc61e000 ---p 0002b000 08:01 302121                     /usr/local/lib/libgnuradio-fft-3.7.11git.so.0.0.0
7f13cc61e000-7f13cc621000 r--p 0002b000 08:01 302121                     /usr/local/lib/libgnuradio-fft-3.7.11git.so.0.0.0
7f13cc621000-7f13cc622000 rw-p 0002e000 08:01 302121                     /usr/local/lib/libgnuradio-fft-3.7.11git.so.0.0.0
7f13cc622000-7f13cc6fe000 r-xp 00000000 08:01 302203                     /usr/local/lib/libgnuradio-filter-3.7.11git.so.0.0.0
7f13cc6fe000-7f13cc8fd000 ---p 000dc000 08:01 302203                     /usr/local/lib/libgnuradio-filter-3.7.11git.so.0.0.0
7f13cc8fd000-7f13cc90b000 r--p 000db000 08:01 302203                     /usr/local/lib/libgnuradio-filter-3.7.11git.so.0.0.0
7f13cc90b000-7f13cc90c000 rw-p 000e9000 08:01 302203                     /usr/local/lib/libgnuradio-filter-3.7.11git.so.0.0.0
7f13cc90c000-7f13cc90d000 rw-p 00000000 00:00 0 
7f13cc90d000-7f13ccd56000 r-xp 00000000 08:01 302208                     /usr/local/lib/python2.7/dist-packages/gnuradio/filter/_filter_swig.so
7f13ccd56000-7f13ccf55000 ---p 00449000 08:01 302208                     /usr/local/lib/python2.7/dist-packages/gnuradio/filter/_filter_swig.so
7f13ccf55000-7f13ccf58000 r--p 00448000 08:01 302208                     /usr/local/lib/python2.7/dist-packages/gnuradio/filter/_filter_swig.so
7f13ccf58000-7f13ccf7b000 rw-p 0044b000 08:01 302208                     /usr/local/lib/python2.7/dist-packages/gnuradio/filter/_filter_swig.so
7f13ccf7b000-7f13cd2fc000 rw-p 00000000 00:00 0 
7f13cd2fc000-7f13cd372000 r-xp 00000000 08:01 35365                      /usr/lib/python2.7/dist-packages/numpy/random/mtrand.x86_64-linux-gnu.so
7f13cd372000-7f13cd572000 ---p 00076000 08:01 35365                      /usr/lib/python2.7/dist-packages/numpy/random/mtrand.x86_64-linux-gnu.so
7f13cd572000-7f13cd573000 r--p 00076000 08:01 35365                      /usr/lib/python2.7/dist-packages/numpy/random/mtrand.x86_64-linux-gnu.soAborted (core dumped)

ptrkrysik, 2016-07-18:
The first problem is a result of recent changes in gr-gsm. However
having the receiver in the capture application doesn't seem like very
needed feature and I will probably solve the problem by removing it
completely.

ptrkrysik, 2016-07-18:
I'm closing the issue as I've corrected it. Regarding the core dump - if it persist it will require separate issue.

closed 2016-07-18 by ptrkrysik