ptrkrysik/gr-gsm #206: grgsm_decode - tch_f_decoder_make - argument 2 of type 'bool'

Varbaek, 2016-07-19:
When I try to use grgsm_decode with TCHF, I get the following error:

```
Traceback (most recent call last):
  File "/usr/local/bin/grgsm_decode", line 377, in <module>
    print_bursts=options.print_bursts)
  File "/usr/local/bin/grgsm_decode", line 98, in __init__
    self.tch_f_decoder = grgsm.tch_f_decoder(speech_codec, speech_file)
  File "/usr/local/lib/python2.7/dist-packages/grgsm/grgsm_swig.py", line 1640, in make
    return _grgsm_swig.tch_f_decoder_make(mode, boundary_check)
TypeError: in method 'tch_f_decoder_make', argument 2 of type 'bool'
```

Everything else works great :-)

ptrkrysik, 2016-07-19:
Thank you for that report!

This is result of another change in gr-gsm. I removed saving to a file
from TCH/F decoding block and added output for voice frames. PDU
messages can be turned into signal samples and stored to a file. An
example of that is in the

gr-gsm/examples/voice_decoding_in_grc

directory.

Also - as GNU Radio has GSM Full Rate codec implemented it is possible
to play speech coded with Full Rate codec in GNU Radio directly.

However, I haven't integrated that into grgsm_decode and thank you for
reporting it.

Varbaek, 2016-07-19:
Thanks for the detailed response.
Is there an easy/quick fix that I can do without using GNU Radio?

Varbaek, 2016-07-19:
Also, thanks for the tch_f_reception.grc file, I'll play with that for now :-)

Varbaek, 2016-07-20:
I tried to use "tch_f_reception.grc", after modifying the input file, frequency, sample rate, Kc and algorithm (A5/3). As it didn't work (with my capture file), I tried to modify a few more parameters. (Didn't work either, but I only tried it with my capture file.)

Ended up downloading an older version of your code, where grgsm_decode works 100% with the srlabs test file. (Awesome)

In the channel description I saw that it said Hopping Channel: Yes (for the TCH/F assignment command), and I guess that's why I can't "decode" my own captured file (from my own phone). 

I've closed the ticket as well.

ptrkrysik, 2016-07-20:
I'm reopening the issue as there is in fact there is little problem in grgsm_decode. tch_f_reception.grc reception should work with the srlabs file - but when you are running the generated python script in it's directory. I haven't figured out how to use relative paths to data in gnuradio-companion yet.

Varbaek, 2016-07-20:
You're right that relative paths don't seem to work. However, absolute paths work fine which is better than nothing working at all.

ptrkrysik, 2016-07-20:
I've added output to a file to grgsm_decode for TCHF channel. However it is still not ideal as one of the blocks I used is apparently causing that application doesn't end when processing TCHF.

JamesonBeebe, 2017-03-30:
Was this issue ever resolved or did someone ever find a way around it? I am getting the same error with the vf_call6... test file.

Varbaek, 2017-03-31:
@JamesonBeebe Download the packaging branch and try that instead. I don't think the master/main branch ever had this issue fixed code-wise.

ptrkrysik, 2017-04-02:
The problem is that I don't see this problem on my side.
When I run:
grgsm_decode -c vf_call6_a725_d174_g5_Kc1EF00BAB3BAC7002.cfile -s
$((100000000/174)) -a 725 -m TCHF -t 5 -e 1 -k
0x1E,0xF0,0x0B,0xAB,0x3B,0xAC,0x70,0x02 -d FR -o /tmp/speech.au.gsm

then:
vlc /tmp/speech.au.gsm

everything is fine.

Maybe version of gnuradio makes difference. I've got 3.7.9.

JamesonBeebe, 2017-04-03:
@Varbaek Thanks for the advice with the Packaging Distro. That seemed to do the trick. 

The strange part is that the master branch did work at a time. I must have upgraded something that broke it. 

Currently I am running:
OS: Ubuntu 16.04
Gnu Radio: 3.7.12git-50-g1d477a70

Varbaek, 2017-04-03:
@JamesonBeebe No problem, the master branch had this feature removed for unknown reasons.

ptrkrysik, 2017-04-03:
W dniu 03.04.2017 o 17:39, Jamdman pisze:
> @Varbaek <https://github.com/Varbaek> Thanks for the advice with the
> Packaging Distro. That seemed to do the trick.
>
> The strange part is that the master branch did work at a time. I must
> have [...]
I checked on GNU Radio 3.7.12git-29-gac5109d2 and Ubuntu 16.04 but I'm
unable to get that error. Can you post the exact command you run and the
result?

ptrkrysik, 2017-04-03:
@Varbaek, I'm looking closer at the error that you posted at the
beginning (on July). Currently in grgsm_decode there is no such line as:

    self.tch_f_decoder = grgsm.tch_f_decoder(speech_codec, speech_file)

in grgsm_decode.

It looks that you had or still have old grgsm_decode and new
installation of grgsm library.

Can you check what you see currently when you try to decode TCHF with
grgsm_decode?

Varbaek, 2017-04-03:
@ptrkrysik Sure. 

I git cloned the newest version from the master branch, compiled, installed, and ldconfig.

Then I did the usual decoding which worked all fine until I tried to decode TCHF. As you can see in the dump below, the line numbers are different compared to my first issue that I opened. 

root@kali:~/grgsm_new/test_data# grgsm_decode -a 725 -s $((100000000/174)) -c vf_call6_a725_d174_g5_Kc1EF00BAB3BAC7002.cfile -m TCHF -t 5 -e 1 -k  '1EF00BAB3BAC7002' -d FR -o ./voice.gsm
Traceback (most recent call last):
  File "/usr/local/bin/grgsm_decode", line 387, in <module>
    print_bursts=options.print_bursts, ppm=options.ppm)
  File "/usr/local/bin/grgsm_decode", line 100, in __init__
    self.tch_f_decoder = grgsm.tch_f_decoder(speech_codec, enable_voice_boundary_detection)
  File "/usr/local/lib/python2.7/dist-packages/grgsm/grgsm_swig.py", line 1815, in make
    return _grgsm_swig.tch_f_decoder_make(mode, boundary_check)
TypeError: in method 'tch_f_decoder_make', argument 2 of type 'bool'

If you look at the following file, you can see that:
https://github.com/ptrkrysik/gr-gsm/blob/master/apps/grgsm_decode

You can see that the line exists: 
self.tch_f_decoder = grgsm.tch_f_decoder(speech_codec, enable_voice_boundary_detection)

Varbaek, 2017-04-03:
If you run grgsm_decode with --voice-boundary, the script seems to work. 

The help page (--help) does not state that this is a mandatory option that the user needs to set for the script to work though. This is the line I used the second time:

grgsm_decode -a 725 -s $((100000000/174)) -c vf_call6_a725_d174_g5_Kc1EF00BAB3BAC7002.cfile -m TCHF -t 5 -e 1 -k '1EF00BAB3BAC7002' -d FR -o ./voice.gsm

In that case the script runs forever until the user kills the script with CTRL+C.

The produced file contains valid audio which can be played back with VLC.

E3V3A, 2017-06-20:
I have indeed the same exact experience on Kali. This issue is almost one year old...so a fix would be nice and timely.

romankh, 2017-06-20:
@Varbaek: --voice-boundary is not intended to be mandatory, it is an option that in some cases may improve the sound quality of the audio.

Varbaek, 2017-06-20:
@romankh Did you even read my responses? The packaging branch doesn't have this issue. I already solved this issue on my own computer last year. This year the developer asked me to check this on the latest version, so I did and I got the same issue again that there's a missing argument, so I added --voice-boundary and it worked, despite running forever. I don't get the point of your comment as it seems like you're either using a version that doesn't have this bug anymore, or you haven't even tried running this program yet. 

Did you read this line of code I posted about:
self.tch_f_decoder = grgsm.tch_f_decoder(speech_codec, enable_voice_boundary_detection)

And also this error:
TypeError: in method 'tch_f_decoder_make', argument 2 of type 'bool'

Just because the "Help page" says it's not mandatory, does not mean the code reflects the same. The evidence is pretty clear if you just read my responses and recommended workarounds, instead of just quoting the help page without having tried to run this code.

Varbaek, 2017-06-20:
@romankh Just because the developer states that --voice-boundary is not intended to be mandatory, does not mean you should wait until he provides a fix for it if there's a quick workaround like I posted. In fact there are two work arounds:
a) Download packaging branch; or
b) Set --voice-boundary so that it returns true like a bool.

There is another option: c) Fix the developers code, but the developer is more than capable of fixing this issue himself as gr-gsm is after all not a simple program at all.

romankh, 2017-06-20:
@Varbaek : wow, sweet little rage post. 

Where exactly is your quick fix ? Adding the option ? That is not fix.

> There is another option: c) Fix the developers code, but the developer is more than capable of fixing this issue himself as gr-gsm is after all not a simple program at all.

Thats a nice one. I implemented this particular feature.
And ok, why provide a pull request, if you can let others do all the work, right ? 

So, let me clear this out for you:
If you have to add that option to make it stop, then it's a bug.

Varbaek, 2017-06-20:
@romankh You need to learn what "workarounds" mean. They're not solutions, they're temporary fixes so that people can continue to use the code.

If you implemented this feature/bug, then how come you didn't test it properly in your QA? Or did the other main developer change your code? 

I do agree this is a bug, because the program doesn't work properly without it in the current branch.

romankh, 2017-06-20:
> You need to learn what "workarounds" mean. They're not solutions, they're temporary fixes so that people can continue to use the code.

Thanks for the lesson, professor. I somehow knew your are not the kind of guy who can take that without a cheeky response. Ok, let's go back to the beginning, this was your text:

> The help page (--help) does not state that this is a mandatory option that the user needs to set for the script to work though. This is the line I used the second time:

Basically you claim that the help page is missing to state that --voice-boundary is mandatory. 
And I dared to post that 

> --voice-boundary is not intended to be mandatory, it is an option that in some cases may improve the sound quality of the audio.

Truely offensive, so, lets bring that to an end:
Sorry, princess, didnt mean to upset you. 

> If you implemented this feature/bug, then how come you didn't test it properly in your QA? Or did the other main developer change your code?

This is an free and open source project, where some developers invested their free time in creating something and share that something with others, for free. No foss developer owes you any level of quality, nor any QA tests, nor any functionality or anything at all.

But, as I am always interested in learning from the best, give us the honour, enlighten us with pull requests fixing that bug and raise the test coverage of this project to unimagined heights.

Varbaek, 2017-06-20:
@romankh At least test your code like a proper developer. How can you take yourself serious? I also used my own free time, free of charge, pro bono to report this issue, and even follow up on it after the main developer asked me to. Did I really want to? No not really but I did it anyway. You started this by trying to make my statement invalid when it's completely valid.

Varbaek, 2017-06-20:
@romankh Calling me names just makes you look more unprofessional.

E3V3A, 2017-06-21:
Come on guys, this is totally not the way to get anything fixed. 

@ptrkrysik  Can you please remove the last couple of comments and implement any temporary workaround until master branch is fixed?

E3V3A, 2017-06-21:
And yes, the documentation of this project could use some TLC and updates! 
The following links are simply outdated and doesn't reflect all the command options, some which are completely unexplained. 

* https://github.com/ptrkrysik/gr-gsm/wiki/Usage
* https://github.com/ptrkrysik/gr-gsm/wiki/Usage:-Decoding-How-To

Unfortunately, this lack of documentation on close to all GitHub projects are very counter productive and make people post bug/issues for just about anything.

closed 2017-07-23 by ptrkrysik

ptrkrysik, 2017-07-23:
Hi all,

The problem was very simple to fix but it was present for long (a year). Big sorry for that. I don't think it was worth this argument and to lose nerves though. As suggested by @E3V3A I will remove the comments with the argument.

Varbaek, 2017-07-23:
Thanks @ptrkrysik, sounds great 👍

commit 2770fc48 (2017-07-23): Added missing default value of voice-boundary parameter. Fixes #206