ptrkrysik/gr-gsm #208: grgsm_capture.py sample rate error

danyhm, 2016-08-14:
I tried to run 
grgsm_capture.py -f 957400000 -s 6M -b file -T 60

but i get the following error
```
linux; GNU C++ version 6.1.1 20160802; Boost_106100; UHD_003.009.004-0-unknown

gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.10
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy redpitaya 
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Exact sample rate is: 1000000.026491 Hz
[R82XX] PLL not locked!
Traceback (most recent call last):
  File "/home/daniel/gr-gsm/bin/grgsm_capture.py", line 264, in <module>
    rec_length=options.rec_length, args=options.args)
  File "/home/daniel/gr-gsm/bin/grgsm_capture.py", line 89, in __init__
    self.gsm_clock_offset_control = grgsm.clock_offset_control(fc-shiftoff, sample_rate, osr=4)
NameError: global name 'sample_rate' is not defined
```
it seemed to me that it should be samp_rate instead i tried changing it but got another error related to swag.
 any ideas ?

danyhm, 2016-08-14:
i edited the grgsm_capture.py and it didn't work and then I reverted the changes and it worked!
don't know what really caused the problem strange.

danyhm, 2016-08-14:
i checked again it's a problem with -b parameter

romankh, 2016-09-20:
Samplerate of 3000000 is to high for R820T, as far as I remember 2M is the max. supported by R820T

romankh, 2016-10-01:
Seems there is a wrong variable name used for setting sample rate in clock offset control, will try to fix that

romankh, 2016-12-05:
@danyhm: can this issue be closed ?

danyhm, 2016-12-07:
@romankh

It works now.
Thanks for fixing the problem roman !

The issue can be closed.

romankh, 2016-12-08:
@danyhm Thanks for feedback

closed 2016-12-08 by romankh