ptrkrysik/gr-gsm #209: grgsm_livemon rise an error

tinyexplorer, 2016-08-15:
I follow wiki to install the tools on Ubuntu 16.04 Desktop (x32). Looks like installation without any errors, but..

I got an error when try to run - 
> grgsm_livemon

> Traceback (most recent call last):
>   File "/usr/local/bin/grgsm_livemon", line 31, in <module>
>     import grgsm
>   File "/usr/local/lib/python2.7/dist-packages/grgsm/__init__.py", line 45, in <module>
>     from grgsm_swig import *
>   File "/usr/local/lib/python2.7/dist-packages/grgsm/grgsm_swig.py", line 28, in <module>
>     _grgsm_swig = swig_import_helper()
>   File "/usr/local/lib/python2.7/dist-packages/grgsm/grgsm_swig.py", line 24, in swig_import_helper
>     _mod = imp.load_module('_grgsm_swig', fp, pathname, description)
> ImportError: /usr/local/lib/python2.7/dist-packages/grgsm/_grgsm_swig.so: undefined symbol: _ZN2gr3gsm34controlled_fractional_resampler_cc4makeEff

some other commands 

> grgsm_scanner

> Traceback (most recent call last):
>   File "/usr/local/bin/grgsm_scanner", line 33, in <module>
>     import grgsm
>   File "/usr/local/lib/python2.7/dist-packages/grgsm/__init__.py", line 45, in <module>
>     from grgsm_swig import *
>   File "/usr/local/lib/python2.7/dist-packages/grgsm/grgsm_swig.py", line 28, in <module>
>     _grgsm_swig = swig_import_helper()
>   File "/usr/local/lib/python2.7/dist-packages/grgsm/grgsm_swig.py", line 24, in swig_import_helper
>     _mod = imp.load_module('_grgsm_swig', fp, pathname, description)
> ImportError: /usr/local/lib/python2.7/dist-packages/grgsm/_grgsm_swig.so: undefined symbol: _ZN2gr3gsm34controlled_fractional_resampler_cc4makeEff
> 

It's possible to fix?

ognistyi, 2016-08-18:
I setup latest Ubuntu 14.04, then install gr-gsm (pybombs) bundle on it. So all working great, but after I reboot PC I got -

> $ sudo grgsm_livemon
Traceback (most recent call last):
  File "/usr/local/bin/grgsm_livemon", line 31, in <module>
    import grgsm
  File "/usr/local/lib/python2.7/dist-packages/grgsm/__init__.py", line 45, in <module>
    from grgsm_swig import *
  File "/usr/local/lib/python2.7/dist-packages/grgsm/grgsm_swig.py", line 28, in <module>
    _grgsm_swig = swig_import_helper()
  File "/usr/local/lib/python2.7/dist-packages/grgsm/grgsm_swig.py", line 24, in swig_import_helper
    _mod = imp.load_module('_grgsm_swig', fp, pathname, description)
ImportError: /usr/local/lib/python2.7/dist-packages/grgsm/_grgsm_swig.so: undefined symbol: _ZN2gr3gsm34controlled_fractional_resampler_cc4makeEff

really disappointed with it.

ptrkrysik, 2016-08-25:
W dniu 18.08.2016 o 13:00, Andrew pisze:
> I setup latest Ubuntu 14.04, then install gr-gsm (pybombs) bundle on it.
> So all working great, but after I reboot PC I got -
> 
>     $ sudo grgsm_livemon
>     Traceback (most recent call last):
>     File "/usr/local/bin/grgsm_livemon", line 31, in
>     import grgsm
>     File "/usr/local/lib/python2.7/dist-packages/grgsm/*init*.py", line
>     45, in
>     from grgsm_swig import *
>     File "/usr/local/lib/python2.7/dist-packages/grgsm/grgsm_swig.py",
>     line 28, in
>     _grgsm_swig = swig_import_helper()
>     File "/usr/local/lib/python2.7/dist-packages/grgsm/grgsm_swig.py",
>     line 24, in swig_import_helper
>     _mod = imp.load_module('_grgsm_swig', fp, pathname, description)
>     ImportError:
>     /usr/local/lib/python2.7/dist-packages/grgsm/_grgsm_swig.so:
>     undefined symbol: _ZN2gr3gsm34controlled_fractional_resampler_cc4makeEff
> 
> really disappointed with it.

Did you install everything on a fresh Ubuntu installation? Did you run:
sudo ldconfig?

tinyexplorer, 2016-09-11:
I install fresh copy of ubuntu, install pybombs, gr-gsm, than install gnuradio and do reboot.. after reboot got the issue. I'm not run sudo ldconfig. So, I try this but it not helps.

tinyexplorer, 2016-09-15:
is there right way to try reinstall this bundle?

ptrkrysik, 2017-06-11:
I'm closing the issue. It it is still present after reinstall I'll reopen it.

closed 2017-06-11 by ptrkrysik