ptrkrysik/gr-gsm #213: Incorrect or missing CID/LAC/MCC/MNC?

tkuester, 2016-08-18:
On a test scan in my area, I observed discrepancies between gr-gsm and a separate tool on the reported CID and MNC. Some MNC's for my area are T-Mobile (260) and AT&T (410). Are you able to provide any insight?

<pre>
$ grgsm_scanner -b PCS1900
linux; GNU C++ version 4.8.4; Boost_105400; UHD_003.009.002-0-gf18abe54

ARFCN:  512, Freq: 1930.2M, CID:     0, LAC: 10454, MCC: 310, MNC:  41, Pwr: -58
ARFCN:  535, Freq: 1934.8M, CID:     0, LAC: 10454, MCC: 310, MNC:  41, Pwr: -45
ARFCN:  692, Freq: 1966.2M, CID:  3492, LAC: 32451, MCC: 310, MNC:  26, Pwr: -57
ARFCN:  693, Freq: 1966.4M, CID:     0, LAC:     0, MCC:   0, MNC:   0, Pwr: -61
ARFCN:  694, Freq: 1966.6M, CID:     0, LAC: 32451, MCC: 310, MNC:  26, Pwr: -52
ARFCN:  701, Freq: 1968.0M, CID:     0, LAC: 32451, MCC: 310, MNC:  26, Pwr: -51
</pre>

With an alternate tool, I was able to obtain the following.

<pre>
        {
            "arfcn": 701, 
            "bsic": 53, 
            "id": 0, 
            "lac": 32451, 
            "mcc": 310, 
            "mnc": 260, 
            "rxlev": -87
        },
        {
            "arfcn": 692, 
            "bsic": 46, 
            "id": 3492, 
            "lac": 32451, 
            "mcc": 310, 
            "mnc": 260, 
            "rxlev": -94
        }, 
        {
            "arfcn": 694, 
            "bsic": 31, 
            "id": 4653, 
            "lac": 32451, 
            "mcc": 310, 
            "mnc": 260, 
            "rxlev": -84, 
        }
</pre>

danyhm, 2016-08-18:
I'm not sure but i think MNC is a 2 digit code. So probably your software
is adding a zero at the end maybe?
Also grgsm_scanner can be buggy sometimes and not report the values some
times. Try running the test multiply times and also try changing the
scanning speed.

tkuester, 2016-08-18:
The MNC can definitely be three digits, at least according to [Wikipedia's list (warning, large page)](https://en.wikipedia.org/wiki/Mobile_country_code)

I am unable to find any pairings of 310-26 or 310-41 on the above page, but I am able to find 310-260 (AT&T) and 310-410 (T-Mobile). I can confirm the T-Mobile one, as it's what my equipment is currently using.

danyhm, 2016-08-18:
Then i was wrong :)

tkuester, 2016-08-18:
Thank you for the suggestions, btw! Sorry, I wrote that while in the middle of another conversation.

romankh, 2016-09-27:
@tkuester: can you provide a capture file of a few seconds, containing some messages with a 3 digit MNC? I can fix that in the next few days, just would be fine to have something to verify with.

tkuester, 2016-12-01:
Hey, I don't quite have the time to test this right now... but I'd love to get back to this soon! Thanks for the commit!

romankh, 2016-12-02:
@tkuester, would be great. If you find out the fix didnt work, please consider providing a cfile or burst file with system information message containing a mnc with 3 digits

ptrkrysik, 2026-11-09:
This was fixed in 2016 (three-digit MNC handling in extract_system_info) and is present in the current code. Please open a new issue with a capture file if you still see wrong MNC/CID values.

I am closing this as part of a clean-up for the new release (v0.43.0), which ports gr-gsm to GNU Radio 3.10 (plain CMake build, no SWIG, installation steps in the README). If the problem still occurs with it, please open a new issue with your operating system, the output of `gnuradio-config-info --version`, how gr-gsm and its dependencies were installed, the exact command and the complete output.

closed 2026-11-09 by ptrkrysik