ptrkrysik/gr-gsm #214: gsm_wideband_input2 runtime error

doraexploradora2014, 2016-08-25:
Hi, I have a problem with GSM Wideband Input Adaptor block. When I run the flowgraph this it  stop  at line 51 of  gsm_wideband_input.py, with runtime error:

self.gsm_clock_offset_corrector_0 = grgsm.gsm_clock_offset_corrector(
AttributeError: 'module' object has no attribute 'gsm_clock_offset_corrector'

I  saw another class, gsm_clock_offset_corrector_tagged, so I replaced with this but when I did it I had new errors. 

Thanks!
Dora

doraexploradora2014, 2016-08-26:
well, I think is rigth change by clock_offset_corrector_tagged, but I see I have another problem by the gnuradio : RuntimeError: invalid msg port in connect() or disconnect()

Traceback (most recent call last):
  File "/home/u/Documents/Aptana Studio 3 Workspace/test/wideband.py", line 293, in <module>
    main()
  File "/home/u/Documents/Aptana Studio 3 Workspace/test/wideband.py", line 287, in main
    tb = top_block_cls(fc=options.fc, osr=options.osr)
  File "/home/u/Documents/Aptana Studio 3 Workspace/test/wideband.py", line 175, in __init__
    band="PCS1900",
  File "/home/u/Documents/Aptana Studio 3 Workspace/test/wideband.py", line 101, in __init__
    self.msg_connect(self, "ppm_in", self.gsm_clock_offset_corrector_0, "ppm_in")
  File "/usr/lib/python2.7/dist-packages/gnuradio/gr/hier_block2.py", line 59, in wrapped
    func(self, src.to_basic_block(), srcport, dst.to_basic_block(), dstport)
  File "/usr/lib/python2.7/dist-packages/gnuradio/gr/hier_block2.py", line 131, in msg_connect
    self.primitive_msg_connect(*args)
  File "/usr/lib/python2.7/dist-packages/gnuradio/gr/runtime_swig.py", line 3584, in primitive_msg_connect
    return _runtime_swig.hier_block2_sptr_primitive_msg_connect(self, *args)
RuntimeError: invalid msg port in connect() or disconnect()

ptrkrysik, 2016-08-27:
The name of the input of the clock offset corrector changed.
I forgot to change the gsm_wideband_input block because I wasn't aware
anyone is using it and forgot about this block.

In my opinion having one gsm_input block that just group other blocks in
order to make a simple flowgraphs smaller in gnuradio-companion is
enough. I would rather remove gsm_wideband_input block and instead add
some block (or blocks) that is doing channelization of wideband signal.
In case clock offset correction is needed one could place
clock_offset_corrector in front of that block.

closed 2017-06-11 by ptrkrysik