ptrkrysik/gr-gsm #216: Feature request: grgsm_capture & grgsm_livemon wideband capture

Varbaek, 2016-08-29:
Hi Ptrkrysik,

I wanted to do a wideband capture recently, but as grgsm did not have that feature, I added it myself.

I basically added a -B (BANDWIDTH) parameter, and made the bandwidth dynamic and set it by default to 1.5MHz, as the BladeRF library was complaining about the minimum bandwidth and auto-setting it to 1.5MHz anyway. (By default it's a static value of 200KHz +450KHz)

Generally the wideband capture worked with a BladeRF, up to 9MSPS (sample rate) and 9MHz or 10MHz of bandwidth without any issues of using grgsm_channelize later on. (Of course, you have to set the "center frequency correctly" as well, but that's the easy part.)

I think it was around 3-4 places I had to add/modify code, so it was a small "fix", despite that it's not perfect, it does work in my case.

ptrkrysik, 2016-09-25:
@Varbaek , can you clarify what features exactly you request?

Varbaek, 2017-03-31:
I thought I had responded to your question. What I want is wideband capturing natively in grgsm_capture. 

I've already modified the code once to allow the user to set the bandwidth dynamically instead of 200KHz + 450KHz. It wasn't perfect but I was able to capture several channels and split them as well, and confirm that the split-files worked.

(Only problem was reassembling the files because of channel hopping, which didn't work, not even with airprobe-hopping.)

ptrkrysik, 2017-04-02:
W dniu 31.03.2017 o 16:05, Hans-Michael Varbaek pisze:
> I thought I had responded to your question. What I want is wideband
> capturing natively in grgsm_capture.
>
> I've already modified the code once to allow the user to set the
> bandwidth dynamically [...]

I plan to simplify grgsm_capture.py. My conviction is that if it tries
to do too much for the user (like shifting frequency, deciding what
value bandwidth should be set to) it get less universal. It shouldn't
try to write bursts files as well - just plain signal samples. Maybe
some other tool should process capture files into bursts files (for
compression/speeding up future processing of the captures).

jiangyf61, 2017-08-04:
@Varbaek 
I just want use gr-gsm to capture wideband signal.
I am truely interested in it but find difficult to modify the code.
can you help share the gr-gsm with wideband capture feature.
or maybe e-mail.
Thanks

Varbaek, 2017-08-04:
@jiangyf61 Open grgsm_capture.py and edit the lines that specify the bandwidth. Then add a new option so that you can specify the bandwidth to capture as well. Keep in mind this hack, will have that nasty "DC line" near the middle of the wideband capture, so you need to address that by e.g. shifting the frequency you capture on. (Refer to `self.shiftoff = shiftoff = 400e3`)

It also depends on your hardware, an RTL-SDR can only reliably do 2.4Msps, while a BladeRF or USRP can do significantly more.

I'm a pretty bad developer, but here's my "grgsm hack" that allows you to specify a higher amount of bandwidth to capture. Keep in mind I haven't tested this hack recently, I only tested it last year or so and just re-posted it here.

### Important capture notes:

1. When you capture this way, the **frequency** you specify is the **center** of the capture. (If you don't include the **shiftoff** that is.) 
2. If you want to capture from 950MHz to 960MHz, the center frequency would e.g. be 955MHz and you would specify 10MHz bandwidth (5MHz on each side.)
3. To compensate for the **shiftoff**, you will probably need to shift the center frequency 400KHz depending on which channels you want to capture. I recommend doing a capture, and then replay that capture in GQRX or similar program. This allows you to visually see which channels you captured, and more importantly, where that nasty DC line is. The DC line may not be a problem on equipment like BladeRF and USRPs if you calibrate them first. I'm not a software defined radio expert or pro in this topic btw, I just do this for fun out of interest. (Please feel free to correct me.) 
Concerning BladeRFs and the DC offset and IQ Imbalance please refer to:
https://github.com/Nuand/bladeRF/wiki/DC-offset-and-IQ-Imbalance-Correction 
4. The file sizes are going to be very large when you capture 10MHz, even if it's just for 60 seconds.
5. I don't remember the exact highest setting that worked with a BladeRF, but I think it was around 12MHz before my results became unreliable.
6. You can also do a raw capture with BladeRF's own tools, and most likely USRP's own tools too. (They should have one, but I only got a BladeRF and several RTL-SDRs.) From what I can recall, the raw capture using BladeRF's own tools, needs to be converted though, so I generally preferred using grgsm instead.
7. grgsm_channelize.py (https://github.com/ptrkrysik/gr-gsm/blob/master/apps/helpers/grgsm_channelize.py) actually works if you use my "hack" correctly. However, at the time of testing, I was unable to decode channel hopping voice with it. The parameters used, which are also mentioned in other guides, seem different to what I find, when I open the packets in wireshark. 

With that said though, I was able to "channelize" the wideband capture, and actual decode the individual files with grgsm_decode. 

Good luck and let us know if you succeed in decoding channel hopping SMS/Voice/etc. in a stable and reproducible manner. 

File to be edited:
https://github.com/ptrkrysik/gr-gsm/blob/9494c8940daa2a9bf09c8c1dd8eec2ea3aeb15c0/apps/helpers/grgsm_capture.py

Lines: 
```
59, add new line:
self.custom_bandwidth = custom_bandwidth= 250e3 # Default bandwidth 250KHz

75: Change: 
self.rtlsdr_source.set_bandwidth(250e3+abs(shiftoff), 0)
To the following: 
self.rtlsdr_source.set_bandwidth(custom_bandwidth+abs(shiftoff), 0)

220, add new line:
    parser.add_option("-w", "--bandwidth", dest="custom_bandwidth", type="eng_float", 
                      help="Set bandwidth [default=%default]")
```

ptrkrysik, 2018-06-02:
I'm closing the issue here as I opened another issue #409 of improving grgsm_capture. Regarding grgsm_livemon I don't know why is it in the issue title.

closed 2018-06-02 by ptrkrysik