ptrkrysik/gr-gsm #217: grgsm scanner error

dmaynor, 2016-09-07:
# grgsm_scanner -b P-GSM
linux; GNU C++ version 6.2.0 20160822; Boost_106100; UHD_003.009.005-0-unknown

Traceback (most recent call last):
  File "/usr/local/bin/grgsm_scanner", line 393, in <module>
    for info in sorted(found_list):
  File "/usr/local/bin/grgsm_scanner", line 293, in __cmp__
    return self.getKey().__cmp__(other.getKey())
AttributeError: 'NoneType' object has no attribute '__cmp__'

dmaynor, 2016-09-07:
duplicate of #166

closed 2016-09-07 by dmaynor

ptrkrysik, 2016-09-11:
Hi David,

This is a problem with channel number calculation (arfcn) - it sometimes
returns None type - in case of false positives that are outside of the
frequency band. There is need to improve scanning application (and
applications in general) but I'm too busy to do it currently.

-- 
Best Regards,
Piotr Krysik