ptrkrysik/gr-gsm #219: need help for capturing  uplink

bdyaods, 2016-09-21:
hi, Ptrkrysik,
   I'm working on capturing GSM downlink and uplink data, and I bind two rtl receivers based on your document on http://ptrkrysik.github.io/,
and downloaded your project from https://github.com/ptrkrysik/gr-gsm and https://github.com/ptrkrysik/multi-rtl, and compiled them. 
   I captured two file with mutlirtl_rx_to_cfile_2chan.py and decode them by uplink_decoding_multi_rtl.py, but there are only downlink messages
 in wireshark, no any uplink messages.
   Please give me some advices, thanks a lot.
   BTW, if run the mutlirtl_rx_to_cfile_2chan.py for several minutes, the program will be corrupted with a message: "thread[thread-per-block[31]: 
   <block vector_sink_c (10)>]: std::bad_alloc"

Yradio, 2016-09-21:
Hi Ptrkrysik,

I would like to decode multiple frequency at the same time with a lower 
CPU cost.

Is there any specific developpement already done for Ettus USRP in order 
to take advantage of the FPGA ?
  - with gr-gsm ?
  - with other projects ?

Regards,

--
Y.

ptrkrysik, 2016-09-22:
W dniu 21.09.2016 o 21:16, Yradio pisze:
> Hi Ptrkrysik,
> 
> I would like to decode multiple frequency at the same time with a lower
> CPU cost.
> 
> Is there any specific developpement already done for Ettus USRP in order
> to take advantage of the FPGA ?
> - with gr-gsm ?
> - with other projects ?
> 
> Regards,
> 
> --
> Y.

Hi Yradio and others. For discussions which are not directly related to
gr-gsm development (bugs, specific propositions of improvements) please
use the project's mailing list (the link is in the project's readme).

-- 
Best Regards,
Piotr Krysik

ptrkrysik, 2016-09-25:
@bdyaods only the "message: "thread[thread-per-block[31]: ]: std::bad_alloc" fits into github issues. Other things belong to gr-gsm or (better) multi-rtl mailing lists. However - the above possible problem might be a problem of multi-rtl not gr-gsm.

ptrkrysik, 2016-09-25:
@Yradio, regarding taking advantage of USRP's FPGA's - look for RFNoC branch of Ettus FPGAs code. Better place to ask for it is Ettus mailing list. If you want to discuss it here - please use gr-gsm's mailing list.

closed 2016-09-25 by ptrkrysik