ptrkrysik/gr-gsm #221: error on with gnuradio 3.7.10 (manual compilation)

SonyKucing, 2016-09-24:
my gnuradio has update with version 3.7.10 with PPA myriadrf , but when run grgsm_livemon got the error, then i try to manual compilation but the same error show the libgnuradio version.... 

```
Traceback (most recent call last):
  File "/usr/local/bin/grgsm_livemon", line 32, in <module>
    import osmosdr
  File "/usr/lib/python2.7/dist-packages/osmosdr/__init__.py", line 26, in <module>
    from osmosdr_swig import *
  File "/usr/lib/python2.7/dist-packages/osmosdr/osmosdr_swig.py", line 28, in <module>
    _osmosdr_swig = swig_import_helper()
  File "/usr/lib/python2.7/dist-packages/osmosdr/osmosdr_swig.py", line 24, in swig_import_helper
    _mod = imp.load_module('_osmosdr_swig', fp, pathname, description)
ImportError: libgnuradio-runtime.so.3.7.9: cannot open shared object file: No such file or directory

```

ptrkrysik, 2016-09-24:
This is problem with gr-osmosdr - one of gr-gsm's dependencies. And it
looks that the problem is how you installed it - not some bug in
gr-osmosdr. Did you recompile gr-gsm with dependencies after you changed
your gnuradio version?

SonyKucing, 2016-09-24:
thanks bro....  i recompile manualy gr-osmosdr then recompile gr-gsm.. now all works fine... thanks.... :*

closed 2016-09-25 by ptrkrysik