ptrkrysik/gr-gsm #227: Ошибка установки GR-GSM

Jusse, 2016-10-02:
Петр помоги решить проблему установки GR-GSM. При установке выходит следующие:
src/thrift/transport/TSSLSocket.cpp: In constructor ‘apache::thrift::transport::SSLContext::SSLContext(const apache::thrift::transport::SSLProtocol&)’:
src/thrift/transport/TSSLSocket.cpp:143:37: error: ‘SSLv3_method’ was not declared in this scope
     ctx_ = SSL_CTX_new(SSLv3_method());
                                     ^
Makefile:1292: ошибка выполнения рецепта для цели «src/thrift/transport/TSSLSocket.lo»
make[4]: *** [src/thrift/transport/TSSLSocket.lo] Ошибка 1
make[4]: выход из каталога «/root/pybombs/src/apache-thrift/lib/cpp»
Makefile:1586: ошибка выполнения рецепта для цели «all-recursive»
make[3]: *** [all-recursive] Ошибка 1
make[3]: выход из каталога «/root/pybombs/src/apache-thrift/lib/cpp»
Makefile:550: ошибка выполнения рецепта для цели «all-recursive»
make[2]: *** [all-recursive] Ошибка 1
make[2]: выход из каталога «/root/pybombs/src/apache-thrift/lib»
Makefile:620: ошибка выполнения рецепта для цели «all-recursive»
make[1]: *** [all-recursive] Ошибка 1
make[1]: выход из каталога «/root/pybombs/src/apache-thrift»
Makefile:542: ошибка выполнения рецепта для цели «all»
make: *** [all] Ошибка 2
PyBOMBS.Packager.source - ERROR - Build failed. See output above for error messages.
PyBOMBS.Packager.source - ERROR - Problem occurred while building package apache-thrift:
Build failed.
PyBOMBS.install_manager - ERROR - Error installing package apache-thrift. Aborting.

ptrkrysik, 2016-10-03:
You have problem with compilation of one of gr-gsm's dependencies
(apache-thrift) on Kali linux. I can't help to make it work -
installation on Kali proved to be difficult many times before. Unless
there appear someone who will take care of installation process on Kali
- this state won't change. My current priority is to keep gr-gsm
installable on Ubuntu and Debian and (I hope in the near future) to
prepare a package for these distributions.

Please use the mailing list for reporting problems with installation in
the future:
https://github.com/ptrkrysik/gr-gsm#mailing-list

axilirator, 2016-10-19:
Also, it is better to use English in international correspondence.

closed 2016-10-21 by ptrkrysik