ptrkrysik/gr-gsm #229: somebody help please

Mahmoudshakra, 2016-10-20:
i have installed gr-gsm by Compiling and installing   Pybombs
and when i am try to run airprobe_rtlsdr.py it gives me this msg : 

Traceback (most recent call last):
  File "/usr/local/bin/airprobe_rtlsdr.py", line 28, in <module>
    from gnuradio.qtgui import Range, RangeWidget
ImportError: cannot import name Range

Mahmoudshakra, 2016-10-20:
when i tried to run gnuradio-companion it gives me : 

Traceback (most recent call last):
  File "/usr/local/bin/gnuradio-companion", line 99, in <module>
    run_main()
  File "/usr/local/bin/gnuradio-companion", line 87, in run_main
    from gnuradio.grc.main import main
ImportError: No module named main

ptrkrysik, 2016-10-21:
Probably you have problem with your gnuradio installation. If you intend to continue this thread please go to the gr-gsm's mailing list.

closed 2016-10-21 by ptrkrysik