ptrkrysik/gr-gsm #237: Feature request: decoding TCH/9.6

bdyaods, 2016-11-23:
hi, all:
    I want to decode the data in TCH/F9.6, but I can't find any guide, can you give me some advices?
    Thanks a lot

closed 2017-03-21 by bdyaods

ptrkrysik, 2017-03-21:
W dniu 23.11.2016 o 09:19, bdyaods pisze:
> hi, all:
> I want to decode the data in TCH/F9.6, but I can't find any guide, can
> you give me some advices?
> Thanks a lot
> 
Currently decoding of data channels is not supported. The support can be
quite easily added as libosmocore has the decoding implementation.

axilirator, 2017-03-22:
> The support can be quite easily added as libosmocore has the decoding implementation.

It already has ;)

E3V3A, 2017-07-01:
@ptrkrysik So was this implemented?

ptrkrysik, 2017-07-24:
The support of encoding/decoding different channel codes was ported by
Vadim to libosmocore. But it will need to be integrated into gr-gsm.