ptrkrysik/gr-gsm #243: Why I can‘t get any data when I pass argument the "-o " that is a large value to grgsm_capture?

xWeiwei, 2016-12-27:
I want to modify the shiftoff default value with grgsm_capture, I can't get any data when i set the value that is 1200K, but pass 1000k to grgsm_capture iis work fine.  I use SDR device is USRP1.

ptrkrysik, 2016-12-29:
@weiwei, what is the sample rate of your capture?

xWeiwei, 2016-12-29:
@ptrkrysik, the sample rate is 100000

ptrkrysik, 2016-12-29:
Then the shiftoff 1.2e6 Hz doesn't make sense in this case. On another
side - the shiftoff 1e6 doesn't do much on. It's like 360 degrees rotation.

The idea of shiftoff is to shift DC offset out of the bandwidth of a
captured GSM channel. You shouldn't shift more than
sample_rate/2-gsm_channel_bandwidth/2 (+/- no more than about 0.4e6 Hz
in case of sample_rate=1e6).

If you have USRP you can do the shifting inside of your USRP's fpga
(pass tune_request(...) as frequency to the UHD object) - but you will
have to make your own program for capturing signal in gnuradio-companion.

xWeiwei, 2016-12-29:
@ptrkrysik , thanks for your reply.

closed 2017-06-11 by ptrkrysik