ptrkrysik/gr-gsm #246: Manual install problem. in gr_gsm

pgiz, 2017-01-08:
Install type : Manual install make step

OS : 14.04.1-Ubuntu x86_64 GNU/Linux

Before run make. cmake ran with option to solve include directory issues. 
cmake -DCMAKE_INSTALL_PREFIX=/usr .. 

Terminal
[ 61%] Built target grgsm
[ 63%] Built target _grgsm_swig_doc_tag
[ 66%] Built target grgsm_swig_swig_doc
[ 67%] Built target _grgsm_swig_swig_tag
[ 73%] Built target grgsm_swig_swig_2d0df
[ 75%] Generating grgsm_swig.py
[ 76%] Generating grgsm_swigPYTHON_wrap.cxx
[ 76%] Generating grgsm_swigPYTHON_wrap.cxx
[ 77%] Building CXX object swig/CMakeFiles/_grgsm_swig.dir/grgsm_swigPYTHON_wrap.cxx.o
c++: error: /home/something/gr-gsm/build/swig/grgsm_swigPYTHON_wrap.cxx: No such file or directory
c++: fatal error: no input files
compilation terminated.
make[2]: *** [swig/CMakeFiles/_grgsm_swig.dir/grgsm_swigPYTHON_wrap.cxx.o] Error 4
make[1]: *** [swig/CMakeFiles/_grgsm_swig.dir/all] Error 2
make: *** [all] Error 2

Thank you.

pgiz, 2017-01-09:
gcc version 4.8.4 (Ubuntu 4.8.4-2ubuntu1~14.04.3)

SWIG Version 3.0.11
Compiled with g++ [x86_64-pc-linux-gnu]
Configured options: +pcre

GNU Make 3.81

SuperJesseH, 2017-02-17:
It seems like I'm having the same problem.

ptrkrysik, 2017-06-11:
The manual installation was written by soemeone who tested it on Kali Linux. For Ubuntu 14.04 this description won't work. You will need to compile GNU Radio (version >= 3.7.3) manually in this case.

closed 2017-06-11 by ptrkrysik