ptrkrysik/gr-gsm #249: Could not find port: bursts in: bursts system

DemonLord98, 2017-01-18:
I installed gr-gsm from ppa on ubuntu 16.04 LTS and I have this problem when I run grgsm_livemon

`~$ grgsm_livemon
linux; GNU C++ version 5.4.0 20160609; Boost_105800; UHD_003.010.001.000-release

gr-osmosdr v0.1.x-xxx-xunknown (0.1.5git) gnuradio 3.7.9
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy soapy redpitaya 
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Exact sample rate is: 2000000.052982 Hz
[R82XX] PLL not locked!
Could not find port: bursts in:
bursts
system

Traceback (most recent call last):
  File "/usr/bin/grgsm_livemon", line 270, in <module>
    tb = grgsm_livemon(fc=options.fc, gain=options.gain, ppm=options.ppm, samp_rate=options.samp_rate, shiftoff=options.shiftoff, args=options.args)
  File "/usr/bin/grgsm_livemon", line 143, in __init__
    self.gsm_sdcch8_demapper_0 = grgsm.gsm_sdcch8_demapper(1)
  File "/usr/lib/python2.7/dist-packages/grgsm/gsm_sdcch8_demapper.py", line 39, in __init__
    self.msg_connect((self.gsm_universal_ctrl_chans_demapper_0, 'bursts'), (self, 'bursts'))    
  File "/usr/lib/python2.7/dist-packages/gnuradio/gr/hier_block2.py", line 59, in wrapped
    func(self, src.to_basic_block(), srcport, dst.to_basic_block(), dstport)
  File "/usr/lib/python2.7/dist-packages/gnuradio/gr/hier_block2.py", line 131, in msg_connect
    self.primitive_msg_connect(*args)
  File "/usr/lib/python2.7/dist-packages/gnuradio/gr/runtime_swig.py", line 3584, in primitive_msg_connect
    return _runtime_swig.hier_block2_sptr_primitive_msg_connect(self, *args)
RuntimeError: invalid msg port in connect() or disconnect()`

PS: I have gnu radio 3.7.10.1

ptrkrysik, 2017-06-11:
I removed installation from packages as this work wasn't done (and I don't know when it will be finished).

closed 2017-06-11 by ptrkrysik

paulkilby, 2017-07-08:
I'm having exactly the same issue, even down to the line numbers in the error message.
I installed using PyBOMBS. It looks a little bit like issue #163 but the solution there did not solve the problem. Any assistance, greatly appreciated.