ptrkrysik/gr-gsm #251: cannot import name qtgui

tom-2015, 2017-01-21:
When I try to start grgsm_livemon I get error:

File "/usr/local/bin/grgsm_livemon", line 25, in <module>
from gnuradio import qtgui
ImportError: cannot import name qtgui

Andru1999, 2017-01-22:
May be you must install qt library. Tray `pip install qt`

tom-2015, 2017-01-22:
I get error: `could not find qt`
I tried: `pip install qtgui` which installed qtgui but still have the same error.

Andru1999, 2017-01-22:
Which version of python do you use?

Andru1999, 2017-01-22:
Try to change the 25 line in the grgsm_livemon on 'from PyQt4 import qtgui'

tom-2015, 2017-01-22:
Python 2.7.12+
I tried to reinstall everything but got error: `Error installing package gr-iqbal unable to fetch recipe gr-iqbal`

Also changed line 25 to 'from PyQt4 import qtgui' but this gives the same error message it cannot import qtgui.

Andru1999, 2017-01-22:
try this 'sudo add-apt-repository -y ppa:ptrkrysik/gr-gsm sudo apt update && sudo apt install gr-gsm'

tom-2015, 2017-01-22:
```
W: The repository 'http://ppa.launchpad.net/ptrkrysik/gr-gsm/ubuntu yakkety Release' does not have a Release file.
N: Data from such a repository can't be authenticated and is therefore potentially dangerous to use.
N: See apt-secure(8) manpage for repository creation and user configuration details.
E: Failed to fetch http://ppa.launchpad.net/ptrkrysik/gr-gsm/ubuntu/dists/yakkety/main/binary-amd64/Packages  404  Not Found
E: Some index files failed to download. They have been ignored, or old ones used instead.
```

ptrkrysik, 2017-01-23:
This probably have more to do with your installation of GNU Radio than
with gr-gsm. Try to reinstall GNU Radio correctly.

ptrkrysik, 2017-01-23:
W dniu 22.01.2017 o 11:44, Andru1999 pisze:
> try this 'sudo add-apt-repository -y ppa:ptrkrysik/gr-gsm sudo apt
> update && sudo apt install gr-gsm'
> 

This repository was a result of my experiments with packaging gr-gsm.
As this effort is not finished yet I don't recommend you to use these
packages.

ptrkrysik, 2017-02-17:
I'm closing the issue here. If you didn't find the solution - please ask on the project's mailing list.

closed 2017-02-17 by ptrkrysik