ptrkrysik/gr-gsm #252: grgsm_scanner TypeError

pachulo, 2017-01-23:
Hi!

I'm getting an error like this when using the grgsm_scanner app:

    root@latitude:~# grgsm_scanner --ppm=22 --band=E-GSM --verbose                                            
    linux; GNU C++ version 5.4.0 20160609; Boost_105800; UHD_003.010.001.000-release
    
    ....

    ARFCN:  118, Freq:  958.6M, CID:     0, LAC:  XXXX, MCC: 214, MNC:   1, Pwr: -38
    |---- Configuration: Unknown
    |---- Cell ARFCNs: 118, 122
    |---- Neighbour Cells: 

    Traceback (most recent call last):
    File "/usr/local/bin/grgsm_scanner", line 394, in <module>
    for info in sorted(found_list):
    File "/usr/local/bin/grgsm_scanner", line 286, in __cmp__
    return self.getKey().__cmp__(other.getKey())
    TypeError: int.__cmp__(x,y) requires y to be a 'int', not a 'NoneType'

ptrkrysik, 2017-01-23:
This usually happens when there is false positive from outside GSM band
detected. The frequency to ARFCN converter returns None in this
situation. It is needed to correct this bug but also source of the
problem which is false positives coming from one edge of band of RTL-SDR
receiver.

E3V3A, 2017-06-21:
@ptrkrysik What do you mean with "false positive from outside GSM"? (What is the scanner looking for, that can get false positives?)

ptrkrysik, 2017-07-24:
Hi. RTL-SDR's have some non-linearities that cause copy of signal from one frequency to be seen on another one. grgsm_scanner tries to process 2MHz of bandwidth at once, so it processes also the copy. If the copy falls outside of GSM frequency band function converting frequency to ARFCN returns none. Returning NoneType is not that bad but this situation will need to be handled.

Having false positives is pretty bad though. It can be solved through simplification of the scanning algorithm, but scanning will take longer.

E3V3A, 2017-07-24:
I see those "images" more often when the center freq is where I measure. 

@ptrkrysik 
 * Do you think it would it help to use the `-o <freq_shift>` option? 
 * Could we use some kind of algo to find those copies? (Eg. Are they equally spaced across the spectrum?)

petterreinholdtsen, 2017-08-21:
I ran into this backtrace:

```
$ grgsm_scanner -v
linux; GNU C++ version 6.3.0 20170221; Boost_106200; UHD_003.009.005-0-unknown

Traceback (most recent call last):
  File "/usr/bin/grgsm_scanner", line 394, in <module>
    print info
  File "/usr/bin/grgsm_scanner", line 296, in __repr__
    return "ARFCN: %4u, Freq: %6.1fM, CID: %5u, LAC: %5u, MCC: %3u, MNC: %3u, Pwr: %3i" % (self.arfcn, self.freq/1e6, self.cid, self.lac, self.mcc, self.mnc, self.power)
TypeError: %u format: a number is required, not NoneType
```

I worked around the crash by simply not adding entries without arfcn entries to the output list, like this:

```
--- /usr/bin/grgsm_scanner.orig      2016-04-24 17:00:00.000000000 +0200
+++ /usr/bin/grgsm_scanner  2017-08-21 22:49:48.653906776 +0200
@@ -388,7 +389,10 @@
                 neighbour_list = scanner.gsm_extract_system_info.get_neighbours(chans[i])
 
                 info = channel_info(grgsm.arfcn.downlink2arfcn(found_freqs[i], options.band), found_freqs[i], cell_ids[i], lacs[i], mccs[i], mncs[i], ccch_confs[i], powers[i], neighbour_list, cell_arfcn_list)
-                found_list.append(info)
+                if info.arfcn:
+                    found_list.append(info)
+                else:
+                    print("Found strange entry witout ARFNC")
                             
             for info in sorted(found_list):
                 print info
```

Perhaps something to include in the official source?

petterreinholdtsen, 2017-08-23:
I just ran into a similar problem:

```
linux; GNU C++ version 6.3.0 20170221; Boost_106200; UHD_003.009.005-0-unknown

Traceback (most recent call last):
  File "/usr/bin/grgsm_scanner", line 396, in <module>
    for info in sorted(found_list):
  File "/usr/bin/grgsm_scanner", line 288, in __cmp__
    return self.getKey().__cmp__(other.getKey())
AttributeError: 'NoneType' object has no attribute '__cmp__'
```

E3V3A, 2017-08-23:
@petterreinholdtsen 
* Please show the full command you run to get that output.
* Please try to add the `-o <freq_shift>`  option and see what happens. 

**A not so perfect explanation**: When not using a properly DC or FM filtered antenna input, you often get huge DC junk and other images from strong sources when using the center frequency. By offsetting the center frequency, you can avoid some of this.

petterreinholdtsen, 2017-08-29:
I believe this issue is fixed in commit aed9f5cc80e2c123eb0294dff21e853c1ffa376b .

petterreinholdtsen, 2017-09-02:
The fix for this is in the master branch and the 0.41.1 release.  Time to close this issue?

closed 2017-09-11 by ptrkrysik