ptrkrysik/gr-gsm #255: Multithread glitches in output

Alster, 2017-02-11:
Sometimes i got multithread glitches in output:
`200641 309497: 110011100111001000010011001001111100001101100010100100010100010101000011010010110110001100000100101100110001000011
200713 310163: 100000010000010111010100000000100001000101111101000000001000100101010101010100010100100000010000010101110101010010
200714 310196: 10101110110101110101000100100010111110111101110101010010000010200713 310163:  03 20 0d 05 18 03 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b
200729 310665:  05 01 03 03 49 06 1d 00 00 00 00 00 40 40 01 00 40 22 02 00 00 24 61
1111110101010100000000000010111111111101010101001010
200715 310229: 000000001111110100010100001010000100010111110001010000001001000101011101010100000010000001000011011101000001101000
200716 310262: 010101001000001111101101111101010000000000101111111111110100010000000010111010010111010000010010101010101001111101`
Looks like it's only with "--print-bursts" option enabled.

ptrkrysik, 2017-02-12:
Bursts and messages are printed by separate blocks.
Some way of synchronizing printing is needed. It probably should be done
by a single block.

One solution is to create one block for printing everything.

Another solution is to add a block that simply prints strings and turn
bursts printer and message printer into blocks that convert bursts and
messages into strings.

Alster, 2017-02-12:
I chosen last solution with new block for finally printing messages. I never used GNU Radio before, and bad in cpp. As a result:
https://github.com/Alster/gr-gsm/commit/6c78eed733609ec03fa66d9ee7eb49a81041e07e
Two problems:
1 Big performance degradation
2 Messages drawn well, but order in sequence is wrong:
214564 329964: 101010101101010111111110001010101010110111111111101100000011001010010101111110110001001010101001000111111110000110
214565 329997: 000000000101110101010100000000000001010101010101010000001000000000010101110100010000000000010101011101010100011000
214566 330030: 101010101101010101000100001010101110111111110101000000001010111011111101010000000000001010101011010101010101000000
214567 330063: 000000010101010100000000001000010001011111010101000010100001010101010101010101001000000000000111110101000000101000
214568 330096: 010000000000101110111111010100010000001010101011110101010100010000001010111011111101010101001000101011111111010101
214565 329997:  40 94 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b
214569 330129: 000000000101110101010100000000000001010101010101010000001000000000010101110100010000000000010101011101010100010000
214569 330129:  40 94 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b
214570 330162: 101010101101010101000100001010101110111111110101000000001010111011111101010000000000001010101011010101010101000000
214571 330195: 000000010101010100000000001000010001011111010101000010100001010101010101010101001000000000000111110101000000101000

Looks like i just to synchronize messages printing, but its generation is still async or am I missing something?

Alster, 2017-02-12:
How it works? Maybe just add shared mutex into bursts/message_printer at directly std::cout line? Where i can add shared mutex?

ptrkrysik, 2017-02-13:
These blocks are asynchronous as they are using message passing.

It might be not that simple as adding mutex to get the streams
synchronized. I was thinking about some buffer that would hold few
bursts/messages, would sort them based on frame number and then output
them with slight delay. But this approach might not work in 100% cases
(i.e. if delay between streams is higher than size of the buffer).

Alster, 2017-02-14:
Thanks for the reply.
My point is to got "non-glitched" messages. Correct order is not necessary for me, but performance is important. Finally, avoiding adding separate block (see my commit in previous comment) i just make small modify of bursts_printer to write messages into file (and not into stdout), its enough for me.
Issue still will be opened for better solution.

commit 75ae9cc3 (2018-09-11): Format the output into ostringstream and then write it at once.

This fixes the garbled output when multiple printers are used in a
flowgraph.

closes #255
closes #420

Change-Id: I1012ed26371b4c67163545652f0a1ce0f576af9e
closed 2018-09-11 by osmocom-gerrit